mne-tools/mne-python #11373: Automatic epoch removal shifts index in epochs.plot() but not in epochs.drop()

nvbln, 2022-12-16:
### Description of the problem

One epoch in my dataset was automatically removed due to it being too close to the start of the experiment (making it impossible to apply a baseline with negative times). Due to this, the indices displayed in epochs.plot() and the bad epochs indices that are printed after going through the epochs are shifted by one index (i.e. removing the first and second epochs removes [1, 2] not [0, 1]). So far so good. However, epochs.drop() starts with index 0 as if the automatically dropped epoch had never existed. As a result, in the case of the example, not epochs 1 and 2 get dropped, but 2 and 3.

Possibly related to #7260

### Steps to reproduce

```Python
import mne

# Load the sample dataset.
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')

raw = mne.io.read_raw_fif(sample_data_raw_file).crop(tmax=60)
events = mne.find_events(raw, stim_channel='STI 014')

# Set the first event to time index 0, such that it'll be too close to the start of the data.
events[0, 0] = 0

# Now try to extract the epoch with a baseline before it.
epochs = mne.Epochs(raw, events, tmin=-1, tmax=0.7, baseline=(-1, 0))

# Plot the epochs, the first epoch will be automatically dropped.
epochs.plot()

# Now if we click on the epochs that we want to drop in the plot,
# the dropping will work as expected. But if we take the list of
# dropped epochs, recreate the epochs object and use those indices
# to drop the epochs automatically, the dropped epochs will be shifted.
# Let's try just that. Imagine we just dropped epoch 1 in the plot.
# We get the following array:
dropped_epochs = [1]

# Now we drop this epochs and plot again.
epochs.drop(dropped_epochs)
epochs.plot()

# In the plot you should see that not epoch 1, but epoch 2 got dropped.
```

### Expected results

Either epoch 1 is dropped when giving index 1 to epochs.drop() or epochs.plot() is modified such that the index starts at 0 when epochs are automatically dropped.

### Actual results

The indices in epochs.plot() do not correspond to the epoch indices in epoch.drop() after epochs are dropped automatically.

### Additional information

Output exceeds the [size limit](command:workbench.action.openSettings?%5B%22notebook.output.textLineLimit%22%5D). Open the full output data [in a text editor](command:workbench.action.openLargeOutput?3595f6e4-e29e-4346-bca4-682b8b067b67)
Platform:         Linux-5.15.0-56-generic-x86_64-with-glibc2.35
Python:           3.10.6 (main, Nov 14 2022, 16:10:14) [GCC 11.3.0]
Executable:       /home/nathan/Workspace/NE2/Cybathlon/venv/bin/python
CPU:              x86_64: 4 cores
Memory:           7.7 GB

mne:              1.2.3
numpy:            1.23.0 {OpenBLAS 0.3.20 with 4 threads}
scipy:            1.8.1
matplotlib:       3.5.2 {backend=module://matplotlib_inline.backend_inline}

sklearn:          1.1.1
numba:            0.56.4
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           1.4.3
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             Not found
ipympl:           Not found
...
mne_features:     0.2.1
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

larsoner, 2022-12-16:
> However, epochs.drop() starts with index 0 as if the automatically dropped epoch had never existed. 

To me this behavior is correct (or at least one way to be correct) and should be better documented, perhaps in a tutorial. If you want to drop based on the *original* indices, you probably want to use `epochs.drop(np.where(np.in1d(epochs.selection, original_to_drop)))` or so. In other words, whenever you `.drop`, any new indexing operation refers to the indices of the *remaining* epochs, and if you want to think about the old ones, you need to use `epochs.selection`. Does that make sense?

nvbln, 2022-12-16:
Yes I realise now that this behaviour makes sense and it is how I normally use it.
Something that I couldn't reproduce in the sample script but what happened to me was for the first epoch to drop as soon as I created them using mne.Epochs(). That is why the indices didn't match to the ids of the epochs when I wanted to drop the bad ones (which I do immediately after creating them, to prevent having to reselect them during re-analysis).

My problem was in that sense two-fold:
1. I did not realise that an epoch was dropped automatically when creating the epochs.
2. I was not aware of how to use the original indices to drop the correct epochs reliably.

For the second point I completely agree with your solution. For the first point, I would suggest to be more explicit about automatically dropped epochs if possible, perhaps by printing the log message in red, as it is easy to overlook when a lot of log messages are printed.

larsoner, 2022-12-16:
> I did not realise that an epoch was dropped automatically when creating the epochs.

I don't think they are actually -- but rather they are/should be when the epochs are accessed/loaded. In your code snippet, they are accessed by `epochs.plot`. Does that make sense?

So I think the thing we should tell people is to be careful about indexing when data might be accessed, and to use `epochs.drop_bad()` after creation to make sure any invalid epochs have been removed.

nvbln, 2022-12-17:
Makes sense, thanks a lot!

Dev-satish, 2023-02-15:
@larsoner can you provide with the direction for this issue? In which file of `doc` shall I do the change?

larsoner, 2023-02-15:
It wouldn't actually be in the `doc/` folder. It would either be in one of our `tutorials/` or in `mne/epochs.py` for the `Epochs` class, which ends up populating our API docs. If you are new to MNE I would recommend going through some basic tutorials (if you haven't already) until this issue makes sense, then thinking about where it would fit best. I don't know offhand...

divyani-22, 2026-10-07:
Hi! Is anyone still working on this? If not, I'd like to take it. From larsoner's note, I plan to document in the Epochs docstring (mne/epochs.py) that indices after dropping refer to the remaining epochs, and that epochs.selection and drop_bad() should be used for original indices. Does that sound right?