mne-tools/mne-python #11376: [BUG/Feature request] plotting dipoles gives unsatisfying and buggy results

jasmainak, 2022-12-18:
### Description of the problem

There doesn't seem to be a satisfying way to plot the dipoles obtained from dipole fit in MNE-Python. There are 4 options:

1. Orthoview: this is a bit low-contrast and it's not possible to traverse the slices. It's very difficult to identify anatomical landmarks as a result
[image]

2. Outlines: Similar problem as 1. If you get the dipoles at the wrong location, you can't interactively explore how far from the expected location you are.

3. Arrow/Sphere: seems buggy for me (see below)

Here is an example figure from a paper:

[image]

Why not have a vanilla option that resembles this with an option to interactively traverse slices? I saw this issue: https://github.com/mne-tools/mne-python/issues/2995, not sure if it's related. Please feel free to migrate the discussion there if relevant.

### Steps to reproduce

```Python
import os.path as op
import numpy as np
import matplotlib.pyplot as plt

import mne
from mne.forward import make_forward_dipole
from mne.evoked import combine_evoked
from mne.simulation import simulate_evoked

from nilearn.plotting import plot_anat
from nilearn.datasets import load_mni152_template

data_path = mne.datasets.sample.data_path()
subjects_dir = data_path / 'subjects'
fname_ave = data_path / 'MEG' / 'sample' / 'sample_audvis-ave.fif'
fname_cov = data_path / 'MEG' / 'sample' / 'sample_audvis-cov.fif'
fname_bem = subjects_dir / 'sample' / 'bem' / 'sample-5120-bem-sol.fif'
fname_trans = data_path / 'MEG' / 'sample' / 'sample_audvis_raw-trans.fif'
fname_surf_lh = subjects_dir / 'sample' / 'surf' / 'lh.white'

evoked = mne.read_evokeds(fname_ave, condition='Right Auditory',
                          baseline=(None, 0))
evoked.pick_types(meg=True, eeg=False)
evoked_full = evoked.copy()
evoked.crop(0.07, 0.08)

# Fit a dipole
dip = mne.fit_dipole(evoked, fname_cov, fname_bem, fname_trans)[0]

# Plot the result in 3D brain with the MRI image.
dip.plot_locations(fname_trans, 'sample', subjects_dir, mode='arrow')
```

### Expected results

Documentation says:

"Plot in 3D mode using PyVista with the given glyph type."

I am not sure what to expect since there is no example online ...

### Actual results

I see this, not sure if it's an installation problem:

[image]

### Additional information

In [11]: mne.sys_info()
Platform:         macOS-11.5.2-x86_64-i386-64bit
Python:           3.9.7 | packaged by conda-forge | (default, Sep  2 2021, 17:58:46)  [Clang 11.1.0 ]
Executable:       /Users/mainak/anaconda3/envs/mne/bin/python
CPU:              i386: 4 cores
Memory:           8.0 GB

mne:              1.3.dev0
numpy:            1.21.2 {unknown linalg bindings}
scipy:            1.7.1
matplotlib:       3.4.3 {backend=MacOSX}

sklearn:          0.24.2
numba:            Not found
nibabel:          3.2.1
nilearn:          0.8.0
dipy:             1.4.1
openmeeg:         Not found
cupy:             Not found
pandas:           1.3.3
pyvista:          0.32.1 {OpenGL 4.1 INTEL-16.5.2 via Intel(R) HD Graphics 6000}
pyvistaqt:        0.5.0
ipyvtklink:       0.2.2
vtk:              9.0.1
qtpy:             1.11.0 {PyQt5=5.12.9}
ipympl:           0.9.1
pyqtgraph:        0.12.4
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.2.5
mne_connectivity: Not found
mne_icalabel:     Not found

larsoner, 2022-12-20:
Agreed the 3D one looks buggy.

So far our dipole plotting doesn't really support interactivity in the way you want. To me it sound like you want something like a [`vtkImagePlaneWidget`](https://kitware.github.io/vtk-examples/site/Python/Widgets/ImplicitPlaneWidget2/) like freeview has, e.g., the lower-right view here (taken from [here](https://andysbrainbook.readthedocs.io/en/latest/FreeSurfer/FS_ShortCourse/FS_06_Freeview.html#viewing-your-data)):

[image]

Is that right? If so we could make such a mode, and maybe add it to the broken 3D one while fixing it. But it would be a bit of work

jasmainak, 2022-12-20:
That would of course be super-awesome. The freeview visualizer is really good. But even something simpler like what we [implemented with volume source estimates](https://github.com/mne-tools/mne-python/blob/654248a3ea9d5b106eaba51b9a74b93a531b3746/mne/viz/_3d.py#L2168) might be an improvement. Being able to click on a point on one of the slices and updating the other slices from that. The main thing is the contrast needs to be better + being able to change slices somehow. Arrow-based keyboard shortcuts to navigate slices can also be a quality-of-life improvement!