mne-tools/mne-python #11382: Inconsistent .get_data() behavior for bad channels in mne.io.Raw and mne.time_frequency.Spectrum objects

moritz-gerster, 2022-12-20:
### Description of the problem

`raw.get_data()` returns all channels (including bad channels) by default. 

`spectrum.get_data()` returns only good channels by default. This is inconsistent and therefore confusing. I just spent hours understanding why the indices in `spectrum.ch_names` don't match the data indices in `spectrum.get_data()`.

### Steps to reproduce

```Python
import mne

from mne import create_info
from mne.io import RawArray
from numpy.random import default_rng

def create_toy_data(n_channels=2, duration=25, sfreq=250, seed=None):
    rng = default_rng(seed)
    data = rng.standard_normal(size=(n_channels, 50 * sfreq)) * 5e-6
    info = create_info(n_channels, sfreq, "eeg")
    return RawArray(data, info)

raw = create_toy_data()

raw.info["bads"] = ['0']
spectrum = raw.compute_psd(picks=raw.ch_names, verbose=False)

assert len(spectrum.get_data()) == len(spectrum.ch_names), "spectrum.get_data() should return bad channels as raw.get_data() does."

---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
Cell In [22], line 1
----> 1 assert len(spectrum.get_data()) == len(spectrum.ch_names), "spectrum.get_data() should return bad channels as raw.get_data() does."

AssertionError: spectrum.get_data() should return bad channels as raw.get_data() does.
```

### Expected results

I would suggest to return all channels by default in both cases. I plotted my channels by looping over `spectrum.get_data(index)` and using the `label=spectrum.ch_names(index)`. I could imagine that more people plan to do this as it is really important to plot the PSD of bad channels to check whether they are actually bad. 

`len(spectrum.get_data()) != len(spectrum.ch_names)` is not intuitive in my opinion.

### Actual results

[inconsistent_get_data.pdf](https://github.com/mne-tools/mne-python/files/10267712/inconsistent_get_data.pdf)

### Additional information

```
Platform:         macOS-12.6.1-arm64-arm-64bit
Python:           3.10.6 | packaged by conda-forge | (main, Aug 22 2022, 20:41:22) [Clang 13.0.1 ]
Executable:       /Users/moritzgerster/anaconda3/envs/bids_neu/bin/python
CPU:              arm: 10 cores
Memory:           16.0 GB

mne:              1.2.2
numpy:            1.23.4 {OpenBLAS 0.3.21 with 10 threads}
scipy:            1.9.3
matplotlib:       3.6.2 {backend=module://matplotlib_inline.backend_inline}

sklearn:          1.1.3
numba:            0.56.3
nibabel:          4.0.2
nilearn:          0.9.2
dipy:             1.5.0
openmeeg:         Not found
cupy:             Not found
pandas:           1.5.1
pyvista:          0.37.0 {OpenGL 4.1 Metal - 76.3 via Apple M1 Pro}
pyvistaqt:        0.9.0
ipyvtklink:       0.2.2
vtk:              9.2.2
qtpy:             2.3.0 {PyQt5=5.15.6}
ipympl:           Not found
pyqtgraph:        0.13.1
pooch:            v1.6.0

mne_bids:         0.12.dev0
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.4.0
mne_connectivity: Not found
mne_icalabel:     Not found
```

agramfort, 2022-12-20:
+ 0.25 on this :)

> Message ID: ***@***.***>
>

larsoner, 2022-12-20:
> raw.get_data() returns all channels (including bad channels) by default.
>
> spectrum.get_data() returns only good channels by default.

Unfortunately across MNE `picks=None` can mean process/return any one of several things:

1. All channels
2. All good channels
3. All data channels
4. All good data channels

And it's quite likely we are not super consistent across functions, possibly even within a single type (e.g., for just Epochs-related functions and methods). Before trying to change this one single default, I think someone needs to look at all (or almost all) such cases, make a table, and we can think about how to make things more consistent.

Such changes are breaking for people's code, so require a deprecation cycle :(

drammock, 2023-01-27:
after dev meeting discussion, @larsoner and @agramfort and I agree that `Spectrum.get_data()` should change to be like `Raw.get_data()`.

closed 2023-10-04 by drammock