mne-tools/mne-python #11383: Current multitaper implementation too slow for continuous raw data

moritz-gerster, 2022-12-20:
### Describe the new feature or enhancement

Currently, it is basically impossible to use `raw.compute_psd(method="multitaper", bandwidth=1)`. For long data, it calculates hundreds of DPSS windows and therefore it takes way too long. In the example below, I used `bandwidth=1` which is one of the fastest settings but it takes 6 minutes for a single raw array with shape (376 x 277s * 600Hz) (taken from this [tutorial](https://mne.tools/stable/auto_tutorials/raw/10_raw_overview.html#sphx-glr-auto-tutorials-raw-10-raw-overview-py)). I did not have time to test how long it takes for `bandwidth=4` but probably more than 20 minutes on an M1 MacBook Pro.

### Describe your proposed implementation

I am not very familiar with multitapers. However, it should be really easy to speed up the implementation, by simply calculating the multitapers on epochs of the raw data and averaging afterwards. 

I discussed this at the CuttingMEEG conference with @britta-wstnr and I think she agreed.

### Describe possible alternatives

[multitaper_too_slow.pdf](https://github.com/mne-tools/mne-python/files/10268084/multitaper_too_slow.pdf)

britta-wstnr, 2022-12-20:
Hi @moritz-gerster, nice to see you on Github! :)
As a little clarification, I meant that it would be possible to _approximate_ the power in the continuous signal by cutting the signal into smaller segments, and then e.g. running multitaper on the resulting Epochs. 
I have a faint memory of a discussion about this or a similar issue within the last two years or so, but did not remember it off the top of my head as I never work with continuous data myself.

moritz-gerster, 2022-12-20:
> I meant that it would be possible to _approximate_ the power in the continuous signal by cutting the signal into smaller segments e.g. using `make_fixed_length_epochs`, and then running multitaper on the resulting Epochs

Right @britta-wstnr, that's exactly what I do in the attached jupyter notebook. The approximation looks very similar to the continuous input but its 60 times faster. I was wondering whether mne shouldn't do this by default.

larsoner, 2022-12-20:
> As a little clarification, I meant that it would be possible to approximate the power in the continuous signal by cutting the signal into smaller segments e.g. using make_fixed_length_epochs, and then running multitaper on the resulting Epochs.

To me this sounds like a combined Welch + multitaper PSD, right? One way to accomplish this would be to still use the `n_per_seg` and `n_overlap` option in `'multitaper'` mode to say how to chop up the `Raw` data before computing the multitaper PSD. This is probably what most people would want anyway from `mode='multitaper'`, and if they want the old behavior they can always set `n_per_seg=len(raw.times)` and live with a computation that takes forever. I think we could consider this a bugfix, and I doubt many users would be affected since the computation is so slow it's probably not widely used (or used at all) in its current form. WDYT @drammock ?

drammock, 2022-12-20:
before the addition of the Spectrum class, we didn't support multitaper on Raw at all, so I agree that it's not likely to be a widespread use case yet.  Using the `n_per_seg` and `n_overlap` seems like a good solution.  Is this worth blocking the 1.3 release?

larsoner, 2022-12-20:
No, we can backport as a bug fix I think

britta-wstnr, 2022-12-20:
Hehe, sorry @moritz-gerster, I admit I did not look at your Notebook, as I am technically on vacation ;-) sorry about that! Looks like there is a good idea how to move on this - @moritz-gerster would you be up for giving this a try?

moritz-gerster, 2022-12-21:
> would you be up for giving this a try?

I just had a look because I thought it could be simple. However, I understand both the functions [psd_array_multitaper](https://github.com/mne-tools/mne-python/blob/654248a3ea9d5b106eaba51b9a74b93a531b3746/mne/time_frequency/multitaper.py#L346) and [psd_array_welch](https://github.com/mne-tools/mne-python/blob/78e099d9c117c01a0545697a6bbb26521659556e/mne/time_frequency/psd.py#L97) very badly. I'm sorry, I don't feel fit for it :/ 

I hope you find someone more competent. Thanks again for the exciting lecture and enjoy your vacation @britta-wstnr!

closed 2026-11-11 by larsoner