mne-tools/mne-python #11386: "stable" docs show docs for MNE 1.2, not the latest 1.3

hoechenberger, 2022-12-24:
### Proposed documentation enhancement

https://mne.tools/stable/index.html

[image]

larsoner, 2022-12-24:
FYI the release is still in progress -- I just pushed a commit to `maint/1.3` with `git commit --allow-empty -m 'CI: Deploy [circle deploy] [skip azp] [skip actions] [skip cirrus]'` but then immediately cancelled the circle run because I realized we need to do this after https://github.com/mne-tools/mne-python/pull/11387 is merged and backported.

drammock, 2022-12-25:
closed by f2f7ddea1

closed 2022-12-25 by drammock