mne-tools/mne-python #11395: coregistration GUI does not exit cleanly

jasmainak, 2022-12-30:
### Description of the problem

If I open the coregistration GUI and it crashes for some reason (e.g., missing subjects_dir), a dangling welcome banner is left which becomes distracting the next time the GUI is opened. One can exit ipython and restart the script, but this might require redoing lengthy computation.

### Steps to reproduce

```Python
# $ unset SUBJECTS_DIR
# then the following
#
from mne.gui import coregistration
coregistration()
```

### Expected results

The welcome banner closes

### Actual results

[image]

### Additional information

n [2]: mne.sys_info()
Platform: Linux-3.10.0-1160.76.1.el7.x86_64-x86_64-with-glibc2.10
Python: 3.8.6 | packaged by conda-forge | (default, Oct 7 2020, 19:08:05) [GCC 7.5.0]
Executable: /autofs/space/meghnn_001/users/mjas/anaconda3/envs/mne/bin/python3.8
CPU: x86_64: 64 cores
Memory: 125.4 GB

Exception ignored on calling ctypes callback function: <function _ThreadpoolInfo._find_modules_with_dl_iterate_phdr..match_module_callback at 0x7f8f64346940>
Traceback (most recent call last):
File "/autofs/space/meghnn_001/users/mjas/anaconda3/envs/mne/lib/python3.8/site-packages/threadpoolctl.py", line 400, in match_module_callback
self._make_module_from_path(filepath)
File "/autofs/space/meghnn_001/users/mjas/anaconda3/envs/mne/lib/python3.8/site-packages/threadpoolctl.py", line 515, in _make_module_from_path
module = module_class(filepath, prefix, user_api, internal_api)
File "/autofs/space/meghnn_001/users/mjas/anaconda3/envs/mne/lib/python3.8/site-packages/threadpoolctl.py", line 606, in init
self.version = self.get_version()
File "/autofs/space/meghnn_001/users/mjas/anaconda3/envs/mne/lib/python3.8/site-packages/threadpoolctl.py", line 646, in get_version
config = get_config().split()
AttributeError: 'NoneType' object has no attribute 'split'
mne: 1.4.dev0
numpy: 1.23.0 {unknown linalg bindings}
scipy: 1.5.3
matplotlib: 3.3.3 {backend=Qt5Agg}

sklearn: 0.23.2
numba: Not found
nibabel: 3.2.1
nilearn: 0.7.0
dipy: 1.3.0
openmeeg: Not found
cupy: Not found
pandas: 1.1.5
pyvista: 0.36.1 {OpenGL 4.5.0 NVIDIA 455.45.01 via Quadro P5000/PCIe/SSE2}
pyvistaqt: 0.9.0
ipyvtklink: Not found
vtk: 9.0.1
qtpy: 2.0.1 {PyQt5=5.12.9}
ipympl: Not found
/autofs/space/meghnn_001/users/mjas/anaconda3/envs/mne/lib/python3.8/site-packages/pyqtgraph/colors/palette.py:1: RuntimeWarning: PyQtGraph supports Qt version >= 5.15, but 5.12.9 detected.
from ..Qt import QtGui
pyqtgraph: 0.13.1
pooch: v1.5.2

mne_bids: Not found
mne_nirs: Not found
mne_features: Not found
mne_qt_browser: 0.4.0
mne_connectivity: Not found
mne_icalabel: Not found

drammock, 2022-12-30:
duplicate of #11394

drammock, 2022-12-30:
FYI for next time @jasmainak:

[image]

closed 2022-12-30 by drammock

jasmainak, 2022-12-31:
Aware of that but you need to be admin or have triaging rights on the repo to do that :)

drammock, 2022-12-31:
> Aware of that but you need to be admin or have triaging rights on the repo to do that :)

Ahh, oops, I thought you did.