mne-tools/mne-python #11396: Missing API documentation for sample dataset's data_path() (and possibly others)

hoechenberger, 2022-12-30:
### Proposed documentation enhancement

To figure out what the parameters of `datasets.sample.data_path()` do, I had to dig into the source code. This is … not great, especially since we constantly ask users to provide MWE's based on `sample`.

[image]

drammock, 2022-12-30:
is this the old issue of "vscode doesn't render doccer-based docstrings"? Here's what I see in ipython, no source-code-digging required:

[image]

hoechenberger, 2023-01-01:
It seems to be caused by this post-hoc injection of the docstring into the function:
https://github.com/mne-tools/mne-python/blob/e5086fcb20e891ef1c4ddc2ce2ae16047b496c5f/mne/datasets/sample/sample.py#L20-L21

And you're right, seems to be limited to Pylance.

larsoner, 2023-01-02:
Closing as wontfix dup of #8863 #8218

closed 2023-01-02 by larsoner