mne-tools/mne-python #11400: spectral connectivity for source estimate without epochs

neural-decoder, 2023-01-02:
### Describe the new feature or enhancement

A simple way to calculate a spectral connectivity for source estimates from non-epoched raw data.

While running 

```
label_ts = mne.extract_label_time_course(sources, labels, src, mode='mean_flip',
                                         return_generator=True)

fmin = 8.
fmax = 13.
sfreq = 256
con = mne_connectivity.spectral_connectivity_epochs(
    label_ts, method='coh', sfreq=sfreq, fmin=fmin,
    fmax=fmax, mt_adaptive=True)
```

I'm getting an error 

```
File "/home/white/APPS/anaconda3/envs/mne/lib/python3.10/site-packages/mne_connectivity/spectral/epochs.py", line 691, in _get_and_verify_data_sizes
    this_n_signals, this_n_times = this_data.shape
ValueError: not enough values to unpack (expected 2, got 1)
```

This error probably means that sources were calculated from non-epoched data (i.e. by mne.minimum_norm.apply_inverse_raw()). 

What would be the simplest solution to this problem?

### Describe your proposed implementation

-

### Describe possible alternatives

-

larsoner, 2023-01-02:
Typically data are chunked into epochs even if they are not event-related, e.g. https://github.com/mne-tools/mne-connectivity/blob/main/examples/mne_inverse_envelope_correlation.py#L89-L92. But this is a question about usage, please check https://mne.discourse.group/ to see if there are answers there, and if not, post a new one!

closed 2023-01-02 by larsoner

neural-decoder, 2023-01-03:
Yeah I put this issue on discourse group, but still there is no explanation why this happens. Moreover there is no clear description in documentation which function to use for non-epoched data, if any.