mne-tools/mne-python #11414: mne plot freezes when pressing add description

Dannapnt1, 2023-01-10:
### Description of the problem

I am running MNE through the Spyder IDE, the MNE version I am using is 1.3.0. I am running this on a MACOS(Monterey).  

Originally I ran this script as indicated in the tutorial- as a previous version. 
fig = raw.plot()
fig.canvas.key_press_event('a')

Here it opens the MNE plot window for creating annotation for bad artifact data, but when I press add description or select visible the application freezes. Also, a left mouse click with a drag does not mark the selection in the data. 

Then I saw there was a newer version so I ran this: 
fig = raw.plot()
fig.fake_keypress('a') 

When I run this nothing in the plot window works I cannot press any of the channels and when I press the add description it again freezes the window. Also, it shows one of the channels in gray as if it was pressed but doesn't allow me to press it again to unselect it. 

### Steps to reproduce

```Python
fig = raw.plot()
fig.canvas.key_press_event('a')

fig = raw.plot()
fig.fake_keypress('a')
```

### Expected results

I expect to be able to mark the bad segments in my data and add the BAD description to create an annotation of the artifacts in my data. 

### Actual results

The screen freezes and doesn't allow me to continue, I must close the window to be able to continue. 

### Additional information

Platform:         macOS-10.16-x86_64-i386-64bit
Python:           3.8.15 | packaged by conda-forge | (default, Nov 22 2022, 08:53:40)  [Clang 14.0.6 ]
Executable:       /Users/dannapinto1/opt/anaconda3/envs/mne/bin/python
CPU:              i386: 4 cores
Memory:           8.0 GB

mne:              1.3.0
numpy:            1.23.5 {OpenBLAS 0.3.21 with 4 threads}
scipy:            1.10.0
matplotlib:       3.6.2 {backend=module://matplotlib_inline.backend_inline}

sklearn:          1.2.0
numba:            0.56.4
nibabel:          4.0.2
nilearn:          0.10.0
dipy:             1.5.0
openmeeg:         2.5.5
cupy:             Not found
pandas:           1.5.2
pyvista:

larsoner, 2023-01-10:
It looks like you are using the `inline` backend for matplotlib. The window that opens is a Qt window, though, right? You might need to do `%matplotlib qt` before making the plot to get the event handler running to make interaction work

Dannapnt1, 2023-01-11:
Thank you so much for the quick reply. I tried this and the problem persists. Also, I set my preferences for graphics to QT5 and this still doesn't work.

larsoner, 2023-01-11:
Can you try also
```
import matplotlib
matplotlib.use('QtAgg', force=True)
import matplotlib.pyplot as plt
plt.ion()
plt.figure()
```
and see if this pops up an interactive figure? And if it does, if the MNE figure is now interactive as well?

It would be good to try this in a plain `python` terminal first, then try it in your Jupyter notebook

Dannapnt1, 2023-01-15:
By plain python terminal do you just mean from my macbook Terminal? I am usually running the scripts on Spyder not the Jupyter notebook. When I run what you wrote on spyder a window does pop up, and it seems like an interactive figure, but it has no data so I can't really do much with it. When I do the MNE figure it also pops up as an interactive figure, however when I try to interact with it in certain ways, the window freezes. I can click on channel to choose them, and I can set a time starting point but it will not let me make an annotation.

mmagnuski, 2023-02-04:
@Dannapnt1 I suspect that when you push "a" key to add annotation the annotation window opens up in the background. This is a window where you can define and select annotations. It might require some action but since it opens in the background it is not obvious and it looks as if the main figure froze. I think I encountered this at least once, but I didn't check on the current mne if this is the case.

Dannapnt1, 2023-02-07:
@mmagnuski This could definitely be the case because it does seem to open the window to insert an annotation but then the window seems to disappear and the plot window freezes.

larsoner, 2023-10-01:
Closing since I think no changes are needed at our end but let's reopen if I'm mistaken

closed 2023-10-01 by larsoner