mne-tools/mne-python #11426: Convert CSD channel type into EEG after LAP rereferencing: unknown unit (117 (FIFF_UNIT_V_M2))

DominiqueMakowski, 2023-01-21:
### Description of the problem

I am exploring ERP analysis after LAP referencing. As mentioned here, the current way to do that in MNE is to apply CSD projection (https://github.com/mne-tools/mne-python/issues/9659), but this transforms the channel type into "csd", creating some issues when using then the normal ERP analysis pipeline (as most function are incompatible with csd). 

I thought that the most straightforward option, when using CSD for re-referencing purposes, would it be to then re-convert the channel type back into EEG.
Unfortunately, it throws an error regarding the unit, which I'm not sure how to address:  

### Steps to reproduce

```Python
raw = nk.mne_data("filt-0-40_raw")
raw= mne.preprocessing.compute_current_source_density(raw)

# Reconvert CSD type to EEG
channels = np.array(raw.ch_names)[mne.pick_types(raw.info, csd=True)]
raw.set_channel_types(dict(zip(channels, ["eeg"] * len(channels))))
```

### Expected results

No error

### Actual results

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Input In [2], in <cell line: 6>()
      4 # Reconvert CSD type to EEG
      5 channels = np.array(raw.ch_names)[mne.pick_types(raw.info, csd=True)]
----> 6 raw.set_channel_types(dict(zip(channels, ["eeg"] * len(channels))))

File <decorator-gen-42>:12, in set_channel_types(self, mapping, verbose)

File ~\AppData\Roaming\Python\Python39\site-packages\mne\channels\channels.py:359, in SetChannelsMixin.set_channel_types(self, mapping, verbose)
    [357](file:///c%3A/Users/domma/AppData/Roaming/Python/Python39/site-packages/mne/channels/channels.py?line=356) unit_new = _human2unit[ch_type]
    [358](file:///c%3A/Users/domma/AppData/Roaming/Python/Python39/site-packages/mne/channels/channels.py?line=357) if unit_old not in _unit2human:
--> [359](file:///c%3A/Users/domma/AppData/Roaming/Python/Python39/site-packages/mne/channels/channels.py?line=358)     raise ValueError("Channel '%s' has unknown unit (%s). Please "
    [360](file:///c%3A/Users/domma/AppData/Roaming/Python/Python39/site-packages/mne/channels/channels.py?line=359)                      "fix the measurement info of your data."
    [361](file:///c%3A/Users/domma/AppData/Roaming/Python/Python39/site-packages/mne/channels/channels.py?line=360)                      % (ch_name, unit_old))
    [362](file:///c%3A/Users/domma/AppData/Roaming/Python/Python39/site-packages/mne/channels/channels.py?line=361) if unit_old != _human2unit[ch_type]:
    [363](file:///c%3A/Users/domma/AppData/Roaming/Python/Python39/site-packages/mne/channels/channels.py?line=362)     this_change = (_unit2human[unit_old], _unit2human[unit_new])

ValueError: Channel 'EEG 001' has unknown unit (117 (FIFF_UNIT_V_M2)). Please fix the measurement info of your data.
```

### Additional information

```
Platform:         Windows-10-10.0.22621-SP0
Python:           3.9.7 (tags/v3.9.7:1016ef3, Aug 30 2021, 20:19:38) [MSC v.1929 64 bit (AMD64)]
Executable:       c:\Program Files\Python39\python.exe
CPU:              Intel64 Family 6 Model 141 Stepping 1, GenuineIntel: 16 cores
Memory:           31.2 GB

mne:              1.2.1
numpy:            1.20.0 {OpenBLAS 0.3.13.dev with 16 threads}
scipy:            1.7.1
matplotlib:       3.4.3 {backend=module://matplotlib_inline.backend_inline}

sklearn:          1.0.2
numba:            0.55.1
nibabel:          3.2.1
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           1.4.3
pyvista:          0.32.1 {OpenGL 4.5.0 NVIDIA 517.51 via NVIDIA GeForce RTX 3080 Laptop GPU/PCIe/SSE2}
pyvistaqt:        0.5.0
ipyvtklink:       0.2.1
vtk:              9.0.3
qtpy:             1.11.2 {PySide2=5.15.2}
ipympl:           0.8.1
```

larsoner, 2023-01-23:
> I thought that the most straightforward option, when using CSD for re-referencing purposes, would it be to then re-convert the channel type back into EEG.

The error is correct, you *shouldn't* be able to do this because the units are not `V` anymore, but current density or something IIRC.

To me the correct fix is to update whatever plotting functions are broken. Usually it's not too much work, mostly just adding `'csd'` to some lists / dicts somewhere. Most of the work actually comes from generating the raw/epochs/evoked/whatever classes needed to hit all the possible plotting combinations in tests :) But that's exactly how we'd want to make sure these plots all work anyway.

berndf, 2024-01-20:
Dear Eric, exactly relating to your suggestion: I just created a patch to mne/viz/evoked.py to allow interactive topomaps as well as GFP with CSD channels.  Thanks and regards, Bernd
[allow_CSD_Topomap.patch.gz](https://github.com/mne-tools/mne-python/files/13997563/allow_CSD_Topomap.patch.gz)

larsoner, 2024-07-12:
Closing but feel free to open more targeted issues for specific viz functions that need updating!

closed 2024-07-12 by larsoner