mne-tools/mne-python #11429: Add `count_events()` function

cbrnr, 2023-01-24:
### Describe the new feature or enhancement

Does anyone else feel the need for a `count_events()` function? I need it all the time to check if the number of specific event types are what I expect.

### Describe your proposed implementation

It's easy enough to implement, for example like this:

```python
from collections import defaultdict
from numpy import bincount

def count_events(events):
    counts = bincount(events[:, 2])
    return defaultdict(lambda: 0, {i: count for i, count in enumerate(counts) if count > 0})
```

This produces a nice overview in a defaultdict like this:

```
defaultdict(<function __main__.count_events.<locals>.<lambda>()>,
            {111: 60, 121: 60, 201: 111, 202: 11, 211: 60, 221: 60})
```

The reason for a defaultdict is that users can access non-existing events and will get 0.

### Describe possible alternatives

Like I said, it is not hard to write this every time it is needed. But my point is that I think it is needed frequently enough to warrant a dedicated function in MNE.

drammock, 2023-01-24:
seems useful. This might be a bit simpler? uses builtin `list.count()` method instead of bincount.

```
unique_events = np.unique(events[:, 2])
counts = {item: events[:, 2].tolist().count(item) for item in unique_events}
return defaultdict(lambda: 0, counts)
```

Other option is `pd.value_counts(events[:, 2]).to_dict()` but then you need pandas installed, which is a soft dependency at present.

cbrnr, 2023-01-24:
Nah, I'd rather not use pandas just for that, let's stick with NumPy. I don't have a preference for either implementation, they seem about equally simple.

cbrnr, 2023-01-24:
The other question is where would you put it? Two options are `mne.count_events()` and `mne.event.count_events()`, what would everyone prefer?

hoechenberger, 2023-01-24:
> The reason for a defaultdict is that users can access non-existing events and will get 0.

I cannot say that I like this approach. I'd expect it to throw an `IndexError` instead, based on my daily work in the "scientific Python ecosystem".

hoechenberger, 2023-01-24:
> `mne.count_events()` and `mne.event.count_events()`,

I'd actually decide between `mne.count_events()` and `mne.events.count()`, no?

drammock, 2023-01-24:
> > The reason for a defaultdict is that users can access non-existing events and will get 0.
> 
> I cannot say that I like this approach. I'd expect it to throw an `IndexError` instead, based on my daily work in the "scientific Python ecosystem".

That was my initial reaction too (though it would be `KeyError` for a dict). However, imagine a scenario where your experiment has an event `77` but in this particular script you've done something to subset the data and none of the event `77`s are left.  Would it be more useful to return a `0` or to error out?  To me it's not really clear which is better.

cbrnr, 2023-01-24:
@drammock that's exactly why I used a defaultdict. I frequently do this kind of substring, or sometimes a particular event type is not present for a subject, e.g. to indicate an incorrect response - but this subject had only correct responses. A normal dict would require a try/except or `.get(type, 0)`, which is cumbersome.

hoechenberger, 2023-01-24:
> , imagine a scenario where your experiment has an event `77` but in this particular script you've done something to subset the data and none of the event `77`s are left. Would it be more useful to return a `0` or to error out? To me it's not really clear which is better.

I'd suggest to control this behavior via a function parameter, then, and default to **not** returning a `defaultdict`

hoechenberger, 2023-01-24:
I am also starting to wonder if we shouldn't consider adding an `Events` class, which is acts like a NumPy array but has a `.count()` method.

drammock, 2023-01-24:
I'm open to the idea of an `Events` class that is a thin wrapper of `np.ndarray`. But even if that is the route we go, I think @cbrnr's idea of counting everything at once is a good one so I'd lobby for an API like `Events.value_counts()` rather than `Events.count(77)`

drammock, 2023-01-24:
> I'd suggest to control this behavior via a function parameter, then, and default to **not** returning a `defaultdict`

you mean something like `on_missing='raise'`?  And if `on_missing='ignore'` or `'warn'` then just do a try-except and return 0 in the except (possibly accompanied by a warning)?

cbrnr, 2023-01-25:
An `Events` class is a good idea in principle, but then we should discuss merging it with our `Annotations`. This historic distinction is something that people have a hard time understanding, and we could create a class which transparently handles both events and annotations.

But this will be a larger change and should be discussed separately. Right now, the question is whether or not we should add the proposed function for existing events or not.

Regarding the behavior of returning 0 by default, I think that's the right thing to do. If I ask "how many events of type 62 are there?", I expect to get 0 if there are none. Even if you would like to control that behavior with a parameter, this should be the default, and raising an error should be optional. I think there's nothing wrong or noteworthy if you query an event that occurs zero times, just return `0`. I have not yet seen an example where this might be problematic other than users might expect it from other Python functions.

hoechenberger, 2023-01-25:
> If I ask "how many events of type 62 are there?", I expect to get 0 if there are none.

I fully agree. But I don't agree that this question should be phrased using a dictionary. If it's a function call, I, alright with it. If it's a dictionary that suddenly spits out results even if the keys I'm requesting are missing, I don't like it -- it deviates greatly from what I know, and introduced an inconsistency.

hoechenberger, 2023-01-25:
> > I'd suggest to control this behavior via a function parameter, then, and default to **not** returning a `defaultdict`
> 
> 
> 
> you mean something like `on_missing='raise'`?  And if `on_missing='ignore'` or `'warn'` then just do a try-except and return 0 in the except (possibly accompanied by a warning)?

No, I only meant a switch to either return a normal dict or the proposed defaultdict. But now that I come to think about it, I don't like my suggestion anymore.

Why not simply add an optional parameter to pass the requested event codes and return a dict where each code is a key? Then I would be okay with simply putting the value 0 in places where the key doesn't exist in the events array, because things are more explicit then.

cbrnr, 2023-01-25:
I often do this in my code (assuming the hypothetical function `count_events()` exists):

```python
counts = mne.count_events(events)
n_responses = counts[1] + counts[2] + counts[3]
```

Events 1, 2, and 3 are three different possible responses. Now depending on the subjects, some data might have zero events of type 3. With the proposed solution, I could use my code above. It is simple and elegant, and a defaultdict is also Pythonic and nothing magical/unexpected.

How would this code look like using your proposed solution? I do not understand your arguments other than claiming it is inconsistent (with the rest of MNE?).

hoechenberger, 2023-01-25:
```
counts = count_events(events, ids=(1, 2, 3))
n_resp = sum(list(counts.values()))
```
and if we returned a DataFrame or Series instead, it could be as simple as `n_resp = counts.sum()`

The point about (in)consistency to me is not a minor one, but actually a huge deal. MNE is complicated to use already, and I believe making things less (instead of more) consistent is not something we should do

cbrnr, 2023-01-25:
Looks good, then let's do it like that!

Last open question: where do we put the function? The consistent location would probably be `mne.count_events()`.

hoechenberger, 2023-01-25:
I think the public namespace should be `mne.count_events()`, yes

But I'm not sure yet if we agreed upon which data type to return?

Because if it's cool to return e.g. a Series we could remove the `ids` param I proposed and suggest something like this:

```python
counts = count_events(events)
n_resp = counts[counts.isin([1, 2, 3])].sum()
```
Or we could have a DataFrame with columns `id` and `count`:

```python
n_resp = counts.loc[counts['id'].isin([1, 2, 3]), 'count']).sum()
```

hoechenberger, 2023-01-25:
> Because if it's cool to return e.g. a Series we could remove the `ids` param I proposed and suggest something like this:
> 
> ```python
> counts = count_events(events)
> n_resp = counts[counts.index.isin([1, 2, 3])].sum()
> ```
> 
> Or we could have a DataFrame with columns `id` and `count`:
> 
> ```python
> n_resp = counts.loc[counts['id'].isin([1, 2, 3]), 'count']).sum()
> ```

But obviously this looks kind of scary to a beginner, right?

So maybe:

- keep the `ids` parameter to filter down event IDs
- but still return a Series so users can simply call `.sum()` to get the count

hoechenberger, 2023-01-25:
(I edited my above post to add an example for returning a Series)

cbrnr, 2023-01-25:
Let's not return any pandas objects, I would like to avoid using optional dependencies. Sure everything can always be made even more convenient, but your example is simple enough I think.

hoechenberger, 2023-01-25:
Ok, a dict it is!

hoechenberger, 2023-01-25:
Proposal:

```python
def count_events(
    events: ndarray,
    ids: Iterable[int] | None = None,
    on_missing: 'raise' | 'warn' | 'ignore' = 'raise
) -> dict[int, int]:
    ...
    if ids and on_missing != 'ignore' and not all [i in events[:,-1] for i in ids]:
        raise IndexError()  # or ValueError? 
    ...
```

cbrnr, 2023-01-25:
IMO we don't need this parameter. If `ids is None` (the default), we return a normal dict with event types as keys and their frequencies as values. If, however, `ids` is a list of `int`, the dict will consist only of keys corresponding to the values in `ids` – and if a specific event type does not occur in the data, its value in the dict will be zero.

hoechenberger, 2023-01-25:
Ok let's do this!

cbrnr, 2023-01-25:
```python
from numpy import bincount

def count_events(events, ids=None):
    counts = bincount(events[:, 2])
    counts = {i: count for i, count in enumerate(counts) if count > 0}
    if ids is not None:
        return {id: counts.get(id, 0) for id in ids}
    return counts

events = np.array([[0, 0, 1], [0, 0, 1], [0, 0, 5]])
```

Behaves as follows:

```
>>> count_events(events)
{1: 2, 5: 1}
>>> count_events(events, ids=[1, 5])
{1: 2, 5: 1}
>>> count_events(events, ids=[1, 11])
{1: 2, 11: 0}
```

I like it. Is everyone OK with putting it into `mne.count_events()`?

cbrnr, 2023-01-25:
@hoechenberger your previous example

```python
n_resp = sum(list(counts.values()))
```

can be simplified to

```python
n_resp = sum(counts.values())
```

- very nice!

PR #11430 (2023-01-25): Add count_events() function
Fixes #11429.

I've included some examples that can be used as doctests, but I'm not sure if we're doing doctests or if "proper" tests would be preferable.
closed 2023-01-25 by drammock