mne-tools/mne-python #11439: Should `mne.Epochs.plot()` plot events and IDs by default?

cbrnr, 2023-01-31:
An `mne.Epochs` instance has associated `events` and `event_id` attributes. These correspond to the events and IDs set during epoch creation.

When plotting existing epochs with `epochs.plot()`, no events are drawn. If events should be included, they need to be passed explicitly, e.g. `epochs.plot(events=epochs.events, event_id=epochs.event_id)`. To me, this is rather cumbersome, since I expect the epoch labels to be included by default. However, maybe there is a good reason for this behavior – if not, I'd be happy to contribute a PR to include events by default.

hoechenberger, 2023-01-31:
I'd love to see this change unless there's, like you said, 

> good reason for this behavior

agramfort, 2023-01-31:
good idea

> Message ID: ***@***.***>
>

cbrnr, 2023-02-01:
Great! I have a related follow-up question (which perhaps should be addressed separately, but I'd like to ask it here): Is it possible to plot multiple events within epochs? That is, what if I epoch relative to a stimulus event (e.g. -0.2 to 0.8s), and typically there are response events say after about 0.6s. These response events are not used to create the epochs, but it would still be useful to see them in the plot. Is this possible? I tried, but it seems like the only events that are plotted are the ones used to define the epochs.

[This figure](https://socialsci.libretexts.org/Bookshelves/Psychology/Book%3A_Applied_Event-Related_Potential_Data_Analysis_(Luck)/02%3A_Processing_the_Data_from_One_Participant_in_the_ERP_CORE_N400_Experiment/2.07%3A_Exercise-_Epoching_and_Baseline_Correction) (Figure 2.4, the third one from the top) shows how EEGLAB handles this case. Epoch 2 contains two events, `B4(221)` (the trigger used to create the epoch) and `201` (the response).

hoechenberger, 2023-02-01:
I'd be concerned about overlapping labels then, no? If you add all events including their names (as labels), it will become difficult to avoid label overlaps (unless we add some sort of vertical stacking). No?

hoechenberger, 2023-02-01:
Yes the event names stored as keys event_id

If there are many events, we'll have overlapping texts, no?

cbrnr, 2023-02-01:
Yes, but overlapping text might also occur with only a few events but long event_id values. To me, that's another relatively independent issue.

drammock, 2023-02-02:
Sorry to chime in late:

> Is it possible to plot multiple events within epochs? That is, what if I epoch relative to a stimulus event (e.g. -0.2 to 0.8s), and typically there are response events say after about 0.6s. These [...]

This already works. MWE:

```py
import mne

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = (sample_data_folder / 'MEG' / 'sample' /
                        'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file, verbose=False, preload=False)

events = mne.find_events(raw, stim_channel='STI 014')
event_dict = {'auditory/right': 2, 'buttonpress': 32}
epochs = mne.Epochs(raw, events, event_id=event_dict, preload=True, proj=False)
epochs['auditory/right'][4:10].plot(events=events)
```

I've cherry-picked the epoch range `4:10` to show what you're asking for: all epochs in the range have event `2` (auditory/right stimulus) but the first and last epoch shown also have event `32` (buttonpress)

cbrnr, 2023-02-03:
Thanks @drammock – but does this mean that I need to create epochs relative to all events I want to display? In your example, I'm not interested in epochs relative to the buttonpress event, but I still need to create it? That's not very intuitive IMO.

cbrnr, 2023-02-03:
Also, I can't even seem to make it work with data available [here](https://ucdavis.app.box.com/v/ERP-Analysis-Book/file/784322947354) (`6_N400_preprocessed.set`):

```python
import mne

raw = mne.io.read_raw_eeglab("6_N400_preprocessed.set", preload=True)
fs = raw.info["sfreq"]
raw.filter(0.2, None)
raw.filter(None, 20)

events, _ = mne.events_from_annotations(raw, event_id=lambda x: int(x))

# merge events from lists 1 and 2 (these are irrelevant)
events = mne.merge_events(events, [111, 112], 111)  # prime related
events = mne.merge_events(events, [121, 122], 121)  # prime unrelated
events = mne.merge_events(events, [211, 212], 211)  # target related
events = mne.merge_events(events, [221, 222], 221)  # target unrelated
events = events[2:]  # discard first two events

event_id = {
    "prime/related": 111,
    "prime/unrelated": 121,
    "target/related": 211,
    "target/unrelated": 221,
    "correct": 201,
    "incorrect": 202
}

epochs = mne.Epochs(raw, events, event_id, tmin=-0.2, tmax=1.8)
epochs["target"].plot(events=events, event_id=event_id)
```

I'd expect to see events of type `201` (`correct`) or `202` (`incorrect`) at the end of each epoch, but there is nothing.

cbrnr, 2023-02-13:
I moved my question regarding plotting extra event types to our forum:

https://mne.discourse.group/t/showing-extra-events-in-epochs-plot/6334

PR #11445 (2023-06-23): Plot epochs.events by default
Fixes #11439, which means that `epochs.events` are now automatically plotted (together with `epochs.event_id`).

I suggest the following changes to the `events` and `event_id` parameters: First, automatic plotting should be enabled when `events=True`. Second, this should be the new default. That way, users can still choose to plot no events with `events=None` (the previous default).
closed 2023-06-23 by drammock