mne-tools/mne-python #11447: permutation test docstrings should link to spatial_src_adjacency

drammock, 2023-02-03:
### Proposed documentation enhancement

currently the docstring (under the `adjacency` parameter) includes a link to `find_ch_adjacency` (for sensor-space data) but not to `spatial_src_adjacency` (for source-space data). It should have both.

hoechenberger, 2023-02-06:
There's actually numerous adjacency-calculating functions that we probably should mention:

### spatial
- `spatial_src_adjacency`
- `spatial_tris_adjacency`
- `spatial_dist_adjacency`
- `spatial_inter_hemi_adjacency`

### spatio-temporal
- `spatio_temporal_src_adjacency`
- `spatio_temporal_tris_adjacency`
- `spatio_temporal_dist_adjacency`

Interestingly, none of them is prefixed with `find_`, contrary to `find_ch_adjacency`.

sawradip, 2023-02-06:
I would like to perform this task.  Can you give me some directions?

drammock, 2023-02-07:
I don't think the `*_tris_*` and `*_dist_*` are worth mentioning actually --- the `*_dist_*` functionality is already available in the `*_src_*` functions (they simply default to `dist=None`), and in the context of permutation testing, users won't be working with `tris`. But adding `spatio_temporal_src_adjacency` makes sense, and `spatial_inter_hemi_adjacency` makes sense for completeness, I guess.

drammock, 2023-02-07:
> I would like to perform this task. Can you give me some directions?

Find the docstring of `permutation_*_test` and `spatio_temporal_*_test` functions, and add to the `adjacency` parameter description cross-references to:

- spatial_src_adjacency
- spatio_temporal_src_adjacency
- spatial_inter_hemi_adjacency

(...with appropriate context for when each might be used)

sawradip, 2023-02-16:
I will be starting working on this, as #11469 is almost finished. I will have to go through the documentations a little, in order to understand the appropriate contexts for those functionalities. 

> (...with appropriate context for when each might be used)

sawradip, 2023-02-16:
Preprocessing:

https://github.com/mne-tools/mne-python/blob/6608b720b214529732542cb2aac11b401770eddf/tutorials/preprocessing/10_preprocessing_overview.py#L39

dominikwelke, 2023-11-15:
working on this in #12212

PR #12212 (2023-11-15): DOC: permutation test docstrings should link to adjacency reader
closes #11447 .
closed 2023-11-15 by sappelhoff