mne-tools/mne-python #11451: mne.viz.plot_compare_evokeds should not draw colorbar legend when `legend=False`

mmagnuski, 2023-02-04:
When a colormap is passed to `mne.viz.plot_compare_evokeds` (for example `cbar='viridis'`) it places a colorbar next to the axis (instead of a normal in-axis legend). While the presence of the standard legend can be controlled with `legend=False`, the colorbar legend is always drawn.

#### Example:
```python
import numpy as np
import mne

info = mne.create_info(ch_names=list('ABCDE'), sfreq=50., ch_types='eeg')

conditions = ['cond1', 'cond2']
evokeds = dict()
for condition in conditions:
    data = np.random.random((5, 100)) * 1e-6
    evokeds[condition] = mne.EvokedArray(data, info, tmin=-0.25)
```

`legend=False` controls whether standard legend is drawn:
```python
mne.viz.plot_compare_evokeds(evokeds)
```
[image]

```python
mne.viz.plot_compare_evokeds(evokeds, legend=False)
```
[image]

However, when `cbar` argument is used, the colormap legend is always drawn:
```python
mne.viz.plot_compare_evokeds(evokeds, cmap='viridis', legend=False)
```
[image]

I think `legend=False` should not plot the colorbar.

hoechenberger, 2023-02-06:
I didn't know that feature existed! Why do we even have that? They're both essentially doing the same, no??

mmagnuski, 2023-02-06:
The colorbar feature might be useful for variables with continuous meaning (contrast for example). I think this scenario is assumed when you specify a colormap with `cmap`.

mmagnuski, 2023-02-06:
take a look at this example for instance:
https://mne.tools/stable/auto_examples/stats/sensor_regression.html

hoechenberger, 2023-02-06:
Makes sense!