mne-tools/mne-python #11462: Cannot view PRs in PyCharm (OAuth restrictions)

cbrnr, 2023-02-08:
I'd like to set up PR integration in PyCharm, but this doesn't work because this repo seems to have OAuth restrictions enabled. Would it be possible to approve PyCharm as described [here](https://docs.github.com/en/organizations/managing-oauth-access-to-your-organizations-data/approving-oauth-apps-for-your-organization)? I'm not sure how it works and if the request is already visible.

agramfort, 2023-02-09:
not sure how to do this. It was setup to avoid having a developper blindly
giving access to the org private data.

Typically for me at work now I don't want to give access to these repos via
SSO.

> Message ID: ***@***.***>
>

cbrnr, 2023-02-09:
Yes, the idea is to only allow access that maintainers explicitly grant, because blindly giving access is not a good idea. As to how it's done, it is described in the link in my top post. So if you agree that this app isn't harmful (it should list all permission it requests, probably just a bunch of read-only things) it would be great if you could enable it (so everyone using PyCharm can benefit).

agramfort, 2023-02-09:
can you at least request it and we can approve?

I see only
 
[image]

cbrnr, 2023-02-09:
I tried again – does it show up now?

drammock, 2023-02-09:
request still does not show up :(

cbrnr, 2023-02-09:
Thanks for checking. Then I guess something's wrong either with what I'm doing or with PyCharm, and I will try to find out.

cbrnr, 2023-04-25:
I'm closing this because there is not really anything we can do I think.

closed 2023-04-25 by cbrnr