mne-tools/mne-python #11463: Changing number of visible channels in the butterfly mode

Genuster, 2023-02-09:
### Description of the problem

For `raw.plot(butterfly=True)` neither keyboard shortcut for changing number of visible channels, nor `n_channels` argument have an effect on number of visible channels.

### Steps to reproduce

```Python
raw.plot(butterfly=True)
```

### Expected results

[image]

### Actual results

[image]

### Additional information

Platform:         Windows-10-10.0.20348-SP0
Python:           3.10.9 (tags/v3.10.9:1dd9be6, Dec  6 2022, 20:01:21) [MSC v.1934 64 bit (AMD64)]
Executable:       C:\Users\Gennadiy\Documents\eeg-processing-pipeline\.venv\Scripts\python.exe
CPU:              Intel64 Family 6 Model 167 Stepping 1, GenuineIntel: 16 cores
Memory:           63.9 GB

mne:              1.3.0
numpy:            1.23.5 {OpenBLAS 0.3.20 with 16 threads}
scipy:            1.9.3
matplotlib:       3.6.2 {backend=QtAgg}

sklearn:          1.2.0
numba:            0.56.4
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             12.0.0b3
pandas:           1.5.2
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             2.3.0 {PyQt5=5.15.2}
ipympl:           Not found
pyqtgraph:        0.13.1
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.4.0
mne_connectivity: Not found
mne_icalabel:     Not found

drammock, 2023-02-09:
that is correct.  In butterfly mode all channels of a given type are shown overlaid on each other.  For a quick fix to get the "expected results" figure you could first reduce to just the channel type you want, i.e.:

```py
raw.copy().pick('eeg').plot(butterfly=True)
```

(the `.copy()` is in there so that your original raw object isn't modified).

As far as whether this is a bug:  I would say that it's correct to ignore `n_channels` when `butterfly=True`, so I would vote to keep that behavior as it is.  However, I can see how it would be useful for the **keyboard shortcuts** to still work for showing/hiding the channel *types* (instead of the individual channels, as is done when `butterfly=False`).

cbrnr, 2023-02-09:
👍 for just enabling the keyboard shortcuts.