mne-tools/mne-python #11476: many CI failures related to pkg_resources

drammock, 2023-02-14:
At first I thought it was just nilearn so I opened this: https://github.com/nilearn/nilearn/pull/3512 but it's also happening with pybtex.

cbrnr, 2023-02-14:
How difficult would it be to solve by removing `pkg_resources`? It should be easy for `bcidata`.

drammock, 2023-02-14:
It needs to be done in our dependencies. For the nilearn PR above it's a one-line fix. I've just opened https://bitbucket.org/pybtex-devs/pybtex/issues/169/replace-pkg_resources-with for pybtex. IDK how many other ones we'll hit though once those two are cleared up.

cbrnr, 2023-02-14:
Ah OK! I thought we could also remove it in MNE, because maybe we can work around it (which is definitely possible in `bcidata`, but I don't know about the other occurrence).

drammock, 2023-02-14:
> Ah OK! I thought we could also remove it in MNE

We could catch / ignore the warnings but AFAICT this isn't something we can properly fix within our own codebase.

larsoner, 2023-02-15:
Workaround for us in #11478, thanks @drammock for doing the harder work of opening upstream bug reports

PR #11478 (2023-02-15): MAINT: Deal with pkg_resources usage bugs
Let's see if it works

Closes #11476
closed 2023-02-15 by larsoner