mne-tools/mne-python #11477: VS Code Setup for MNE should be more detailed

Dev-satish, 2023-02-15:
### Proposed documentation enhancement

## Proposed Documentation Enhancement
While going through the MNE setup for VS Code, the documentation felt very vague as it just mentions the link to VS code and says the following line: `you need to tell the VS Code Python extension where to find the respective Python executable`
Which feels this entire process should be divided step by step and a few pictures could be added instead of a line like: `Using Python: /some/directory/mne-python_1.0.0_1/bin/python` because that `some/directory` can be confusing for first time user or people that do not use terminal very frequently. 

Visual representation for example could be more helpful like:
[image]

larsoner, 2023-02-15:
Can we close as a dup of https://github.com/mne-tools/mne-installers/issues/170#issue-1583813508 ?

Dev-satish, 2023-02-15:
> Can we close as a dup of [mne-tools/mne-installers#170 (comment)](https://github.com/mne-tools/mne-installers/issues/170#issue-1583813508) ?

Sure!

closed 2023-02-15 by larsoner