mne-tools/mne-python #11481: Enable mne.concatenate_raws(raws) to handle varying bad channels in raws

moritz-gerster, 2023-02-15:
### Describe the new feature or enhancement

Maximization of available data is very important for many machine learning tasks. Therefore it is better to concatenate three similar recordings instead of using only one. However, what if the bad channels don't match? Currently, there are only two easy options:

1) Mark the bad channel as bad in all recordings. This means throwing away valuable data from that channel (for example in the case of deep brain stimulation LFPs where channel interpolation is difficult). 
2) Ignore the recording with bad channels. This also means throwing away valuable data from the other channels.

### Describe your proposed implementation

I would suggest to enhance `mne.concatenate_raws` to add the option of setting bad channels as `np.nan`. In that case, the maximum of available data can be used since most algorithms handle nan-values well.

For example, 
`mne.concatenate_raws(raws, preload=None, events_list=None, *, on_mismatch='raise', verbose=None)`
with the option `on_mismatch='set_nan'`.  Or to add another kwarg `on_bads_mismatch='set_nan'`.

### Describe possible alternatives

[Example code for enhancement](https://github.com/mne-tools/mne-python/files/10747126/mne_concatenate_channels.pdf)

larsoner, 2023-02-15:
> I would suggest to enhance mne.concatenate_raws to add the option of setting bad channels as np.nan. In that case, the maximum of available data can be used since most algorithms handle nan-values well.

Rather than doing this it might be easier and more general to make a function like `mne.preprocessing.bads_to_nan` (the name is not good but you get the idea) that takes any bad channels and sets the data to NaN, maybe also clearing `info['bads']` (though this would just be one more line for the end user). Then `concatenate_raws` doesn't have to be changed at all.

Actually this could almost trivially be wrapped in a kind of ugly (naming-wise) way into `raw.interpolate_bads(..., method='nan')` or something that just replaces bad channel data with NaN. That's already where one should probably look into dealing with bad channels across recordings/sessions/runs/subjects, so it's maybe an okay place...

drammock, 2023-02-15:
it would be a bit odd if `raw.interpolate_bads(...)` might result in no interpolation at all (but rather just masking the bad channels)...  but the more I think about it the more I like the idea.

But this raises the question of why current/intended usage of `interpolate_bads()` is not a viable option for @moritz-gerster ?  I.e., why not interpolate rather than setting as `nan` (obviously `nan` is faster, but are there other reasons)?

moritz-gerster, 2023-02-16:
In my case, I work on deep brain stimulation leads. I think these leads are very difficult to reasonably interpolate because they are in a tiny inhomogeneous structure (the subthalamic nucleus) very close to the neuronal sources and in addition they have various leads with various impedances (2 ring leads and 6 smaller directional leads). I attach a figure. 

Even if interpolation would work well, it would not improve my Common Spatial Patterns algorithm anyways as feeding in linear dependent data does not really improve the training sample size I guess. 
[image]

moritz-gerster, 2023-02-16:
There is another functionality that `raw.interpolate_bads(...)` does not provide @drammock, @larsoner. It could be the case, that even the number of channels might not agree across recordings. Maybe one or two channels went missing completely. 

At the moment, one would have to create fake channels with nan values that match the recording length and copy the info from the other recordings where the channel is present. This requires tremendous coding. 

Wouldn't it be nice if their is one simple function that compares channels across recordings and
- sets bad channels as nan (only if they are good in some recordings and otherwise keeps them as bad throughout)
- and adds empty channels with matching info if some channels are missing?

I think this could be added to `mne.concatenate_raws()` and it would probably make it much easier to maximize available data even if it is inhomogeneous.

moritz-gerster, 2023-02-16:
I currently do this:
```Python
try:
    # all channels match, including bad channels
    raw_concat = concatenate_raws(raws)
except ValueError:

    # set bad channels as nan and remove from bad channels
    for raw in raws:
        _interpolate_bads_nan(raw)

    # add missing channels as nan
    raws = _add_missing_channels_as_nan(raws)
    
# now concatenate raws
raw_concat = concatenate_raws(raws)

def _interpolate_bads_nan(raw):
    """Set bad channels to np.nan."""
    set_nan = lambda x: np.full(x.shape, np.nan)
    raw.apply_function(set_nan, picks=raw.info["bads"])
    raw.info["bads"] = []

def _add_missing_channels_as_nan(raws):
    # all channels across all raws
    all_chs = [raw.ch_names for raw in raws]  # as list
    all_chs = {el for list in all_chs for el in list}  # as set

    missing_channels = []
    for ch in all_chs:
        for raw in raws:
            if ch not in raw.ch_names:
                missing_channels.append(ch)

    # add missing channels as nan
    for ch in missing_channels:
        for raw in raws:
            if ch not in raw.ch_names:
                # need to add fake channel to raw
                ch_good = False
                while not ch_good:
                    # loop again through raws to get the very same channel
                    # in a recording where it is good
                    for sub_raw in raws:
                        if ch in sub_raw.ch_names:
                            ch_info = sub_raw.copy().pick(ch).info
                            data = np.empty([1, raw.get_data().shape[1]])
                            data[:] = np.nan
                            empty_channel = RawArray(data, ch_info)
                            ch_good = True
                raw.add_channels([empty_channel], force_update_info=True)
    return raws
```

This is pretty ugly. Is there a more elegant way to do this? Don't you think that many other researchers need the same functionality?

larsoner, 2023-02-16:
> see #11106; maybe your use case is enough to convince @larsoner to change his mind.
>
> As an aside, if I'm understanding your code correctly, it looks like when a channel is missing you're using the info for that channel from a different recording. I'm guessing you thought of this already, but...

It's considerations like this -- that there are a lot of underlying variables to consider -- that make me hesitate to add some new simple one-liner to do this stuff. In some sense the cumbersome nature of needing to create new `RawArray` and then `add_channels` and/or `reorder_channels` them helps see what's really necessary to do what you want, and hopefully implicitly see concerns like what @drammock brings up.

However, I am a bit more convinced seeing that there are more use cases like that of @hoechenberger in #11106 in the wild. Perhaps we can add something new and document sufficiently all the caveats.

We handled a somewhat similar problem in #10533 / #10537 to create a `prepare_empty_room` function. Rather than making `concatenate_raws` any smarter, though, I think I'd rather have a dedicated function for coercing recordings with different channels (and/or different bads) to have the same set of (good) channels. After all you might want this *without* concatenating the recordings as well, just to get the same sort of / compatible Evoked instance from multiple participants. Maybe a new `mne.preprocessing.unify_channels(inst_list, add_method='nan', bad_method='interpolate')` or something to start? I don't love the name and am not sure about the necessary arguments yet, but the idea is that you are somehow unifying all channels across instances so that any subsequent concatenation and processing (epoching, evoking, PSD, etc.) afterward give reasonable results.

hoechenberger, 2023-02-16:
I thought we already had a solution for that but I cannot remember the name anymore. I believe @agramfort implemented this...  I did make a reference to it without ever mentioning the name of the function, see:

https://github.com/mne-tools/mne-bids-pipeline/issues/538

moritz-gerster, 2023-02-18:
> As an aside, if I'm understanding your code correctly, it looks like when a channel is missing you're using the `info` for that channel _from a different recording_. I'm guessing you thought of this already, but if that recording is from a different subject the `pos` info will almost surely be wrong. In the case of EEG even the same subject in different sessions could be wrong if you're using real (digitized) sensor locations rather than a template montage (though IDK if that applies to depth electrodes; I would hope that they don't shift around much day-to-day!)

Yes, thank you! It's always the same subject and its intracranial electrodes which hopefully don't move 😊

agramfort, 2023-02-18:
I like the idea of having a function :

mne.preprocessing.unify_bad_channels

there are indeed some places eg the mne bids pipeline where there is indeed
some logic like this.

PR #12027 (2023-09-29): Adding nan method to interpolate channels
Partially fixes #11481 (the other part was fixed in #12014).
closed 2023-09-29 by drammock