mne-tools/mne-python #11482: SSP remove ECG/EOG artifacts: Unable to generate artificial ECG channel

Serendipityzzz, 2023-02-16:
### Description of the problem

Hi there,

I tried to run the tutorial [Repairing artifacts with SSP](https://mne.tools/stable/auto_tutorials/preprocessing/50_artifact_correction_ssp.html) with my own data.

I have problems with using function create_ecg_epochs, compute_proj_ecg.
I tried both EEG and MEG data, and the bug goes like:
"Produces a ValueError: Unable to generate artificial ECG channel"

I want to run ssp before ica, since ica is not enough for me to repair ecg/eog artifacts, 
but I fail to compute the projections with SSP.
Any tips or hints will be appreciated - Thx!

Wentao

### Steps to reproduce

```Python
The first is what I do for EEG

base = os.path.join(os.getcwd(), os.pardir, os.pardir)
path_data = os.path.join(base, "data", "eeg", "demo","005", "20221223")
path_eeg = os.path.join(path_data, "eeg", "image-audio-post.vhdr")
path_montage = os.path.join(path_data, "standard-1020-cap64.locs")
path_output = os.path.join(os.getcwd(), "output")

# Execute preprocess pipeline.
data = preprocess_session(path_eeg, path_montage, path_output)
data = mne.io.read_raw_brainvision(path_eeg, preload=True)
# Annotate channel types according to config.
data = init_channel_types(data, channel_types=config.channel_types)
# Update the locations of channels for future plot.
data = init_channel_montage(data, path_montage)
## Preprocess data.
# Remove bad channels from data.
data = remove_bad_channels(data, bad_channels=config.bad_channels)
# Set reference channels of data.
data = set_ref_channels(data, ref_channels=config.ref_channels)
# Filter parts of signals' spectrum.
data = filter_spectrum(data, band_frange=config.band_frange, notch_freqs=config.notch_freqs)
# SSP
rojs_ecg, events_ecg  = mne.preprocessing.compute_proj_ecg(data1,n_eeg = 2,reject=None)

The Second is what I do for MEG
base = os.path.join(os.getcwd(), os.pardir, os.pardir)
path_data = os.path.join(base, "data", "meg", "demo","001", "20230214")
path_meg = os.path.join(path_data, "meg", "FL2_TEST1.fif")
path_output = os.path.join(os.getcwd(), "output")

# Execute preprocess pipeline.
data = mne.io.read_raw_fif(path_meg, preload=True)
ecg_evoked = mne.preprocessing.create_ecg_epochs(data).average()
```

### Expected results

Created an SSP operator (subspace dimension = 3)
3 projection items activated
Using data from preloaded Raw for 58 events and 61 original time points ...
0 bad epochs dropped
Adding projection: planar--0.200-0.400-PCA-01
Adding projection: axial--0.200-0.400-PCA-01
Adding projection: eeg--0.200-0.400-PCA-01

### Actual results

Traceback (most recent call last):
  File "F:\PKU2\Research2\Yunzhe_Liu\Code\eeg2label\preprocess\meg\session.py", line 71, in <module>
    data = preprocess_session(path_meg, path_output)
  File "F:\PKU2\Research2\Yunzhe_Liu\Code\eeg2label\preprocess\meg\session.py", line 56, in preprocess_session
    ecg_evoked = mne.preprocessing.create_ecg_epochs(data).average()
  File "<decorator-gen-434>", line 12, in create_ecg_epochs
  File "E:\Anaconda3\envs\tensorflow\lib\site-packages\mne\preprocessing\ecg.py", line 346, in create_ecg_epochs
    events, _, _, ecg = find_ecg_events(
  File "<decorator-gen-433>", line 12, in find_ecg_events
  File "E:\Anaconda3\envs\tensorflow\lib\site-packages\mne\preprocessing\ecg.py", line 194, in find_ecg_events
    ecg, _ = _make_ecg(raw, start=None, stop=None)
  File "<decorator-gen-435>", line 12, in _make_ecg
  File "E:\Anaconda3\envs\tensorflow\lib\site-packages\mne\preprocessing\ecg.py", line 383, in _make_ecg
    raise ValueError('Unable to generate artificial ECG channel')
ValueError: Unable to generate artificial ECG channel

### Additional information

Platform:         Windows-10-10.0.19044-SP0
Python:           3.9.13 (main, Oct 13 2022, 21:23:06) [MSC v.1916 64 bit (AMD64)]
Executable:       E:\Anaconda3\envs\tensorflow\python.exe
CPU:              AMD64 Family 23 Model 96 Stepping 1, AuthenticAMD: 12 cores
Memory:           15.4 GB

mne:              1.3.0
numpy:            1.24.1 {MKL 2022.1-Product with 6 threads}
scipy:            1.10.0
matplotlib:       3.6.3 {backend=QtAgg}

sklearn:          1.2.0
numba:            Not found
nibabel:          5.0.0
nilearn:          0.10.0
dipy:             1.6.0
openmeeg:         Not found
cupy:             Not found
pandas:           1.5.2
pyvista:          0.37.0 {OpenGL could not be initialized}
pyvistaqt:        0.9.0
ipyvtklink:       0.2.3
vtk:              Not found
qtpy:             2.3.0 {PyQt5=5.15.2}
ipympl:           Not found
pyqtgraph:        0.13.1
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.4.0
mne_connectivity: Not found
mne_icalabel:     0.4

Serendipityzzz, 2023-02-16:
# For the code that gives ValueError, it's 
[image]

larsoner, 2023-02-16:
Artificial ECG requires MEG channels. We should update this error message to be clearer that this is the case

Serendipityzzz, 2023-02-16:
Thx! I've solved the problem. And I wonder that after we calculate the ecg/eog projection with compute_proj_ecg/eog, what should we do to clean the raw data with these peojections. Maybe I should use **raw.apply_proj()** to rroject raw data onto a hyperplane orthogonal to noise?

which gose like 
“”“
projs_eog, events_eog = mne.preprocessing.compute_proj_eog(raw,ch_name="EOG",reject=None)
raw.add_proj(projs_eog)
raw.apply_proj()
”“”

Is it right?

larsoner, 2023-02-16:
This is venturing into usage territory rather than code bug fix territory, please open a mne.discourse.group forum post for usage questions!

PR #12716 (2024-07-19): DOC: Update install docs for Spyder
The bulk of this has to do with us no longer packaging Spyder in our installers and adjusting the install docs (mostly) minimally to comply with this change. #10707 and #11615 for example ask for something more comprehensive and aren't meant to be addressed here. This does help a tiny bit (maybe!) with #10707 in the sense that it deduplicates some stuff. I also snuck in a couple of tiny single-digit-line changes to close three other issues:

1. Pinning `numpy<3`
2. Adding a note that `compute_rank` operates on data channels
3. Improving an error message

@ho [... cut]
closed 2024-07-19 by hoechenberger