mne-tools/mne-python #11483: Arguments of the layout reader

mscheltienne, 2023-02-16:
### Describe the new feature or enhancement

While working on #11473, it was not that straightforward to add support for `Path` in the I/O roundtrip for layouts because of the strange arguments from [`read_layout`](https://mne.tools/dev/generated/mne.channels.read_layout.html#mne.channels.read_layout).

`kind` is supposed to be the name of the layout. According to the docstring, without the suffix.
`path` is supposed to be the folder in which the layout file is.

In reality, `kind` can be provided with or without suffix, and it can be the path to a layout file in which case the value provided in `path` is ignored. https://github.com/mne-tools/mne-python/blob/d740e47c8bbce81e6c0beda304748e606f20fc6c/mne/channels/tests/test_layout.py#L60-L61

------------------

I propose to add a new argument `fname`:

```
fname : path-like | str
    Either the path to a .lout or .lay file or the name of a built-in layout.
    c.f. Notes for a list of the available built-in layouts.
```

And to deprecate both `kind` and `path`.

### Describe your proposed implementation

-

### Describe possible alternatives

-

drammock, 2023-02-16:
I think it's useful to be able to just say `'standard-1020'` it whatever so that users don't have to know where the MNE built-in layouts are stored. Is there a way to keep that but otherwise do as you suggest?

mscheltienne, 2023-02-16:
> And make the list of built-in layouts include the suffix to reduce ambiguity.

This point can be omitted, in which case `fname` is either the path to a layout or the name of a layout.

drammock, 2023-02-16:
SGTM.

PR #11500 (2023-02-27): Deprecate 'kind' and 'path' in favor of 'fname' in the layout reader
Closes #11483

As discussed in #11483, those 2 arguments don't play well and could be replaced by a common explicit `fname` argument.
closed 2023-02-27 by larsoner