mne-tools/mne-python #11486: mne.io.read_raw_edf cannot load data correctly with argument `include` or `exclude`

datalw, 2023-02-16:
### Description of the problem

I just tried to load EEG, EMG and bio channels separately, unfortunately I did not suceed to load the data correctly. 

### Steps to reproduce

```Python
eeg_raw=mne.io.read_raw_edf('test.edf', exclude=['Resp', 'BVP-0','EMG-0','EMG-1','EMG-2','EMG-3', 'BVP-1'], preload=True)
bio_raw=mne.io.read_raw_edf('test.edf', include=[ 'Resp','BVP-0','BVP-1'], preload=True)
emg_raw=mne.io.read_raw_edf('test.edf', include=[ 'EMG-0','EMG-1','EMG-2','EMG-3'], preload=True) 
```
```

### Link to data

https://drive.google.com/file/d/1IMP_HK0JvzNOzd7worc3zDTBy7f3E9Zj/view?usp=sharing

### Expected results

`eeg_raw` should have all 21 eeg channels.
`bio_raw` should have the 3 channels 'Resp','BVP-0','BVP-1'
`emg_raw` should have the 4 emg channels  'EMG-0','EMG-1','EMG-2','EMG-3'

### Actual results

The result is that,

`eeg_raw` only excluded one bio channel ‘Resp’.
`emg_raw` had no channel.
`bio_raw` had only 1 channel ‘Resp’…

Then I tried upgraded mne to 1.3, the result is that,

`eeg_raw` only excluded one bio channel ‘Resp’.
The following 2 lines did not work and I got TypeError: `read_raw_edf() got an unexpected keyword argument 'include' `. I looked into the function and found that `include` was not implemented in the function `mne.io.read_raw_edf`. According to the documentation it should be there…

### Additional information

```
Platform:         Windows-10-10.0.22621-SP0
Python:           3.9.13 | packaged by conda-forge | (main, May 27 2022, 16:50:36) [MSC v.1929 64 bit (AMD64)]
Executable:       [c:\Users\luwan\anaconda3\envs\mne1_1\python.exe](file:///C:/Users/luwan/anaconda3/envs/mne1_1/python.exe)
CPU:              AMD64 Family 25 Model 80 Stepping 0, AuthenticAMD: 12 cores
Memory:           15.3 GB

mne:              1.1.1
numpy:            1.23.3 {blas=NO_ATLAS_INFO, lapack=lapack}
scipy:            1.9.1
matplotlib:       3.6.0 {backend=Qt5Agg}

sklearn:          1.1.2
numba:            0.56.2
nibabel:          4.0.2
nilearn:          0.9.2
dipy:             1.5.0
cupy:             Not found
pandas:           1.5.0
pyvista:          0.36.1 {OpenGL 4.5.0 Compatibility Profile Context 22.20.02.04.220711 via AMD Radeon(TM) Graphics}
pyvistaqt:        0.9.0
ipyvtklink:       0.2.3
vtk:              9.1.0
qtpy:             2.2.1 {PyQt5=5.12.9}
ipympl:           0.8.7
pyqtgraph:        [c:\Users\luwan\anaconda3\envs\mne1_1\lib\site-packages\pyqtgraph\colors\palette.py:1](file:///C:/Users/luwan/anaconda3/envs/mne1_1/lib/site-packages/pyqtgraph/colors/palette.py:1): RuntimeWarning: PyQtGraph supports Qt version >= 5.15, but 5.12.9 detected.
  from ..Qt import QtGui
0.13.1
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.4.0
mne_connectivity: 0.4.0
mne_icalabel:     Not found
```

cbrnr, 2023-02-16:
It should be noted that this file has signals with different sampling frequencies. When fixing `include` and `exclude`, picking a particular signal group should *not* trigger resampling, which I think is currently the case.

datalw, 2023-03-21:
> It should be noted that this file has signals with different sampling frequencies. When fixing `include` and `exclude`, picking a particular signal group should _not_ trigger resampling, which I think is currently the case.

Hi, I would like to have a quick follow-up: is fixing this bug already in plan?