mne-tools/mne-python #11487: Deleting annotation in `raw.plot()` does not work

cbrnr, 2023-02-16:
### Description of the problem

When deleting an annotation in `raw.plot()` with a right-click, a residual annotation (with duration around zero) remains right where the mouse was clicked.

### Steps to reproduce

```Python
import mne

folder = mne.datasets.sample.data_path()
file = folder / "MEG" / "sample" / "sample_audvis_raw.fif"
raw = mne.io.read_raw_fif(file)

raw.plot()
```

### Expected results

The annotation should be completely removed.

### Actual results

A very short annotation remains.

https://user-images.githubusercontent.com/4377312/219396442-29041e00-f56d-48e7-87c1-41a90fb98c5a.mov

I also get this error message in the console:

```
Traceback (most recent call last):
  File "/Users/clemens/Projects/mne-python/.direnv/python-3.10.10/lib/python3.10/site-packages/matplotlib/cbook/__init__.py", line 307, in process
    func(*args, **kwargs)
  File "/Users/clemens/Projects/mne-python/mne/viz/_mpl_figure.py", line 187, in _close
    super()._close(event)
  File "/Users/clemens/Projects/mne-python/mne/viz/_mpl_figure.py", line 111, in _close
    self.mne.parent_fig.mne.child_figs.remove(self)
ValueError: list.remove(x): x not in list
```

### Additional information

Platform:         macOS-11.7.4-x86_64-i386-64bit
Python:           3.10.10 (main, Feb  8 2023, 05:44:44) [Clang 13.0.0 (clang-1300.0.29.30)]
Executable:       /Users/clemens/Projects/mne-python/.direnv/python-3.10.10/bin/python
CPU:              i386: 8 cores
Memory:           Unavailable (requires "psutil" package)
mne:              1.4.dev0
numpy:            1.24.2 {OpenBLAS 0.3.21 with 4 threads}
scipy:            1.10.0
matplotlib:       3.6.3 {backend=MacOSX}
sklearn:          1.2.1
numba:            Not found
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           Not found
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             Not found
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0
mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

mscheltienne, 2023-02-16:
It looks like it's because in the legacy `matplotlib` backend you can define an annotation either with the left click or with the right click. However, I don't get any error traceback in the console.

Also, interestingly, if you do left click first on the created annotation and then right-click it will delete it without creating a new short annotations where you right clicked.

cbrnr, 2023-02-16:
Thanks for investigating @mscheltienne! I don't think creating an annotation with a right click is intentional (or is it?). I thought it was mapped to the green line which indicates the time under the mouse cursor.

mscheltienne, 2023-02-16:
> I don't think creating an annotation with a right click is intended (or is it?)

I don't think so either, and disabling this "feature" would be a good fix.

On the Qt backend, the right click is not drawing annotations.

drammock, 2023-02-16:
agree that right-click is not supposed to create annotations. In non-annotation mode it clears the green line (left-click sets it), in annotation mode it's supposed to clear an annotation (left-drag creates it).

PR #11491 (2023-02-17): Accept only left-clicks for adding annotations
Fixes #11487.

A little heads up: Matplotlib 3.7 is out, and it breaks our raw plot in several places (meaning that the annotations window cannot even be opened). This is for another PR though.
closed 2023-02-17 by drammock