mne-tools/mne-python #11492: Matplotlib 3.7 breaks annotations window

cbrnr, 2023-02-17:
### Description of the problem

Matplotlib 3.7 breaks the annotation window in `raw.plot()` (which appears when pressing <kbd>a</kbd>).

### Steps to reproduce

```Python
1. Open `raw.plot()`
2. Press <kbd>a</kbd>
```

### Expected results

The annotation window should appear.

### Actual results

The annotation window does not appear. Instead, I get the following error:

```
>>> Traceback (most recent call last):
  File "/Users/clemens/Projects/mne-python/.direnv/python-3.10.10/lib/python3.10/site-packages/matplotlib/cbook/__init__.py", line 304, in process
    func(*args, **kwargs)
  File "/Users/clemens/Projects/mne-python/mne/viz/_mpl_figure.py", line 676, in _keypress
    self._toggle_annotation_fig()
  File "/Users/clemens/Projects/mne-python/mne/viz/_mpl_figure.py", line 1113, in _toggle_annotation_fig
    self._create_annotation_fig()
  File "/Users/clemens/Projects/mne-python/mne/viz/_mpl_figure.py", line 936, in _create_annotation_fig
    self._update_annotation_fig()
  File "/Users/clemens/Projects/mne-python/mne/viz/_mpl_figure.py", line 1046, in _update_annotation_fig
    ax.buttons = RadioButtons(ax, labels)
  File "/Users/clemens/Projects/mne-python/.direnv/python-3.10.10/lib/python3.10/site-packages/matplotlib/widgets.py", line 1670, in __init__
    self.value_selected = labels[active]
IndexError: list index out of range
```

### Additional information

<details>

```
Platform:         macOS-11.7.4-x86_64-i386-64bit
Python:           3.10.10 (main, Feb  8 2023, 05:44:44) [Clang 13.0.0 (clang-1300.0.29.30)]
Executable:       /Users/clemens/Projects/mne-python/.direnv/python-3.10.10/bin/python
CPU:              i386: 8 cores
Memory:           Unavailable (requires "psutil" package)
mne:              1.4.dev0
numpy:            1.24.2 {OpenBLAS 0.3.21 with 4 threads}
scipy:            1.10.0
matplotlib:       3.7.0 {backend=MacOSX}

sklearn:          1.2.1
numba:            Not found
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           Not found
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             Not found
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found
```
</details>

cbrnr, 2023-02-17:
The problem seems to be [this line](https://github.com/mne-tools/mne-python/blob/main/mne/viz/_mpl_figure.py#L1046):

```python
ax.buttons = RadioButtons(ax, labels)
```

When there are no annotations, `labels` is an empty list. This worked in previous Matplotlib versions < 3.7.0 (see [here](https://github.com/matplotlib/matplotlib/blob/v3.6.3/lib/matplotlib/widgets.py#L1400-L1455)), but now this is not possible anymore (see [here](https://github.com/matplotlib/matplotlib/blob/v3.7.0/lib/matplotlib/widgets.py#L1670)).

I guess it was probably by chance that this worked previously, so now we will have to adapt our code. @drammock do you agree?

larsoner, 2023-02-17:
From [CircleCI](https://app.circleci.com/pipelines/github/mne-tools/mne-python/17994/workflows/0a54bb02-ba11-43bf-ad3a-90bf0d18dca1/jobs/52520) also:
```
  File "/home/circleci/project/tutorials/preprocessing/50_artifact_correction_ssp.py", line 183, in <module>
    fig = epochs.average().plot_topomap(times, proj='interactive')
  File "/home/circleci/project/mne/evoked.py", line 435, in plot_topomap
    return plot_evoked_topomap(
  File "/home/circleci/project/mne/viz/topomap.py", line 1826, in plot_evoked_topomap
    _draw_proj_checkbox(None, params)
  File "/home/circleci/project/mne/viz/utils.py", line 487, in _draw_proj_checkbox
    for rect in proj_checks.rectangles:
...
matplotlib._api.deprecation.MatplotlibDeprecationWarning: The rectangles attribute was deprecated in Matplotlib 3.7 and will be removed two minor releases later. Any custom property styling may be lost.
```
And:
```
Traceback (most recent call last):
  File "/home/circleci/project/tutorials/inverse/20_dipole_fit.py", line 25, in <module>
    from nilearn.plotting import plot_anat
  File "/home/circleci/python_env/lib/python3.10/site-packages/nilearn/plotting/__init__.py", line 47, in <module>
    from . import cm
  File "/home/circleci/python_env/lib/python3.10/site-packages/nilearn/plotting/cm.py", line 226, in <module>
    _cm.register_cmap(name=k, cmap=v)
...
matplotlib._api.deprecation.MatplotlibDeprecationWarning: The register_cmap function was deprecated in Matplotlib 3.7 and will be removed two minor releases later. Use ``matplotlib.colormaps.register(name)`` instead.
```
And:
```
Traceback (most recent call last):
  File "/home/circleci/project/tutorials/clinical/30_ecog.py", line 150, in <module>
    rgba = cm.get_cmap("viridis")
...
matplotlib._api.deprecation.MatplotlibDeprecationWarning: The get_cmap function was deprecated in Matplotlib 3.7 and will be removed two minor releases later. Use ``matplotlib.colormaps[name]`` or ``matplotlib.colormaps.get_cmap(obj)`` instead.
```

drammock, 2023-02-17:
closing as dup of #11332, will be addressed by #11409

closed 2023-02-17 by drammock