mne-tools/mne-python #11494: Broken link in GSoC ideas

Dev-satish, 2023-02-18:
### Proposed documentation enhancement

# Problem Area
- Broken link in [GSoC ideas page](https://github.com/mne-tools/mne-python/wiki/GSoC-Ideas#mne-python)
  - I discovered a broken link `https://mne.tools/dev/install/mne` for the **Installation instructions** on the [GSoC ideas page](https://github.com/mne-tools/mne-python/wiki/GSoC-Ideas) that needs to be updated with a properly functional one.

mscheltienne, 2023-02-18:
Thank you for the report, since the wiki page can't be edited by everyone as the code can be through a pull request, I fixed it.

Dev-satish, 2023-02-18:
> Thank you for the report, since the wiki page can't be edited by everyone as the code can be through a pull request, I fixed it.

Alright! Shall I close this issue then?

mscheltienne, 2023-02-18:
Yes, you can :)

closed 2023-02-18 by Dev-satish

larsoner, 2023-02-18:
Thanks for the report @Dev-satish and quick fix @mscheltienne !