mne-tools/mne-python #11495: Event names getting overlapped It

Dev-satish, 2023-02-20:
### Description of the problem

While I was trying to reproduce the issue in [#130 in mne-tools/mne-qt-browser](https://github.com/mne-tools/mne-qt-browser/issues/130) I saw that the event name are getting overlapped with each other 

### Steps to reproduce

```Python
# %%
import mne

sample_dir = mne.datasets.sample.data_path()
sample_fname = sample_dir / 'MEG' / 'sample' / 'sample_audvis_raw.fif'

raw = mne.io.read_raw_fif(sample_fname)
raw.crop(tmax=60)

events = mne.find_events(raw, stim_channel='STI 014')
event_id = {'auditory/left': 1, 'auditory/right': 2, 'visual/left': 3,
            'visual/right': 4, 'face': 5, 'buttonpress': 32}

raw.plot(events=events, event_id=event_id)
```

### Expected results

It should have been consistent with the event duration like shown by @hoechenberger in the issue

[image]

### Actual results

[image]

### Additional information

none

Dev-satish, 2023-02-20:
I realized that as soon as I press `- time` button it the event name separates but I feel this issues can still be worked upon
[image]

Dev-satish, 2023-02-20:
What if we add a header and display the description there.
I think I could explain it better visually- 
[image]
[image]

cbrnr, 2023-02-20:
You will always be able to have long enough event names to overflow even in a dedicated area.

cbrnr, 2023-02-20:
> Yes it's definitely an issue; I wonder if we could detect the collision and display just a marker or truncated event description in that case, and only expose the full description on hover.

If this is possible I think a truncated version would be very nice! I'd truncate first, and if even that doesn't work (e.g. zoom out far enough) I'd remove the label completely.

Color-coding event types would be a good idea not just in that particular case – is this supported?

hoechenberger, 2023-02-20:
I'd like to second @cbrnr's proposal, I like that

marsipu, 2023-02-20:
> Color-coding event types would be a good idea not just in that particular case – is this supported?

You can pass a dict with custom colors to ˋevent_colorˋ in [raw.plot()](https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.plot). Or do you mean something else?

marsipu, 2023-02-20:
A qt-favoring solution in terms of simplicity might also be giving the event-labels a background with the color of the plot-background. If they stack you could at least read the one on top. And to expand we could intercept `hoverEventˋ of [ˋEventLineˋ](https://github.com/mne-tools/mne-qt-browser/blob/e3fec001106fab34db0dc87f121f50dc278ddc40/mne_qt_browser/_pg_figure.py#L1380l) to put the hovered EventLine on top (ˋself.setZValue(1)ˋ).
You find the code for the ˋInfiniteLineˋ from pyqtgraph which `EventLine` inherits [here](https://github.com/pyqtgraph/pyqtgraph/blob/master/pyqtgraph/graphicsItems/InfiniteLine.py).

cbrnr, 2023-02-21:
> You can pass a dict with custom colors to event_color in [raw.plot()](https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw.plot). Or do you mean something else?

Yes, but I meant the default behavior should be different colors for different events.