mne-tools/mne-python #11513: plot_topomap produces spurious output in jupyter

sam-s, 2023-02-28:
### Description of the problem

The following code:
```python
fig = plt.figure(figsize=(10,2))
axes = fig.subplots(ncols=len(bdf.columns))
for ax, band in zip(axes, bdf.columns):
    x = mne.viz.plot_topomap(data=bdf[band], pos=info, axes=ax)
    # print(type(x))
    bot,top = hg.band_range(band)
    ax.set_title(f"{band}\n({bot}--{top} Hz)")
```
produces 5 spurious "outputs" in Jupyter (one for each band):
[image]

### Steps to reproduce

```Python
fig = plt.figure(figsize=(4,2))
df = pd.DataFrame({"delta":[0.1, 0.2, 0.4], "gamma":[0.3, 0.6, 0.2]},
                  index=["C4","P4","O1"])
info = mne.create_info(ch_names=list(df.index), sfreq=500, ch_types="eeg")
info.set_montage("standard_1020")
axes = fig.subplots(ncols=len(df.columns))
for ax, band in zip(axes, df.columns):
    x = mne.viz.plot_topomap(data=df[band], pos=info, axes=ax)
    ax.set_title(band)
```
```

### Expected results

no output 

### Actual results

2 "lines" (1 image, 1 text)
![image](https://user-images.githubusercontent.com/1596894/221993012-a4cbd9e2-1c45-4c38-9ebb-c403205fd4d8.png)

### Additional information

Platform:         Linux-6.0.12-76060006-generic-x86_64-with-glibc2.35
Python:           3.10.6 (main, Nov 14 2022, 16:10:14) [GCC 11.3.0]
Executable:       /usr/bin/python3
CPU:              x86_64: 8 cores
Memory:           15.3 GB

mne:              1.3.0
numpy:            1.24.2 {OpenBLAS 0.3.21 with 8 threads}
scipy:            1.10.1
matplotlib:       3.7.0 {backend=module://matplotlib_inline.backend_inline}

sklearn:          1.2.1
numba:            Not found
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           1.5.3
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             2.3.0 {PyQt5=5.15.3}
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: 0.5.0
mne_icalabel:     Not found

sam-s, 2023-02-28:
Work around is to pass `show=False` to `mne.viz.plot_topomap` - this way only _one_ figure is shown instead of 5.

larsoner, 2023-03-01:
@sam-s thanks for investigating and finding a solution to the problem! I actually think this is the correct/intended behavior at the MNE end, so I'd consider the "workaround" more of a bugfix in your use of the MNE viz function, more or less. Our viz functions all (I think?) default to `show=True` so if you want it not to show at the end -- like you do in this loop -- you have to pass `show=False` when calling the function. Make sense?

I'll close since I don't think there is anything to change at the MNE end for this.

closed 2023-03-01 by larsoner

sam-s, 2023-03-01:
I disagree. MNE places unwanted output (text and image) into the notebook, **even with `show=False`**.
IMO, this is a bug.

larsoner, 2023-03-02:
> Still I get the spurious output, just less of it.

This is what I see when adding `show=False`:

[image]

Is it not the same for you? To me this looks correct

larsoner, 2023-03-02:
... or at least, not an MNE-specific bug/behavior:

[image]

drammock, 2023-03-03:
@sam-s there is a detailed discussion of root of this problem on the matplotlib forum: https://discourse.matplotlib.org/t/custom-plot-function-in-jupyter-notebooks/23185

TL;DR: the notebook behavior is not something we can fix/control from within MNE-Python.

I will concede that it's odd / unexpected for a parameter `show=False` to be *required* to get *only one (complete) figure to show up* in a Jupyter notebook --- ideally `show=True` would show one complete figure, and `show=False` wouldn't show any figures.  The `show` param was originally conceived for windows that pop open during an interactive console session (not a notebook), and (perhaps unfortunately?) we tried to make it work for notebooks too.  As @larsoner demonstrated, it *does* work for notebooks, in the sense that it does what is necessary to prevent the figures being shown prematurely with empty axes. I'll open an issue about improving our documentation / possibly renaming the `show` parameter.