mne-tools/mne-python #11514: plot_topo_psd is impossible when epochs are concatenated

Farzaneh-Taleb, 2023-03-01:
### Description of the problem

Since the output of concatenate_epochs is an instance of BaseEpochs, it is impossible to apply plot_psd_topomap directly on that.

### Steps to reproduce

```Python
.
```

### Expected results

.

### Actual results

.

### Additional information

.

mscheltienne, 2023-03-01:
I think you are running an outdated version of MNE-Python. `concatenate_epochs` returns an `EpochsArray`: https://mne.tools/stable/generated/mne.concatenate_epochs.html

The following code snippet works for me:

```
from mne import make_fixed_length_epochs, concatenate_epochs
from mne.datasets import sample
from mne.io import read_raw_fif

raw = read_raw_fif(
    sample.data_path() / "MEG" / "sample" / "sample_audvis_raw.fif", 
    preload=False,
)
raw.crop(0, 30)
raw.load_data()
epochs1 = make_fixed_length_epochs(raw)
raw = read_raw_fif(
    sample.data_path() / "MEG" / "sample" / "sample_audvis_raw.fif", 
    preload=False,
)
raw.crop(40, 70)
raw.load_data()
epochs2 = make_fixed_length_epochs(raw)

epochs = concatenate_epochs([epochs1, epochs2])
epochs.plot_psd_topomap()
```

Note also that `.plot_psd_topomap` is a legacy methd. New code should use `.compute_psd` which returns an instance of the `EpochsSpectrum`.

Mathieu

Farzaneh-Taleb, 2023-03-01:
You are right! Sorry about that and thanks for the help!

closed 2023-03-01 by Farzaneh-Taleb