mne-tools/mne-python #11520: raw.compute_psd().plot() output a blank figure with black background

impPie, 2023-03-02:
### Description of the problem

raw.compute_psd().plot() output a blank figure with black background in my win10 computer.
But when I ran the same codes in google colab, it shows the correct figure.

### Steps to reproduce

```Python
psd=raw.compute_psd()
print(psd)
psd.plot()
```

### Expected results

[image]

### Actual results

[image]

### Additional information

when I use the raw.plot(). It can show a proper result

agramfort, 2023-03-02:
can you try psd.plot(color="white")?

it seems your machine has a dark background setup

> Message ID: ***@***.***>
>

impPie, 2023-03-06:
[image]

I tried but it still not work.

impPie, 2023-03-16:
This was a vscode jupyter issue, just close the `Jupyter: Theme Matplotlib Plots
`  in VScode settings can fix it.

closed 2023-03-16 by impPie

drammock, 2023-03-16:
Thanks for reporting back the solution! I never would have found that since I didn't even know about that vscode setting 😅