mne-tools/mne-python #11522: Add `colorbar` parameter to `mne.viz.plot_topomap()`?

cbrnr, 2023-03-02:
I was wondering if there is a reason why `mne.viz.plot_topomap()` does not have a `colorbar` parameter. Although it is easy to add a colorbar after the figure has been created, it is easy only if you know how. Many other topomap functions do have such a parameter, but not the generic `mne.viz.plot_topomap()`.

kar-hik, 2023-03-26:
can you assign me this issue i can work on this.

cbrnr, 2023-03-26:
Thanks @kar-hik!

NadipilliAditya, 2025-04-07:
Hi, I am Aditya. Recently I registered for the GSOC 2025. I would like to work on this issue 
so, can you please assign me this issue .

drammock, 2025-04-07:
@NadipilliAditya there's already an open PR #13187 that addresses this one.

NadipilliAditya, 2025-04-07:
> [@NadipilliAditya](https://github.com/NadipilliAditya) there's already an open PR [#13187](https://github.com/mne-tools/mne-python/pull/13187) that addresses this one.

Thank you for the update! I’ll explore other open issues in the repo and see where I can contribute. Please feel free to suggest if there's any beginner-friendly issue or area that could use help.