mne-tools/mne-python #11529: raw.plot does not work with show_first_samp

jasmainak, 2023-03-03:
### Description of the problem

Plotting raw data with `show_first_samp=True` gives strange results. 

### Steps to reproduce

```Python
import numpy as np
import mne

sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = (sample_data_folder / 'MEG' / 'sample' /
                        'sample_audvis_filt-0-40_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)

events = mne.make_fixed_length_events(raw)
raw.plot(events=events, event_color='r', show_first_samp=True)
```

### Expected results

We would expect the data to be shown but it is not. It seems to suggest some indexing bug.

### Actual results

[image]

### Additional information

```py
Platform:         Linux-3.10.0-1160.76.1.el7.x86_64-x86_64-with-glibc2.10
Python:           3.8.6 | packaged by conda-forge | (default, Oct  7 2020, 19:08:05)  [GCC 7.5.0]
Executable:       /autofs/space/meghnn_001/users/mjas/anaconda3/envs/mne/bin/python3.8
CPU:              x86_64: 64 cores
Memory:           125.4 GB

Exception ignored on calling ctypes callback function: <function _ThreadpoolInfo._find_modules_with_dl_iterate_phdr.<locals>.match_module_callback at 0x7f80b4d9b040>
Traceback (most recent call last):
  File "/autofs/space/meghnn_001/users/mjas/anaconda3/envs/mne/lib/python3.8/site-packages/threadpoolctl.py", line 400, in match_module_callback
    self._make_module_from_path(filepath)
  File "/autofs/space/meghnn_001/users/mjas/anaconda3/envs/mne/lib/python3.8/site-packages/threadpoolctl.py", line 515, in _make_module_from_path
    module = module_class(filepath, prefix, user_api, internal_api)
  File "/autofs/space/meghnn_001/users/mjas/anaconda3/envs/mne/lib/python3.8/site-packages/threadpoolctl.py", line 606, in __init__
    self.version = self.get_version()
  File "/autofs/space/meghnn_001/users/mjas/anaconda3/envs/mne/lib/python3.8/site-packages/threadpoolctl.py", line 646, in get_version
    config = get_config().split()
AttributeError: 'NoneType' object has no attribute 'split'
mne:              1.4.dev0
numpy:            1.23.0 {OpenBLAS 0.3.12 with 1 thread}
scipy:            1.5.3
matplotlib:       3.3.3 {backend=Qt5Agg}

sklearn:          0.23.2
numba:            Not found
nibabel:          3.2.1
nilearn:          0.7.0
dipy:             1.3.0
openmeeg:         Not found
cupy:             Not found
pandas:           1.1.5
pyvista:          0.36.1 {OpenGL 4.5.0 NVIDIA 455.45.01 via Quadro P5000/PCIe/SSE2}
pyvistaqt:        0.9.0
ipyvtklink:       Not found
vtk:              9.0.1
qtpy:             2.0.1 {PyQt5=5.12.9}
ipympl:           Not found
pyqtgraph:        0.13.1
pooch:            v1.5.2

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   0.4.0
mne_connectivity: Not found
mne_icalabel:     Not found

```

cbrnr, 2023-03-03:
How does the plot look if you use the Matplotlib backend?

jasmainak, 2023-03-03:
[image]

But it's still buggy ... it doesn't show the events

agramfort, 2023-03-05:
@jasmainak can you have a look? at least at the matplotlib code?

> Message ID: ***@***.***>
>

kar-hik, 2023-03-26:
If you are using MNE-Python version 0.23.0, you may want to update to a later version to resolve this issue. Alternatively, you could manually adjust the x-axis limits using the xlim argument in raw.plot(). For example, you could set raw.plot(events=events, event_color='r', xlim=(0, raw.times[-1])) to start the x-axis at 0 and end it at the last sample in the data. i this this would we usefull.