mne-tools/mne-python #11531: [DOC/API] clearer guidance and documentation on how to use various .pick* methods

dengemann, 2023-03-03:
### Proposed documentation enhancement

For selecting channels, there are various methods.

The older ways are `.pick_types()` and  `.pick_channels()` whereas the modern way of picking is using `.pick()`.

This may sound like a nitpicky affair [pun intended], but it turns out that passing a list of channels to `.pick_channels()` versus `.pick()` will lead to different results: both functions will pick but only `.pick` will act like similar to a fancy index in numpy, that is, reorder the channels of the instance.

While this is not a bug, it may lead to bugs for e.g. machine learning projects in wich the MNE ways and Numpy/sklearn/PyTorch ways are mixed and combined. Imagine you want to process a large dataset like Temple University Hospital EEG and want to make sure that every recroding is processed not only with the same selection of channels but also with all channels in the same order for all recordings (not a given as channel ordering may change from recording to recording). 

Relying on `.pick_channels()` would not do the job. Only `.pick()` gets this right. Would you have known? Having some bug in your analysis? ^^

Where do we go from here? For me, improving the documentation would be the least we should do. As discussed with @drammock another option may be to soft-depracate the old ways in printing a user warning and providing a legacy API. I don‘t want to create too much work, but adding some warning that makes users aware of the fact that `.pick_channels` will not reorder your channels sounds like a good idea.

Any thoughts, MNE wizzards?

alexrockhill, 2023-03-04:
+1 for depreciate with informative warning, that sounds like the most helpful

agramfort, 2023-03-04:
just to balance the argument here. The original idea was that a group of
channels is to be considered as a "set" so unordered.

That's why the parameter "ordered" was added to pick_channels:
https://mne.tools/stable/generated/mne.pick_channels.html

I would first consider a better documentation.

In sklearn we heard comparable complaints as people where passing
dataframes and sklearn was ignored the column order
and it was creating bugs. sklearn did not change for this. It was
considered user issue.

just thinking out loud...

> Message ID: ***@***.***>
>

drammock, 2023-03-04:
Chiming in to say that there's a middle ground between deprecation (which implies eventual removal) and no deprecation: the `legacy` decorator. It exists so that old, widely-used API elements can be kept around, old scripts will still work, but send a strong message not to use it in new code. If the old methods are *less good* but not technically buggy, I think marking as legacy is a good way to go (combined with documentation improvements). I also like @dengemann idea to add an info message saying if order was ignored

larsoner, 2023-03-06:
> This may sound like a nitpicky affair [pun intended], but it turns out that passing a list of channels to .pick_channels() versus .pick() will lead to different results: both functions will pick but [...]

> just to balance the argument here. The original idea was that a group of
channels is to be considered as a "set" so unordered.

Indeed, at the same time this has been a source of confusion (and probably still silent bugs!) since at least 2015:

https://github.com/mne-tools/mne-python/issues/2300#issuecomment-167115027

One reason I've thought to keep `pick_channels` is in some weird case where you have a channel with the name of a channel type, the `.pick` which tries to decide automatically if you're talking about types, indices, or channel names might be confused. But this is such a rare corner case that people who run into this can figure out the indices to use themselves using `inst.ch_names` and pass those directly.

So overall I'm +0.5 for `@legacy` and +1 for deprecating `pick_channels` completely. And if we try to make this change in our own code and tests we might see if/where using `.pick` falls short.

alexrockhill, 2023-03-06:
Maybe long deprecation cycle like 2.0?

larsoner, 2023-03-07:
I usually agree, but in this case we might very well be fixing 10 year old silent user bugs in the process (which isn't usually the case with our deprecations)

drammock, 2023-03-07:
@jasmainak you're usually a voice saying "don't break 10-year-old scripts" but you also have expressed dissatisfaction with `pick_channels` in the past ([here](https://github.com/mne-tools/mne-python/issues/2300#issuecomment-167424414), [here](https://github.com/mne-tools/mne-python/issues/2300#issuecomment-167438776). Do you vote for "keep it as `@legacy`" or "deprecate"?

dengemann, 2023-03-07:
@jasmainak the cool thing I discovered about pick is that it also accepts channel names - were you aware? Seems like it can be that one way.

larsoner, 2023-03-07:
> Why would you deprecate pick_channels but not pick_types since pick seems to cover both functions?

Because `pick_channels` has been (and probably still is) silently breaking people's code for years. `pick_types` isn't so we can keep it (why not).

>  it seems that pick_channels has introduced an explicit argument called ordered=False. Does that not address the concerns about channel ordering?

No because `ordered=False` is the default, so silent bugs still likely exist.

> For clinical workflows and longer research projects, it is critical that core functionality "just works" and doesn't evolve too fast.

Agreed but in this case we *want* people to change/change/think about their code. We don't know the prevalence, but multiple core devs have made the mistake of assuming `inst.pick_channels(my_ch_list)` would give them the channels in the given `my_ch_list` order, so I assume it's prevalent for end-users, too.

jasmainak, 2023-03-07:
> No because ordered=False is the default, so silent bugs still likely exist.

Depends what you expect from the method. There is also a `reorder_channels` method, so I would expect folks wanting reordering to use that explicitly. But I don't feel strongly either way, just offering an alternate perspective ... am happy to adapt.

dengemann, 2023-03-08:
Looking deeper into `.pick`, I found another surprise. Consider this minimal example below (tested with main branch of today).

```python
import os
import mne

path = mne.datasets.sample.data_path()
sample_data_folder = mne.datasets.sample.data_path()
sample_data_raw_file = os.path.join(sample_data_folder, 'MEG', 'sample',
                                    'sample_audvis_raw.fif')
raw = mne.io.read_raw_fif(sample_data_raw_file)
raw.copy().pick([0])
try:
    raw.copy().pick([5000])
except Exception as e:
    print(e)

try:
    raw.copy().pick([0, 5000])
except Exception as e:
    print(e)

raw.copy().pick(['MEG 1111'])
try:
    raw.copy().pick(['Haha - fake news!'])
except Exception as e:
    print(e)

# THIS PASSES!!!!
raw.copy().pick(['MEG 1111', 'Haha - fake news!'])

```

`.pick` does order and select channels if channels are channel names, but if there is one offender name / a channel that does not exist among valid channels, the selection is implicitly updated and accepted. This although passing that same name in isolation will lead to a ValueError. Not how this is inconsistent with indices, which lead reliably to out-of-bound errors if one bad index is included.

I would say this is on the bug spectrum. Any thoughts?

dengemann, 2023-03-08:
I see that there is a runtime warning at least. But still, this behavior seems quite implicit. My personal assessment is that you don't want this behavior, you want a proper ValueError here.

larsoner, 2023-03-08:
> you want a proper ValueError here.

Agreed, opened #11546 to discuss there if necessary / not change discussion here from what to do about `.pick_channels`.

drammock, 2023-03-10:
Proposal based on today's dev meeting discussion (with some additional suggestions from me for things we didn't discuss):

- for `pick_channels` *method* (part of `UpdateChannelsMixin`)
    - make `ordered=None` the default (which converts to `False`)
    - with deprecation cycle, change default to `ordered=True`
    - meanwhile emit a `FutureWarning` only if `order=None` *and* the provided ordering would be different if they had passed `ordered=True`
    - mark as `@legacy` and recommend using `.pick` method instead

- for `.pick_types` method:
    - mark as `@legacy` and recommend using `.pick` method instead

- convert all our tutorial/example docs to use `.pick` instead of `.pick_channels` or `.pick_types`

- for *functions* `pick_channels_forward`, `pick_channels_cov`, `pick_channels_evoked`: mark all as `@legacy` (or deprecate/remove?) and recommend:
    - `pick_channels_evoked` -> `evoked.copy().pick()` (`.copy()` is optional of course)
    - `pick_channels_cov` -> new method `cov.pick()` that mirrors the method signature and functionality of the `.pick()` method in the Mixin (i.e., `self, picks, exclude=(), *, verbose=None`). I'm not sure if we need to also add the `ordered` kwarg here; not adding it is equivalent to always passing `ordered=True`. Meanwhile `cov.pick_channels(self, ch_names, ordered=False)` should become legacy/deprecated.
    - `pick_channels_forward`. This one is different because `Forward` doesn't keep track of `bads`, but I think otherwise the method signature and functionality of `.pick()` should work? Therefore propose a new method `forward.pick(self, picks, exclude=(), *, verbose=None`). Here again, not sure if we need to preserve `ordered` kwarg. Also here if we go this route then `forward.pick_channels()` should become legacy/deprecated.

- for *functions* `pick_channels` and `pick_channels_regexp` (these operate on *a list of channel names*, not an MNE object). We didn't discuss these. My personal feeling is that these probably aren't used as much as the instance methods, and they should also be marked `@legacy` or even deprecated/removed, because both functions basically boil down to 2 lines of code:
    - for `pick_channels_regexp` it's literally defined in 2 lines: `r = re.compile(regexp); return [k for k, name in enumerate(ch_names) if r.match(name)]`.
    - for `pick_channels` if you already have a list of all ch_names, a list of includes, and a list of excludes (note that `exclude` does not support 'bads' here!): `chs_to_keep = set(ch_names).intersection(include) - set(exclude)`, then optionally something to preserve order like `np.array(ch_names)[np.isin(ch_names, chs_to_keep)]` or `[ch for ch in ch_names if ch in chs_to_keep]` or `list(filter(lambda ch: ch in chs_to_keep, ch_names))`. 

- for function `pick_info(info, sel=(), copy=True, verbose=None)`: this provides unique functionality. For parity there probably should be an `info.pick(self, picks, *, verbose=None)` method (either in addition to the function or instead of it)... pretty sure it doesn't need `exclude` though so I removed that

jasmainak, 2023-03-11:
@drammock thanks for the comprehensive overview of the discussion. Regarding `pick_channels` and `pick_channels_regexp`, my vote would be to actually keep them. For two reasons:

1. They do not operate on an instance. I have had to do several code gymnastics since the introduction of the `Mixin` classes since these operate on only instances while often we are working at the level of `info`, specially with OPM data. In some situations, you could make a copy of the instance and then pick on that, but imagine doing that when you have several raw objects in memory from different runs. Again, all possible ... but annoying to change the structure of old code that already works (and running into other issues ...).
2. While the functionality is a simple 2 line, it is not always obvious what those two lines are, specially if it involves regular expressions.

agramfort, 2023-03-12:
@jasmainak we will not remove pick_channels and pick_channels_regexp but
change the default of ordered to True to avoid the bugs which
triggered this discussion

> Message ID: ***@***.***>
>

dengemann, 2023-03-17:
Thank you @drammock @larsoner and everyone else for this very constructive and profound collective reflection on some of the most basic pieces of  MNE API. 

I like the action plan a lot, which, I think strikes a nice balance between addressing the issue while avoiding disruption of long-term API contracts. 💯 % endorsement from my side.

My only sort of -0 reservation is about investing too much into upgrading / depracting the functions. These have little user attention and who uses them knows what they do. I guess this would be in line with what @jasmainak said.

I would recommend focusing on the instance methods in a first step.

closed 2023-05-01 by larsoner