mne-tools/mne-python #11532: [CI] Issue with new setuptools-scm for version configuration in mne-python

adam2392, 2023-03-05:
With https://github.com/mne-tools/mne-python/commit/3bf04edf744da89b1bb4f84cde5537758ed00865, there seems to be an issue when installing mne-python from GitHub with lines like:

```
python -m pip install --upgrade --progress-bar off --no-deps https://api.github.com/repos/mne-tools/mne-python/zipball/main
```

https://app.circleci.com/pipelines/github/mne-tools/mne-connectivity/644/workflows/914de5ee-39ab-4ca1-8111-0362aec121e8/jobs/701

Also noticed this issue in mne-bids. I presume this is not an issue with mne-python, but just tracking it here in case anyone knows an easy work around.

mscheltienne, 2023-03-06:
I had this issue too on another repository, `pip install git+https://github.com/mne-tools/mne-python` fixes the CI.

larsoner, 2023-03-06:
Thanks @mscheltienne I adapted your solution to a few other repos, so I'll close