mne-tools/mne-python #11540: mne.compute_rank does not return information for all valid channel types

tsbinns, 2023-03-07:
### Description of the problem

`mne.compute_rank` calls `mne.io.pick._picks_by_type` to find the channels to compute the rank for.

`_picks_by_type` only returns channels of types given in the variable `_DATA_CH_TYPES_SPLIT` which does not include types such as 'misc', 'emg', etc..., meaning information is not returned for all valid channel types.
https://github.com/mne-tools/mne-python/blob/3bf04edf744da89b1bb4f84cde5537758ed00865/mne/io/pick.py#L837-L862
https://github.com/mne-tools/mne-python/blob/3bf04edf744da89b1bb4f84cde5537758ed00865/mne/io/pick.py#L939-L943

If this behaviour is intended, perhaps it could at least be noted in the documentation of `compute_rank` that information will not be returned for certain channel types, or raised as a warning message that channels of valid types not marked as 'bads' were dropped.

### Steps to reproduce

```Python
import numpy as np
import mne

rand = np.random.RandomState(44)

n_channels = 3
n_times = 50
data = rand.random((n_channels, n_times))

info = mne.create_info(
    ch_names=[str(i) for i in range(n_channels)],
    sfreq=256,
    ch_types=['misc' for _ in range(n_channels)])

raw = mne.io.RawArray(data, info)

rank_dict = mne.compute_rank(raw)
rank_sum = sum(rank_dict.values())

assert np.linalg.matrix_rank(data) == n_channels  # expected rank (3)
```

### Expected results

rank_dict == {'misc': 3}
rank_sum == 3

### Actual results

rank_dict == {}
rank_sum = 0

### Additional information

Platform:         Windows-10-10.0.22621-SP0
Python:           3.11.0 | packaged by conda-forge | (main, Oct 25 2022, 06:12:32) [MSC v.1929 64 bit (AMD64)]
Executable:       C:\Users\User\anaconda3\envs\mne_ssd\python.exe
CPU:              Intel64 Family 6 Model 140 Stepping 1, GenuineIntel: 8 cores
Memory:           47.7 GB

mne:              1.4.dev0
numpy:            1.24.2 {MKL 2022.1-Product with 4 threads}
scipy:            1.10.0
matplotlib:       3.6.3 {backend=TkAgg}

sklearn:          1.2.1
numba:            Not found
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           1.5.3
pyvista:          0.38.1 {OpenGL 4.5.0 - Build 31.0.101.3889 via Intel(R) Iris(R) Xe Graphics}
pyvistaqt:        0.9.1
ipyvtklink:       Not found
vtk:              9.2.5
qtpy:             Not found
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

larsoner, 2023-03-07:
Yes I think we should document this better, only channels considered [data channels](https://mne.tools/stable/glossary.html#term-data-channels) should be included by default

agramfort, 2023-03-09:
I believe now that EMG is data :)

A

> Message ID: ***@***.***>
>

PR #12716 (2024-07-19): DOC: Update install docs for Spyder
The bulk of this has to do with us no longer packaging Spyder in our installers and adjusting the install docs (mostly) minimally to comply with this change. #10707 and #11615 for example ask for something more comprehensive and aren't meant to be addressed here. This does help a tiny bit (maybe!) with #10707 in the sense that it deduplicates some stuff. I also snuck in a couple of tiny single-digit-line changes to close three other issues:

1. Pinning `numpy<3`
2. Adding a note that `compute_rank` operates on data channels
3. Improving an error message

@ho [... cut]
closed 2024-07-19 by hoechenberger