mne-tools/mne-python #11543: Do not list not installed packages in `mne.sys_info()`

cbrnr, 2023-03-08:
We would like to see the output of `mne.sys_info()` in issue reports, but this list is very long – even when many packages are not installed it looks like this:

```
Platform:         macOS-13.2.1-arm64-arm-64bit
Python:           3.10.10 (v3.10.10:aad5f6a891, Feb  7 2023, 08:47:40) [Clang 13.0.0 (clang-1300.0.29.30)]
Executable:       /Users/clemens/.local/pipx/venvs/ipython/bin/python
CPU:              arm: 12 cores
Memory:           32.0 GB

mne:              1.4.dev0
numpy:            1.24.2 {OpenBLAS 0.3.21 with 12 threads}
scipy:            1.10.1
matplotlib:       3.7.0 {backend=MacOSX}

sklearn:          Not found
numba:            Not found
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           Not found
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             Not found
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.7.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found
```

I propose to exclude packages that are not installed, so the list would be much shorter:

```
Platform:         macOS-13.2.1-arm64-arm-64bit
Python:           3.10.10 (v3.10.10:aad5f6a891, Feb  7 2023, 08:47:40) [Clang 13.0.0 (clang-1300.0.29.30)]
Executable:       /Users/clemens/.local/pipx/venvs/ipython/bin/python
CPU:              arm: 12 cores
Memory:           32.0 GB

mne:              1.4.dev0
numpy:            1.24.2 {OpenBLAS 0.3.21 with 12 threads}
scipy:            1.10.1
matplotlib:       3.7.0 {backend=MacOSX}

pooch:            v1.7.0
```

agramfort, 2023-03-09:
maybe use a flat list for packages not installed? I think it is still
valuable

> Message ID: ***@***.***>
>

cbrnr, 2023-03-09:
Yes, that's probably a good idea – we've been discussing this directly in #11544.

PR #11568 (2023-03-17): Slightly rework `mne.sys_info()`
Fixes #11543, supersedes #11544.
closed 2023-03-17 by drammock