mne-tools/mne-python #11548: Calling maxwell_filter on Yokogawa recordings (from Brainstorm)

ftadel, 2023-03-10:
### Description of the problem

Problem initially posted on the Brainstorm and MNE-Python forums:
https://neuroimage.usc.edu/forums/t/tsss-with-yokogawa-data/38833
https://mne.discourse.group/t/calling-maxwell-filter-on-yokogawa-recordings-from-brainstorm/6452

I could reproduce the error on my end with an example dataset provided by the user.
I confirmed that the problem is not coming too obviously from Brainstorm in the creation of the RawArray object to be passed to maxwell_filter, as I can call other MNE methods on it.

### Steps to reproduce

```Python
import pickle 
import mne
filehandler = open('pyRaw10s.pickle', 'rb') 
raw = pickle.load(filehandler)
filehandler.close()
mne.preprocessing.maxwell_filter(raw, int_order=8, ext_order=3, origin='auto', coord_frame='head', regularize='in', ignore_ref=False, st_duration=10, st_correlation=0.98, st_fixed=True, st_only=False, mag_scale=100, skip_by_annotation=['edge','bad_acq_skip'])
```

### Link to data

Example dataset to reproduce the error (zipped pickle - first 10s of the Yokogawa .con file shared by the user):
https://www.dropbox.com/s/xd7o5m3fyyc5yc7/pyRaw10s.zip?dl=0

To get a link to download the original KIT recordings, please contact privately the owner of the dataset:
hannah.voss@charite.de

### Expected results

tSSS-cleaned data

### Actual results

Maxwell filtering raw data
    No bad MEG channels
    Processing 0 gradiometers and 128 magnetometers (of which 125 are actually KIT gradiometers)
<stdin>:1: RuntimeWarning: Only 5 head digitization points of the specified kind ("extra",), fitting may be inaccurate
<stdin>:1: RuntimeWarning: (X, Y) fit (-7.9, 42.7) more than 20 mm from head frame origin
    Automatic origin fit: head of radius 73.8 mm
    Using origin -7.9, 42.7, 5.3 mm in the head frame
    Processing data using tSSS with st_duration=10.0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<decorator-gen-457>", line 12, in maxwell_filter
  File "C:\Users\franc\AppData\Local\Programs\Python\Python310\lib\site-packages\mne\preprocessing\maxwell.py", line 353, in maxwell_filter
    raw_sss = _run_maxwell_filter(raw, **params)
  File "C:\Users\franc\AppData\Local\Programs\Python\Python310\lib\site-packages\mne\preprocessing\maxwell.py", line 555, in _run_maxwell_filter
    _get_this_decomp_trans(info['dev_head_t'], t=0.)
  File "C:\Users\franc\AppData\Local\Programs\Python\Python310\lib\site-packages\mne\preprocessing\maxwell.py", line 1012, in _get_decomp
    S_decomp, reg_moments, n_use_in = _regularize(
  File "<decorator-gen-460>", line 12, in _regularize
  File "C:\Users\franc\AppData\Local\Programs\Python\Python310\lib\site-packages\mne\preprocessing\maxwell.py", line 1066, in _regularize
    in_removes, out_removes = _regularize_in(
  File "C:\Users\franc\AppData\Local\Programs\Python\Python310\lib\site-packages\mne\preprocessing\maxwell.py", line 1947, in _regularize_in
    u, s, v = _safe_svd(this_S, full_matrices=False, **check_disable)
  File "C:\Users\franc\AppData\Local\Programs\Python\Python310\lib\site-packages\mne\fixes.py", line 84, in _safe_svd
    return linalg.svd(A, **kwargs)
  File "C:\Users\franc\AppData\Local\Programs\Python\Python310\lib\site-packages\scipy\linalg\_decomp_svd.py", line 133, in svd
    raise ValueError('illegal value in %dth argument of internal gesdd'
ValueError: illegal value in 4th argument of internal gesdd

### Additional information

Platform:         Windows-10-10.0.22621-SP0
Python:           3.10.10 (tags/v3.10.10:aad5f6a, Feb  7 2023, 17:20:36) [MSC v.1929 64 bit (AMD64)]
Executable:       C:\Users\franc\AppData\Local\Programs\Python\Python310\python.exe
CPU:              Intel64 Family 6 Model 154 Stepping 4, GenuineIntel: 12 cores
Memory:           15.4 GB

mne:              1.3.1
numpy:            1.23.5 {OpenBLAS 0.3.20 with 12 threads}
scipy:            1.10.1
matplotlib:       3.7.0 {backend=QtAgg}

sklearn:          1.2.1
numba:            0.56.4
nibabel:          5.0.1
nilearn:          0.10.0
dipy:             1.6.0
openmeeg:         Not found
cupy:             Not found
pandas:           1.5.3
pyvista:          0.38.3 {OpenGL 4.5.0 - Build 31.0.101.3616 via Intel(R) Iris(R) Xe Graphics}
pyvistaqt:        0.9.1
ipyvtklink:       Not found
vtk:              9.2.6
qtpy:             2.3.0 {PyQt5=5.15.2}
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.6.0

mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

larsoner, 2023-03-10:
Contacted for original data, hopefully can look next week!

larsoner, 2023-03-15:
FWIW I still need to look at the data, but based on this:
```
Using origin -7.9, 42.7, 5.3 mm in the head frame
```
that looks suspiciously like the Y and Z are swapped (e.g., something somewhere is not RAS). That seems too far forward (+Y) in the Neuromag head coordinate frame.

larsoner, 2023-03-17:
Okay it works with `ignore_ref=True` (here with default kwargs omitted for brevity):

```
with open('pyRaw10s.pickle', 'rb') as fid:
    raw = pickle.load(fid)
mne.preprocessing.maxwell_filter(raw, ignore_ref=True, st_correlation=0.98, verbose=True)
for dig, ident in zip(raw.info['dig'], (2, 1, 3)):
    dig['ident'] = ident
bem = mne.make_sphere_model(info=raw.info, r0=(0., 0., 0.04), head_radius=0.09)
trans = mne.transforms.Transform('head', 'mri', np.eye(4))
mne.viz.plot_alignment(raw.info, bem=bem, coord_frame='head', surfaces=('head', 'inner_skull'), trans=trans, dig=True, meg=('helmet', 'sensors', 'ref'))
```

<details>
<summary>Aside about digitization for @ftadel </summary>

One bug I noticed is that the `info['dig']` are incorrect -- CARDINAL should be POINT_LPA, POINT_NASION, or POINT_RPA but it's EXTRA:
```
>>> pprint.pprint(raw_2.info['dig'][:3])
[{'coord_frame': 4 (FIFFV_COORD_HEAD),
  'ident': 4 (FIFFV_POINT_EXTRA),
  'kind': 1 (FIFFV_POINT_CARDINAL),
  'r': array([2.77555756e-17, 9.81400000e-02, 7.70371978e-34])},
 {'coord_frame': 4 (FIFFV_COORD_HEAD),
  'ident': 4 (FIFFV_POINT_EXTRA),
  'kind': 1 (FIFFV_POINT_CARDINAL),
  'r': array([-7.73700000e-02, -2.12836621e-17, -1.15555797e-33])},
 {'coord_frame': 4 (FIFFV_COORD_HEAD),
  'ident': 4 (FIFFV_POINT_EXTRA),
  'kind': 1 (FIFFV_POINT_CARDINAL),
  'r': array([ 5.52600000e-02,  1.13498856e-17, -3.85185989e-34])}]
```
so I had to hack `plot_alignment` a bit for it to give me an alignment:
```
for dig, ident in zip(raw_2.info['dig'], (2, 1, 3)):
    dig['ident'] = ident
```

</details>

and this is the resulting `plot_alignment` for what worked:

[image]

Maybe this is a reasonable alignment reflective of the measurement setup, maybe not (as the warning suggests, I'm suspicious of that head origin...).

It looks like this is why the reference sensors need to be ignored:
```
>>> raw_2.info['chs'][mne.pick_types(raw_2.info, ref_meg=True)[0]]['loc']
array([-0.0037978 ,  0.01688092,  0.03891309,  0.        ,  0.        ,
        0.        ,         nan,         nan,         nan,         nan,
               nan,         nan])
>>> raw_2.info['chs'][mne.pick_types(raw_2.info, ref_meg=True)[0]]['ch_name']
'M000'
```
The first reference sensor at least has a position in X/Y/Z space (first 3 elements) but no coil rotation EX/EY/EZ (the next 9 elements being zero or nan). We should probably improve our checks in MNE-Python that the coils are defined properly to improve the cryptic GESDD error message, but I think this points to a problem with the channel info population at the brainstorm end (?).

ftadel, 2023-03-20:
Thank you for this thorough report!
I'll fix all this on the Brainstorm side.

Regarding the head position, it looks like this on the Brainstorm side:
[image]  [image]

If the coordinates of the sensors and fiducials are incorrectly converted from Brainstorm to MNE-Python, but all in the same way (e.g. an incorrect rotation-translation was uniformly applied to all the 3D coordinates): can this cause any problem in the SSS/tSSS computation? 
Does this process require the sensors and fiducials to be in the Neuromag coordinate system (or in another coordinate system, but correctly documented in the Info structure)?

ftadel, 2023-03-20:
> The first reference sensor at least has a position in X/Y/Z space (first 3 elements) but no coil rotation EX/EY/EZ (the next 9 elements being zero or nan).

In this Yokogawa file, as read by the Yokogawa library in Matlab, three channels (126, 127, 128) are tagged as reference magnetometers but have no name and no 3D positions.
I fixed the Brainstorm reader to consider that these three channels are NOT MEG reference sensors.

Calling maxwell_filter now works, without any modification to the conversion function (from Brainstorm structure to MNE-Python object).