mne-tools/mne-python #11551: Consider removing some badges

cbrnr, 2023-03-14:
Currently, we have two rows of badges at the top of our README.rst:

[image]

Do we really need the first row? This only shows build results from various workflows, which is not that interesting for users. Developers can look in other places (PRs) to check if our workflows are passing or not. I think removing the first row would make the document less crowded and more pleasant to view. WDYT?

agramfort, 2023-03-14:
maybe just move the first line below?

> Message ID: ***@***.***>
>

cbrnr, 2023-03-14:
> maybe just move the first line below?

Sure, that would also work (maybe to the "Contributing" section?). I still don't find it that informative, because each badge has a different number of independent workflows associated with it and tests various aspects of our codebase (and I'm not sure if we have badges to reflect *all* of our jobs). So I still have to look somewhere else if I want to know exactly what's going on.

larsoner, 2023-03-14:
Indeed NumPy and Pandas don't have CI badges other than codecov, which we could keep

[image]

[image]

So +1 for just removing the CI ones.

We could also consider adding some of the other ones, especially a discourse one (inspired by `gitter` being in the Pandas list). If they don't provide one, we should maybe make one with [anybadge](https://github.com/jongracecox/anybadge) or something similar!

drammock, 2023-03-14:
IMO the audience for those badges is "devs who don't currently work on MNE".  For example if I need to implement some algorithm and I search first to see if someone else already did it, one of the things I look at is "last updated" and another is badges (if they have any).

I'm not sure if that changes anyone's mind though... maybe it supports @larsoner's point that just keeping `codecov` is enough?  I currently don't ever look at our badges to get useful information and I agree that for it to be really useful *to us* the CI badges would have to be a complete set of everything we run... if we keep them it would also be better if "circleci" badge said something like "docs build: passing" IMO

larsoner, 2023-03-14:
Also sometimes they show up as red when we have a spurious failure on `main`, which isn't good. So removing them removes some flakiness / unreliability in our docs/PyPI page

PR #11558 (2023-03-15): Clean up badges
Fixes #11551. I've removed all CI badges, kept Codecov, and added a badge for our Discourse forum.