mne-tools/mne-python #11552: Unable to use RawEDF normally

myh1041893567, 2023-03-14:
### Description of the problem

During the model experiment, the first prompt is that the channel read from the edf file is different from the setting. This query found that a RawEDF cannot be used normally. The data file can be opened with the EDFbrowser file.
2023-03-14 11:26:59,598 INFO : 29 files in total.
2023-03-14 11:26:59,598 INFO : Read recording lengths...
2023-03-14 11:26:59,616 INFO : Load 1 of 29
2023-03-14 11:26:59,848 INFO : Load data...
Reading 0 ... 301249  =      0.000 ...  1204.996 secs...
Traceback (most recent call last):
  File "E:\EEG\eeg1\auto-eeg-diagnosis-example-master\auto_diagnosis.py", line 348, in <module>
    exp = run_exp(
  File "E:\EEG\eeg1\auto-eeg-diagnosis-example-master\auto_diagnosis.py", line 168, in run_exp
    X,y = dataset.load()
  File "E:\EEG\eeg1\auto-eeg-diagnosis-example-master\dataset.py", line 201, in load
    x = load_data(fname, preproc_functions=self.preproc_functions,
  File "E:\EEG\eeg1\auto-eeg-diagnosis-example-master\dataset.py", line 126, in load_data
    assert np.array_equal(cnt.ch_names, selected_ch_names)
AssertionError

Unable to get repr for <class 'mne.io.edf.edf.RawEDF'>
'RawEDF' object has no attribute '_annotations'
'RawEDF' object has no attribute 'info'

### Steps to reproduce

```Python
Recurrence model link：https://github.com/robintibor/auto-eeg-diagnosis-example
```

### Link to data

https://isip.piconepress.com/projects/tuh_eeg/html/downloads.shtml

### Expected results

You want to get the response

### Actual results

D:\Anaconda\python.exe E:/EEG/eeg1/auto-eeg-diagnosis-example-master/auto_diagnosis.py
2023-03-14 11:26:59,580 INFO : Read file names
2023-03-14 11:26:59,580 INFO : Reading E:\EEG\file\text\edf\train\...
2023-03-14 11:26:59,598 INFO : .. 29 files.
2023-03-14 11:26:59,598 INFO : 29 files in total.
2023-03-14 11:26:59,598 INFO : Read recording lengths...
2023-03-14 11:26:59,616 INFO : Load 1 of 29
2023-03-14 11:26:59,848 INFO : Load data...
Reading 0 ... 301249  =      0.000 ...  1204.996 secs...
Traceback (most recent call last):
  File "E:\EEG\eeg1\auto-eeg-diagnosis-example-master\auto_diagnosis.py", line 348, in <module>
    exp = run_exp(
  File "E:\EEG\eeg1\auto-eeg-diagnosis-example-master\auto_diagnosis.py", line 168, in run_exp
    X,y = dataset.load()
  File "E:\EEG\eeg1\auto-eeg-diagnosis-example-master\dataset.py", line 201, in load
    x = load_data(fname, preproc_functions=self.preproc_functions,
  File "E:\EEG\eeg1\auto-eeg-diagnosis-example-master\dataset.py", line 126, in load_data
    assert np.array_equal(cnt.ch_names, selected_ch_names)
AssertionError

### Additional information

Microsoft Windows [版本 10.0.22000.1574]
Python 3.9.12
AMD Ryzen 7 5800H with Radeon Graphics
Package                       Version
----------------------------- --------------------
aiohttp                       3.8.1
aiosignal                     1.2.0
alabaster                     0.7.12
anaconda-client               1.9.0
[... 318 lines cut ...]
typing_extensions             4.1.1
ujson                         5.1.0
Unidecode                     1.2.0
urllib3                       1.26.9
w3lib                         1.21.0
watchdog                      2.1.6
wcwidth                       0.2.5
webencodings                  0.5.1
websocket-client              0.58.0
Werkzeug                      2.0.3
wheel                         0.37.1
widgetsnbextension            3.5.2
win-inet-pton                 1.1.0
win-unicode-console           0.5
wincertstore                  0.2
wrapt                         1.12.1
xarray                        0.20.1
xlrd                          2.0.1
XlsxWriter                    3.0.3
xlwings                       0.24.9
yapf                          0.31.0
yarl                          1.6.3
zict                          2.0.0
zipp                          3.7.0
zope.interface                5.4.0

cbrnr, 2023-03-14:
Can you please post code which shows the problem using MNE functions only? I.e. `raw = mne.io.read_raw_edf("which-file-name.edf")`.

myh1041893567, 2023-03-14:
> 您能否发布仅使用MNE函数显示问题的代码？即.`raw = mne.io.read_raw_edf("which-file-name.edf")`
[image]
Thank you very much for your timely response
import mne

# 加载EDF文件
raw = mne.io.read_raw_edf("E:\\EEG\\file\\text\\edf\\train\\abnormal\\000\\00000077\\s003_2010_01_21\\00000077_s003_t000.edf")

# 打印一些数据信息
print(raw.info)

Python 3.9.12 (main, Apr  4 2022, 05:22:27) [MSC v.1916 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.2.0 -- An enhanced Interactive Python. Type '?' for help.
PyDev console: using IPython 8.2.0
Python 3.9.12 (main, Apr  4 2022, 05:22:27) [MSC v.1916 64 bit (AMD64)] on win32
runfile('E:/EEG/eeg1/auto-eeg-diagnosis-example-master/texi1.py', wdir='E:/EEG/eeg1/auto-eeg-diagnosis-example-master')
Extracting EDF parameters from E:\EEG\file\text\edf\train\abnormal\000\00000077\s003_2010_01_21\00000077_s003_t000.edf...
EDF file detected
Setting channel info structure...
Creating raw.info structure...
<Info | 7 non-empty values
 bads: []
 ch_names: EEG FP1-REF, EEG FP2-REF, EEG F3-REF, EEG F4-REF, EEG C3-REF, ...
 chs: 36 EEG
 custom_ref_applied: False
 highpass: 0.0 Hz
 lowpass: 125.0 Hz
 meas_date: 2010-01-21 13:58:36 UTC
 nchan: 36
 projs: []
 sfreq: 250.0 Hz
>

cbrnr, 2023-03-14:
Can you also share the data file you are using in your example?

myh1041893567, 2023-03-14:
> 您还可以共享您在示例中使用的数据文件吗？
Sorry, I just went out. You can turn to this link
https://isip.piconepress.com/projects/tuh_eeg/downloads/tuh_eeg/v2.0.0/edf/000/aaaaaaaa/s001_2015_12_30/01_tcp_ar/

cbrnr, 2023-03-14:
It asks me for a username and password.

myh1041893567, 2023-03-14:
> It asks me for a username and password.
Sorry, could you provide me with an email? My email is sent to you

cbrnr, 2023-03-15:
The license agreement explicitly states that "The data recipient will not release data to a third party. Please have the third party contact us directly by sending an email to help@nedcdata.org". So unless you are knowingly going to violate that agreement, I guess we have to find a different way to solve this issue.

myh1041893567, 2023-03-15:
> The license agreement explicitly states that "The data recipient will not release data to a third party. Please have the third party contact us directly by sending an email to [help@nedcdata.org](mailto:help@nedcdata.org)". So unless you are knowingly going to violate that agreement, I guess we have to find a different way to solve this issue.

Thank you for your reply and help. If you have any need, please register. It's very simple. Thank you very much

myh1041893567, 2023-03-15:
> > 感谢您的回复和帮助。如果您有任何需要，请注册。这很简单。谢谢
> 
> 我真的不想用我的签名向他们发送此表单，也不想注册他们的邮件列表服务器（自动发生）。我不知道任何其他解决方案，也许可以编写它们并解释您需要私下共
[data.docx](https://github.com/mne-tools/mne-python/files/10980948/data.docx)
享一个数据集，以便我们可以调试问题？

[data.docx](https://github.com/mne-tools/mne-python/files/10980947/data.docx)
Do you think you can use this?

myh1041893567, 2023-03-16:
> 对不起，这不起作用。

I put an edf file in this document

cbrnr, 2023-03-16:
Yes, I can see it, but I don't know how to save/export it.

myh1041893567, 2023-03-16:
> Yes, I can see it, but I don't know how to save/export it.

After downloading the document, can't the EDF file in it be pulled out?

cbrnr, 2023-03-16:
> After downloading the document, can't the EDF file in it be pulled out?

No, it doesn't work for me (maybe because I have a Mac).

myh1041893567, 2023-03-16:
> > 下载文档后，里面的EDF文件不能拉出来吗？
> 
> 不，它对我不起作用（也许是因为我有一台 Mac）。

How do you usually upload a small number of edf files

cbrnr, 2023-03-16:
Can you share a Google Drive (or OneDrive, DropBox) link?

myh1041893567, 2023-03-16:
Recurrence model：https://github.com/robintibor/auto-eeg-diagnosis-example

myh1041893567, 2023-03-16:
> 好的，我可以读取文件：
> 
> ```python
> raw = mne.io.read_raw_edf("/Users/clemens/Downloads/00000077_s003_t000.edf", preload=True)
> ```
> 
> 问题出在哪里？
"I am reproducing a simple model that always prompts me to select channels. The channels read from the edf file are inconsistent, and RawEDF cannot be used during debugging."

cbrnr, 2023-03-16:
Can you post code that lets me reproduce this issue? What are the values of `cnt.ch_names` and `selected_ch_names` in your example?

drammock, 2023-03-16:
> [image]

this entire traceback is within the `auto-eeg-diagnosis-example` namespace. None of the error is coming from within MNE-Python nor does it involve MNE functions or methods. After a quick look at that repo, I suspect (but I'm not sure) that there's a difference between how your channels are named, and the channel names that they hardcode here:

https://github.com/robintibor/auto-eeg-diagnosis-example/blob/9c9b130a3bc724fb74133cff8ce834a9972ba56a/dataset.py#L105-L107

If you want additional support I suggest you reach out to the authors of that repo.

closed 2023-03-16 by drammock

myh1041893567, 2023-03-17:
> 
Thank you very much for your answer, and I will try to contact the repository author

myh1041893567, 2023-03-17:
> > [image]
> 
> 整个回溯都在命名空间内。这些错误都不是来自MNE-Python内部，也不涉及MNE函数或方法。快速浏览该存储库后，我怀疑（但我不确定）您的频道的命名方式与他们在此处硬编码的频道名称之间存在差异：`auto-eeg-diagnosis-example`
> 
> https://github.com/robintibor/auto-eeg-diagnosis-example/blob/9c9b130a3bc724fb74133cff8ce834a9972ba56a/dataset.py#L105-L107
> 
> 如果您需要其他支持，我建议您联系该存储库的作者。
However, a prompt indicating that the method is not available appears during debugging
[image]

myh1041893567, 2023-03-17:
> 你能发布让我重现这个问题的代码吗？在您的示例中，和的值是什么？`cnt.ch_names``selected_ch_names`
https://github.com/robintibor/auto-eeg-diagnosis-example/tree/9c9b130a3bc724fb74133cff8ce834a9972ba56a
This is the case with the code because the RawEDF method is unavailable. I did not find the channel name extracted from the data file during debugging, but the channel names extracted separately are consistent. My only change is this
[image]

drammock, 2023-03-23:
As I said before, the traceback occurs entirely within the `auto-eeg-diagnosis-example` namespace. More concretely, the traceback starts here:

https://github.com/robintibor/auto-eeg-diagnosis-example/blob/9c9b130a3bc724fb74133cff8ce834a9972ba56a/dataset.py#L126

The part where MNE is involved is when the file is loaded, about 25 lines earlier:

https://github.com/robintibor/auto-eeg-diagnosis-example/blob/9c9b130a3bc724fb74133cff8ce834a9972ba56a/dataset.py#L99-L102

I can't tell just by looking at your screenshots what the real problem is, but all signs indicate that this is not an MNE-Python bug: as @cbrnr [indicated above](https://github.com/mne-tools/mne-python/issues/11552#issuecomment-1472090959), the file is readable by MNE.  If you're convinced that MNE is doing something incorrectly, please open a new issue that *does not use the `auto-eeg-diagnosis-example` code, only MNE-Python commands* and shows that the RawEDF object is somehow problematic.