mne-tools/mne-python #11553: BUG: no sensor information after importing elp and hsp data

onelyn, 2023-03-14:
Hello all! I'm looking for help with meg data preprocessing.

My raw data is .con file, and I have .mrk file (marker) and also .pos file (elp and hsp data).

If I read the data using `mne.io.read_raw_kit(data_raw_file)`, then there is correct sensor information. But as the headshape is not digitized, the following analysis can not be applied.

However, if I read the data using `mne.io.read_raw_kit(data_raw_file, mrk=marker_data, elp=elp_data, hsp=hsp_data)`, then there will be no sensor information. And I can plot only one dot when calling `raw.plot_sensors()`.

I'm wondering what's wrong with this process. Why there's a lack of sensor information if I import elp and hsp data? 

I'd appreciate it if anyone could help. Thanks a lot!

[image]
[image]

onelyn, 2023-03-14:
And here are the marker, elp, and hsp data. They are all in array format. The marker is (5,3) array. The elp is (8,3) array. The hsp is (4299,3) array. 

[image]

drammock, 2023-03-14:
> If `mrk`, `hsp` or `elp` are :term:`array_like` inputs, then the
> numbers in xyz coordinates should be in units of meters.

Exactly right. In fact, the plot that shows "one dot" actually shows all the sensors very tightly clustered together, I think, because HSP and ELP were given in mm but MNE treats it as m so the head outline is 1000 times too big

closed 2023-03-14 by drammock