mne-tools/mne-python #11559: Remove `jinja2` from core dependencies?

cbrnr, 2023-03-15:
Currently, we require `jinja2` even though it is imported in only one location (`mne/html_templates/_templates.py`). Although this is a lightweight package (it only depends on `MarkupSafe`), I'd argue that it should not be a core requirement.

larsoner, 2023-03-15:
AFAIK It's lightweight, pure python, well supported, and hasn't caused us any maintenance issues in the years we've had it as a core dependency. So I don't see much gain to making it optional. I do see some drawbacks, like I think Report week no longer just work, as well as possibly some `_repr_html_`, and we need to protect imports and add more test decorators. So overall -1 on making optional.

I'd rather gradually go on the opposite direction, e.g., add things like nibabel that also fit the above description/ attributes. It'll simplify our codebase not to have to nest the import.

> Also, does [our test to detect nested imports](https://github.com/cbrnr/mne-python/blob/8b9baebdf1bb9d476e949262e594250132ef2189/mne/tests/test_import_nesting.py#L30) need to be updated?

No I don't think so. That mostly has to do two things 1) check that only required dependencies are imported and 2) that slow parts of required ones are not imported. I don't think importing jinja2 takes long so no need to nest it

larsoner, 2023-03-15:
> why would we test that its import is nested (even though it's not imported with import mne)?

We must nest it to help ensure that `import mne` stays fast.

Okay to close?

cbrnr, 2023-03-15:
OK. Then I'll add a bit of an explanation to this test.

closed 2023-03-15 by cbrnr

larsoner, 2023-03-15:
(I think the import nesting is effectively achieved one level higher, when the templates themselves are loaded to be used)

cbrnr, 2023-03-15:
It's still hard to understand which packages we are allowed to import directly (and which ones not), and how all of this is related to our core and optional dependencies.

cbrnr, 2023-03-15:
If you have time, can you clarify this in the docstring of the test? I have too many questions about this to do it myself, but I think it would be helpful for (future) developers.