mne-tools/mne-python #11560: interpolate_bads did not work

onelyn, 2023-03-15:
### Description of the problem

Hello all! I have a problem interpolating bad channels of MEG data.

Here is my MEG data description:
[image]

I need to interpolate the MEG 079 channel based on other channels.

The original data looks like this:
[image]

After I called the `meg_data.copy().interpolate_bads(reset_bads=False)`, it seemed that the MEG 079 channel was not changed.
[image]
[image]

I’m wondering what’s wrong during the process. I’d appreciate it if anyone could help! Thanks a lot!

### Steps to reproduce

```Python
raw.info['bads']=['MEG 079']
raw.load_data()
meg_data = raw.copy().filter(l_freq=1, h_freq=40)
meg_data_interp = meg_data.copy().pick_types(meg=True, exclude=[])
meg_data_interp = meg_data_interp.interpolate_bads(reset_bads=False)
picks = mne.pick_channels(meg_data_interp.ch_names, include=['MEG 078','MEG 079','MEG 080'])
meg_data_interp.plot(duration=12, order=picks, n_channels=len(picks))
```

### Expected results

interpolated data of MEG 079 channel

### Actual results

no change to the data

### Additional information

I'm using the data from https://osf.io/ag3kj/
I'm using sub-01_ses-1_task-1_meg.con and corresponding marker, elp, and hsp files

mne.sys_info():
[image]

onelyn, 2023-03-15:
I asked in the MNE Forum but the problem has not been solved. Here is the link:
https://mne.discourse.group/t/interpolate-bads-did-not-work/6531/9

Thanks a lot if anyone could help!

cbrnr, 2023-03-15:
You are using an old MNE version, can you update to the latest version and see if the problem still persists?

onelyn, 2023-03-15:
> 

No, the problem still exists.
[image]

I'm now working with the newest version.
[image]

And I found that it seems the interpolation was done, but made the initial values of channel 079 much smaller (the initial values of this channel was smaller than other channels but not zero).
Here is the initial values of channel 079:
[image]
And this is values after interpolation:
[image]
This is another channel (channel 078):
[image]

Do you know why the interpolation works like this? Thanks a lot!

onelyn, 2023-03-15:
> 

Sadly, the interpolation seemed to fail again. 
[image]

(Can it be because I did not regress the data based on the reference sensors at first? But I did not see this step in the tutorial. )

onelyn, 2023-03-15:
I may provide more information for this problem (I'm afraid it's because I'm doing something wrong before interpolation).

My data is from https://osf.io/ag3kj/. Currently the sample data for this problem is sub-01_ses-1_task-1. And following is all my codes:

```
import mne
import numpy as np
import pandas as pd
from mne.preprocessing import (ICA, corrmap, create_ecg_epochs, create_eog_epochs)

data_folder = '/bigvault/Projects/asymmetry_lang/MEG/sub-01/ses-1/meg'
data_raw_file = data_folder+'/sub-01_ses-1_task-1_meg.con'
marker_file_name = data_folder+'/sub-01_ses-1_task-1_markers.mrk'
marker_data = mne.io.kit.read_mrk(marker_file_name)
elp_file_name = data_folder+'/sub-01_ses-1_acq-ELP_headshape.pos'
hsp_file_name = data_folder+'/sub-01_ses-1_acq-HSP_headshape.pos'

def readPos(filename, start_line, point_num):
    df = pd.read_table(filename, header=None)
    line_num = len(df.values)-3
    save_data_row = []
    item = ''
    indicate_num = 0
    flag = False
    for line_index, m in enumerate(df.values[start_line:]):
        save_data_line = []
        for idx, n in enumerate(df.values[start_line+line_index][0].replace(' ','')):
            if flag:
                indicate_num += 1
            item += n
            if indicate_num == point_num:
                save_data_line.append(float(item))
                item = ''
                indicate_num = 0
                flag = False
            if n == '.':
                flag = True
        save_data_row.append(save_data_line)
    return np.array(save_data_row)

elp_data = readPos(elp_file_name, 3, 4)
hsp_data = readPos(hsp_file_name, 4, 2)

raw = mne.io.read_raw_kit(data_raw_file, mrk=marker_data/1000, elp=elp_data/1000, hsp=hsp_data/1000)
raw.info['bads']=['MEG 079']
raw.load_data()

meg_data = raw.copy().filter(l_freq=1, h_freq=40)
ref_channels = ['MEG 209','MEG 210','MEG 211','MEG 212','MEG 213','MEG 214','MEG 215','MEG 216',
               'MEG 217','MEG 218','MEG 219','MEG 220','MEG 221','MEG 222','MEG 223','MEG 224']
meg_data_interp = meg_data.copy()
meg_data_interp = meg_data_interp.interpolate_bads(reset_bads=False, exclude=ref_channels)
picks = mne.pick_channels(meg_data_interp.ch_names, include=['MEG 078','MEG 079','MEG 080'])
meg_data_interp.plot(duration=12, order=picks, n_channels=len(picks))
```

cbrnr, 2023-03-15:
I think it's not a good idea to discuss this issue in both our Discourse forum and here. Would it be OK if I closed this issue and we continue on Discourse? If it turns out that this is indeed a bug, we can always re-open this issue.

onelyn, 2023-03-15:
> I think it's not a good idea to discuss this issue in both our Discourse forum and here. Would it be OK if I closed this issue and we continue on Discourse? If it turns out that this is indeed a bug, we can always re-open this issue.

Sure. Thanks a lot for your time and help!

cbrnr, 2023-03-15:
Great! Continuing the discussion here: https://mne.discourse.group/t/interpolate-bads-did-not-work/6531

closed 2023-03-15 by cbrnr