mne-tools/mne-python #11576: sys_info fails to find mne-qt-browser

drammock, 2023-03-20:
It runs on my laptop but it took about a minute to print.

Also I get this

```
❯ mne sys_info
Platform:         macOS-13.2.1-arm64-arm-64bit
Python:           3.10.8 | packaged by conda-forge | (main, Nov 22 2022, 08:25:29) [Clang 14.0.6 ]
Executable:       /Users/agramfort/miniconda3/envs/mne/bin/python3.10
CPU:              arm (10 cores)
Memory:           32.0 GB

Core
├☑ mne:              1.4.dev0
├☑ numpy:            1.23.5 (OpenBLAS 0.3.21 with 10 threads)
├☑ scipy:            1.10.0
├☑ matplotlib:       3.7.0 (backend=MacOSX)
├☑ pooch:            1.6.0
└☑ jinja2:           3.1.2

Numerical (optional)
├☑ sklearn:          1.2.0
├☑ numba:            0.56.4
├☑ nibabel:          5.0.0
├☑ nilearn:          0.10.0
├☑ dipy:             1.5.0
├☑ openmeeg:         2.5.5
├☑ pandas:           1.5.3
└☐ unavailable:      cupy

Visualization (optional)
├☑ pyvista:          0.37.0 (OpenGL 4.1 Metal - 83 via Apple M1 Pro)
├☑ pyvistaqt:        0.9.0
├☑ ipyvtklink:       0.2.2
├☑ vtk:              9.2.5
├☑ qtpy:             2.3.0 (PyQt5=5.15.6)
├☑ ipympl:           0.9.2
├☑ pyqtgraph:        0.13.1
└☐ unavailable:      mne-qt-browser

Ecosystem (optional)
├☑ mne_bids:         0.13.dev0
├☑ mne_nirs:         0.6.dev0
└☐ unavailable:      mne_features, mne_connectivity, mne_icalabel
❯ which mne
/Users/agramfort/miniconda3/envs/mne/bin/mne
❯ mne browse_raw ~/mne_data/MNE-sample-data/MEG/sample/sample_audvis_raw.fif
❯ conda list | grep mne-qt
mne-qt-browser            0.4.0           py310hbe9552e_0    conda-forge
```

you can see that mne-qt-browser is installed and works but the new `mne sys_info` does not detect this. To me it's a bug...

_Originally posted by @agramfort in https://github.com/mne-tools/mne-python/issues/11568#issuecomment-1475991673_

cbrnr, 2023-03-20:
I can confirm that `mne-qt-browser` does not show up (because we should replace `-` with `_`). I can do that.

I cannot confirm the long execution time though. On my laptop, it's almost instantaneous (like it was before this change).

agramfort, 2023-03-20:
what takes the most time for me is the Visualization block. but it runs
faster the second time apparently...

> Message ID: ***@***.***>
>

agramfort, 2023-03-20:
I mean the block Visualization printed in sys_info.

> Message ID: ***@***.***>
>

cbrnr, 2023-03-20:
> I mean the block Visualization printed in sys_info.

Yes I know – when this block is being processed, the MNE logo briefly shows up in my Dock. Or at least it used to, I think it depends on if Qt bindings are installed or not.

closed 2023-03-20 by larsoner

cbrnr, 2023-03-20:
> I don't get this on Linux at least. We could look into trying to fix it (I assume for macOS?) but it might not be easy/doable

Yes, on macOS. I guess if we add no splash options to all our GUIs (and use those in `sys.info()` that issue might resolve itself.

PR #11577 (2023-03-20): Fix bug in new `mne.sys_info()`
Fixes #11576. I'm replacing `-` with `_` at in the import call, and for a better look I think we should use `-` in the output. I've also removed the colons after the package names (which @hoechenberger suggested previously but we forgot) and fixed the alignment.