mne-tools/mne-python #11588: Add short documentation for `RawEDF`?

DimitriPapadopoulos, 2023-03-23:
### Proposed documentation enhancement

Function [`mne.io.read_raw_edf()`](https://mne.tools/stable/generated/mne.io.read_raw_edf.html) returns an instance of **`RawEDF`**, but as far as I can see this class is not documented.

Since **`RawEDF`** inherits base class **`BaseRaw`**, it might be sufficient to point to the [`mne.io.BaseRaw`](https://mne.tools/stable/generated/mne.io.BaseRaw.html#mne-io-baseraw) documentation.

DimitriPapadopoulos, 2023-03-23:
Class **`RawEDF`** is actually documented in the source code:

https://github.com/mne-tools/mne-python/blob/603a96da58ddba3c2c05d3a01639abd669cc9543/mne/io/edf/edf.py#L47-L54

But I cannot find the actual documentation using the search function:

[image]

DimitriPapadopoulos, 2023-03-23:
Also:
* The class is missing from the [Python API Reference](https://mne.tools/dev/python_reference.html).
* The URL that should in theory display the **`RawEDF`** documentation returns a 404:
  https://mne.tools/dev/generated/mne.io.RawEDF.html

larsoner, 2023-03-23:
Yes that's why -- people should look at `Raw` or `BaseRaw` for methods, so I'll close

closed 2023-03-23 by larsoner

larsoner, 2023-03-23:
I guess we could. I'd go for something like `The raw instance. See :class:`mne.io.BaseRaw` for details.` or something

DimitriPapadopoulos, 2023-03-23:
I suggest:

> The raw instance. See base class [`mne.io.BaseRaw`](https://mne.tools/stable/generated/mne.io.Raw.html#mne.io.Raw) for documentation of attribute and methods.

This would match the [`mne.io.BaseRaw`](https://mne.tools/stable/generated/mne.io.BaseRaw.html) documentation itself:
[image]

Perhaps also add a link in the **See also** section of [`mne.io.read_raw_edf()`](https://mne.tools/stable/generated/mne.io.read_raw_edf.html)?

larsoner, 2023-03-23:
Yes but `BaseRaw` doesn't actually document any of the methods, just lists them. I'd rather link to `Raw` directly since it contains the method docs

DimitriPapadopoulos, 2023-03-23:
Will do.

But then, the detailed documentation of methods could be moved from `Raw` to `BaseRaw`, couldn't it? I mean, does `Raw` contain the detailed documentation for historical reasons, or for real design reasons?

larsoner, 2023-03-23:
Historical mostly I think. Probably not worth changing it to `BaseRaw` for now