mne-tools/mne-python #11593: Strange Unicode characters in `.github/config.yml`

DimitriPapadopoulos, 2023-03-23:
### Proposed documentation enhancement

File [`.github/config.yml`](https://github.com/mne-tools/mne-python/blame/main/.github/config.yml) contains some strange Unicode characters. While my browser or editor do not display them, they do appear with a `cat` in the console:

[image]

I don't think they were inserted on purpose. Is it OK to remove them?

drammock, 2023-03-23:
those are unicode combining characters. They are supposed to render as a single glyph (in this case "man riding bicycle" probably).  I see no harm in making these emoji gender-unspecified.

drammock, 2023-03-23:
you can look up the codepoints:

```py
In [1]: glyph = "🚴🏽‍♂️"
In [2]: glyph.encode('raw_unicode_escape').decode()
Out[2]: '\\U0001f6b4\\U0001f3fd\\u200d\\u2642\\ufe0f'
```

PR #11594 (2023-03-23): DOC: Get rid of garbage Unicode characters
#### Reference issue

Fixes #11593.

#### What does this implement/fix?

Remove garbage Unicode characters that are not displayed by my editor or browser.
closed 2023-03-23 by larsoner