mne-tools/mne-python #11598: Get rid of `# -*- coding: utf-8 -*-`?

DimitriPapadopoulos, 2023-03-26:
### Description of the problem

UTF-8 is implicit in Python 3.

Is it OK to remove line `# -*- coding: utf-8 -*-` from Python files?

### Steps to reproduce

```Python
$ grep -R 'coding: utf-8'
tutorials/epochs/20_visualize_epochs.py:# -*- coding: utf-8 -*-
tutorials/epochs/15_baseline_regression.py:# -*- coding: utf-8 -*-
tutorials/epochs/10_epochs_overview.py:# -*- coding: utf-8 -*-
tutorials/epochs/60_make_fixed_length_epochs.py:# -*- coding: utf-8 -*-
...
```
```

### Expected results

Remove `# -*- coding: utf-8 -*-` which should be implicit in Python 3 files.

### Actual results

Explicit `# -*- coding: utf-8 -*-` in Python files.

### Additional information

-

cbrnr, 2023-03-26:
Fine with me.

agramfort, 2023-03-26:
works for me

PR #11599 (2023-03-27): MAINT: `coding: utf-8` is implicit in Python 3
#### Reference issue

Fixes #11598.

#### What does this implement/fix?

In Python 3, `# -*- coding: utf-8 -*` is implicit, no need to add it explicitly.

One could argue this might be useful for tools such as editors. However, tools that don't interpret Python source code as UTF-8 by default in 2023 shouldn't be used.

#### Additional information

See [PEP 3120 – Using UTF-8 as the default source encoding](https://peps.python.org/pep-3120/).
closed 2023-03-27 by drammock