mne-tools/mne-python #11602: EDF / EDF+ header I/O

DimitriPapadopoulos, 2023-03-27:
### Description of the problem

The header I/O code in [`mne.io.edf._read_edf_header()`](https://github.com/mne-tools/mne-python/blob/main/mne/io/edf/edf.py#L635) does not look entirely correct to me.

### Patient birthdate in 'local patient identification' field

According to [2.1.3. Additional specifications in EDF+](https://www.edfplus.info/specs/edfplus.html#additionalspecs):
> birthdate in dd-MMM-yyyy format using the English 3-character abbreviations of the month in capitals.

Therefore, this code is incorrect:
https://github.com/mne-tools/mne-python/blob/84259a673caafc3dfb37e07e9d73a3593c5c6acf/mne/io/edf/edf.py#L651
Indeed, `%b` stands for _Month as locale’s abbreviated name_, so it doesn't fit the _English 3-character abbreviations of the month_. It is perfectly OK to support locale month names for leniency and backwards compatibility, but the code should mainly support `JAN`, `FEB`, `MAR` and so on.

### Check for 'Startdate' in 'local recording identification'

According to [2.1.3. Additional specifications in EDF+](https://www.edfplus.info/specs/edfplus.html#additionalspecs):
> 4. The 'local recording identification' field must start with the subfields (subfields do not contain, but are separated by, spaces):
>    - The text 'Startdate'.

The following code should check for `Startdate`:
https://github.com/mne-tools/mne-python/blob/84259a673caafc3dfb37e07e9d73a3593c5c6acf/mne/io/edf/edf.py#L662

Typically:
```python
        if len(rec_info) == 5 and rec_info[0] == 'Startdate':
```
The code could only emit a warning for leniency and backwards compatibility.

### Startdate in 'local recording identification' field

Same as the birthdate above. According to [2.1.3. Additional specifications in EDF+](https://www.edfplus.info/specs/edfplus.html#additionalspecs):
> The startdate itself in dd-MMM-yyyy format using the English 3-character abbreviations of the month in capitals.

Therefore, this code is incorrect:
https://github.com/mne-tools/mne-python/blob/84259a673caafc3dfb37e07e9d73a3593c5c6acf/mne/io/edf/edf.py#L664
Indeed, `%b` stands for _Month as locale’s abbreviated name_, so it doesn't fit the _English 3-character abbreviations of the month_. It is perfectly OK to support locale month names for leniency and backwards compatibility, but the code should mainly support `JAN`, `FEB`, `MAR` and so on.

### Steps to reproduce

```Python
Read the code:
https://github.com/mne-tools/mne-python/blob/84259a673caafc3dfb37e07e9d73a3593c5c6acf/mne/io/edf/edf.py#L643-L672
```

### Expected results

Follow the EDF+ specification, with appropriate forgiveness if needed.

### Actual results

Not compliant with the EDF+ specification.

### Additional information

[EDF and EDF+ specifications](https://www.edfplus.info/specs/index.html)

DimitriPapadopoulos, 2023-03-27:
The date stuff can be fixed by using [Babel](https://babel.pocoo.org/), or perhaps temporarily switching the locale to avoid an additional dependency.

DimitriPapadopoulos, 2023-03-27:
Not a Python bug after all. The documentation states:
> The [`locale`](https://docs.python.org/3/library/locale.html#module-locale) module is implemented on top of the `_locale` module, which in turn uses an ANSI C locale implementation if available.

So, the default locale in Python is ANSI C, not the system locale. This is, of course, a good thing for reproducibility, but slightly disturbing. This is especially true when you're attempting to debug I/O on EDF files that are not compliant – I am attempting to read EDF files that contain Dutch month abbreviations, and the date is not read. What about adding a warning in that case?

DimitriPapadopoulos, 2023-03-27:
And what about (temporarily) forcing the locale in case MNE is used as a library from 3rd-party code that may have set the locale?

cbrnr, 2023-03-27:
It isn't really important for this issue, because the date is supposed to use digits anyway.

cbrnr, 2023-03-27:
Ah sorry, it is supposed to use the English three-letter abbreviations. What about checking against this set of 12 names manually?

DimitriPapadopoulos, 2023-03-27:
It works as it is, unless someone changes the Python locale. In the short therm, I feel it is more important to log warnings if something goes wrong while reading non-compliant EDF+ files.

cbrnr, 2023-03-27:
Yes, we should definitely raise warnings when encountering non-standard EDF files, at least for situations that currently work.