mne-tools/mne-python #11605: Simplify license headers in source files

cbrnr, 2023-03-28:
Currently, our source files show individual authors as well as the license in a comment near the top, for example:

```python
# Authors: Jaakko Leppakangas <jaeilepp@student.jyu.fi>
#          Robert Luke <mail@robertluke.net>
#
# License: BSD-3-Clause
```

I think this is not ideal for the following reasons:

1. It is not trivial to keep the list of authors up to date. Since this information is available from Git anyway, it is also not necessary.
2. I don't think we have always updated these lists, so currently this information is likely outdated.
3. There's always the question what kind of change(s) qualifies an author to get added to that list.

I suggest that we replace our headers with the following license text across all source files:

```python
# Authors: MNE-Python development team
#
# License: BSD-3-Clause
```

The exact wording can be changed of course (feel free to make suggestions), but the point is to get rid of individual authors.

sappelhoff, 2023-03-28:
This was also brought up by @drammock in https://github.com/mne-tools/mne-python/pull/9569#issuecomment-879969307 some time ago.

I generally agree. Do we need a copyright notice in each file? Isn't the license in the root of the repository enough, to cover it all?

cbrnr, 2023-03-28:
Removing is even better. I don't think every file needs to have the license when there is a license in the root.

larsoner, 2023-03-28:
> I don't think every file needs to have the license when there is a license in the root.

I was originally on board with this until I read [this stackexchange thread](https://softwareengineering.stackexchange.com/questions/125836/do-you-have-to-include-a-license-notice-with-every-source-file) where people sometimes (usually?) generally recommend that you do include the license so that a copied-away file still makes it clear what the license is. So I'm +0 on removing the `BSD 3-clause` line, I'm fine with keeping it or removing it.

Last time the "authors" stuff came up, IIRC (I can't find the thread, maybe it was in person at a dev meeting!) most people thought "yes these should go away" until @agramfort pointed out that it's a nice way for newer contributors to get some credit for what they do. I think we to get rid of the `authors` in `mne/` but keep those lines in `examples/` and `tutorials/` because -- even though they, too, end up out of date -- it's nice for newer and/or less experienced people to have their name visible on something that people are more likely to read (i.e., an example or tutorial) than our `whats_new.rst`.

drammock, 2023-03-28:
> I think the changelog is a lot less visible to end-users in general than examples/tutorials

this is the point (originally made by @rob-luke as @larsoner mentioned) that convinced me to keep authors on tutorials/examples, even though I still dislike the fact that they can get stale/out of date/inaccurate.

I like the idea of removing authors from the source `.py` files within `mne/` though. Anyone who is reading the source code I think it's fair to assume they at least know about changelogs, and many will know about `git blame` too. Plus GitHub shows avatars of all authors if you view the source on GitHub, so that's another way authors are visible for source files.

I'm +0.5 to keep the license lines.

agramfort, 2023-03-28:
my take on this and why I think it's valuable to keep names:

- having your name on a file gives a sense of code ownership and
responsibility.
- names on top of files survives github and where the code is stored.
- I've been contacted by people to ask questions in sklearn code as they
found my name and email on the files.
- for some contributors it's important to them to have their name. There
can be a sense of pride to have your name there and can maybe encourage
them to contribute more.

my 2c

Message ID: ***@***.***>
>

jasmainak, 2023-03-28:
> names on top of files survives github and where the code is stored.

I find this to be the most important reason to keep names ... when files are re-organized "git blame" stops working. E.g.: https://github.com/mne-tools/mne-python/blob/main/mne/report/report.py ... I don't see my name in the git commits 

[image]

even though I might be able to answer questions related to the MNE-report better than the average contributor. For functionality that is old, this may be even more critical as it might be hard to dig up who originally wrote that part of the code.

drammock, 2023-03-28:
ok, I withdraw my objection to names in the source files then. It's weird that you aren't listed by GitHub as a contrib @jasmainak; you show up in the blame of that file:

[image]

cbrnr, 2023-03-29:
I still think names in the source code only make sense if they are kept up to date. I rarely bother to put my name there, but now this makes it look as though I did not contribute. I can imagine that this is the case for others as well.

@drammock made some good points: people should not be looking at the source to find developers and ask questions. They should use our official channels. If these are not obvious, we can make them more obvisous in our documentation.

The sense of ownership and responsibility can be instilled by other means, and I think we should try to find something that works automatically. The spirit of open source is not that someone "owns" a piece of code, but we as a community have a shared responsibility for everything we create in MNE-Python. Putting some names in the sources seems to be biased and unfair unless we keep everything up to date all the time (which is not feasible).

Even if some previous contributor knows most about a specific topic I trust current developers to either answer the question or forward it to someone who they think will be able to help.

So I still think that removing names (but keeping the license) in `mne/` (but keeping them in `examples/` and `tutorials/`) is the better option.

agramfort, 2023-03-29:
@cbrnr i would say add your name to the files you feel a clear ownership for. I don't want your name in the files then don't.

cbrnr, 2023-03-29:
> i would say add your name to the files you feel a clear ownership for. I don't want your name in the files then don't.

It's not about me. It's that I don't like it in general for the reasons I mentioned.

So this is settled? No chance that *something* can be done here?

cbrnr, 2023-03-29:
Regarding the comment by @jasmainak, you show up in the blame of the file you linked to. GitHub shows the most recent contributer per line. It's the list that's buggy, we should probably report that to GitHub.

But this file demonstrates the issue I have: @hoechenberger and @larsoner have contributed a lot, but they are not mentioned at the top of the file. Whether or not that's intentional is also beside the point, because if there is a list this indicates to others that these are the developers who contributed most and are responsible for that file. Which is simply not true.

cbrnr, 2023-03-29:
> I'm okay with having the names of the original authors of a module / function listed on top. I don't care if I'm listed even if I contributed a lot, for as long as the commit history is preserved. Should we at some point be forced to kill the history, we should add all contributors to the top.

But some (new) contributors might care, but not dare to ask to include their name because they might think that only original contributors or "worthy" contributions should go in this list.

That's exactly my suggestion, either we copy the full list to the header, or none. Everything in between is not fair.

Coming back to a point I thought we'd converged on (also in a previous discussion): I would leave authors in tutorials and examples, but not in sources. This was the consensus in the linked discussion in my opinion.

Re Black, it is possible to [ignore revisions](url) - is that not sufficient? GitHub should support it.

britta-wstnr, 2023-03-29:
I agree with @agramfort here: it can be valuable for some, especially junior, developers to have their name in the source files. In my opinion, this should be marked by "substantial contribution". 

The reason why I think this is valuable is the following: having a name on a source file is an easy way to showcase that one has substantially contributed to a specific method etc. Recruitment committees of various disciplines will probably not use a command line tool or other git/GitHub functionality to figure out contributions. However, adding a link to a file where your name is up to is easy to do. 
Furthermore, people who look at the source code locally will make a connection with the names - thus increasing visibility for the contributors.

Now I agree with @cbrnr that the practice how this is done is disadvantaging contributors who are less inclined to just put themselves on the files. I wonder if what we actually need is guidelines what makes a contribution substantial enough to have your name on the file (I also see how this would always be an arbitrary cut-off).

I am against having all names up, as this would add a lot of text for potentially less substantial commits (e.g., having fixed a typo in the docstring etc.) and would decrease visibility. 

Another possibility could be to outsource the contributor list to another file we keep on our homepage - that could fix it for my first point, but again not for the visibility.

hoechenberger, 2023-03-29:
> But some (new) contributors might care, but not dare to ask to include their name 

Ok right, very valid point. I've been in this situation before and I never knew when it's justified to add my name.

drammock, 2023-03-29:
> git --no-pager log --follow --pretty=format:"%aN <%ae>" mne/report/report.py | nl | sort -uk2 | sort -nk1 | cut -f2-

If you can add this to a commit hook that automatically runs only on the changed files and then auto updates the author list (and deduplicates with our mailmap!) then I think that would be quite a good solution. Names who only made small typo fixes will still be in there but I'd rather err in that direction than to not include names who made large contributions

cbrnr, 2023-03-29:
> I'd argue that then the name list become meaningless, no? For all points mentioned here: ownership, visibility, pride, ...

But everyone who contributed has the same right to get ownership, visibility, etc.?

Since I wanted to avoid defining what an "important" contribution is, the best solution IMO is to remove author names from source files and try to add other means to provide these perks. For everyone.

britta-wstnr, 2023-03-29:
Well, my point is that with all names on there, it would not work that way.
Would someone feel ownership/responsibility if there are 15+ other names? Probably no.
Would someone get visibility if there are 15+ other names? Probably also no.

But I do think that if someone made substantial changes to a part of the code base, they should be able to get recognition for that beyond having their name in a long list of names or not mentioned at all. 
At this point I also want to throw in that this might be especially important for marginalized groups in open source, which might not be just "believed" to have "really contributed".

sappelhoff, 2023-03-29:
Having observed this a bit, I want to revise my opinion. I think we should:

- use `CODEOWNERS` for "ownership" --> this allows people to get notifications when files are touched
- use our changelog for "credit"/"attribution" --> after all it's more relevant to be able to point to a specific PR that you have implemented, rather than say "I worked on this file in general"
    - I think what would be **really** awesome would be a way to filter the changelog by contributor ... something like: list me all contributions by `@example-person` across all MNE versions
- `git blame` for specifics (the "black" commit can be filtered via a `.git-blame-ignore-revs` file, which does also work on GitHub, next to the CLI)
- keep author names in tutorials and examples, AND write some loose guidelines (preferably examples) when a contribution is considered big enough to add a name there ... OR add a "author contributions" section to each tutorial/example ... something like the [CREDIT taxonomy](https://credit.niso.org/)
- remove names and emails from files, and only keep the license abbreviation there --> credit/ownership/blame is IMHO properly solved by the above mentioned items

hoechenberger, 2023-03-29:
@britta-wstnr 
> But I do think that if someone made substantial changes to a part of the code base, they should be able to get recognition for that beyond having their name in a long list of names or not mentioned at all.

While I do agree, I also see the difficulty here that @cbrnr mentioned: how do we quantify that? How big or important is "substantial"? Changed lines of code (how many?)? Performance improvements (how much faster?)? Continuous maintenance of a module (for how long?)? Do the barriers we set here depend on seniority of the contributor (beginners -> lower, senior devs -> higher barrier?)?

There must be a clear path that contributors can follow to get their name added at the top-level. Otherwise it'll be a recipe for conflict and disappointment.

britta-wstnr, 2023-03-29:
@hoechenberger as mentioned in one of my earlier comments, I agree with that as well. 
However, that also is a problem for tutorials and examples as @sappelhoff acknowledges and proposes to write guidelines for - i.e., we have to solve that anyway.

drammock, 2023-03-29:
@britta-wstnr 
> But I do think that if someone made substantial changes to a part of the code base, they should be able to get recognition for that beyond having their name in a long list of names or not mentioned at all. At this point I also want to throw in that this might be especially important for marginalized groups in open source, who might not be just "believed" to have "really contributed".

I am very sympathetic to and supportive of these points.  But until we have a proposal for how "substantial" is defined and adjudicated, I feel I cannot support inclusion/exclusion of any names --- either show all names, or show none.

I think several of us agree that letting the contributor decide whether or not to include their name isn't a very good solution: it disadvantages anyone feeling shy/humble/insecure and advantages anyone feeling arrogant/confident/entitled.  But I also really don't want the mechanism to be "whoever is reviewing the PR decides" --- clearly there are differences of opinion among members of our steering council about how this should be handled, and whether a contributor gets credit in this way should not be determined by luck of the draw in who is available for PR reviews.

@sappelhoff 
> ideally, such a query would be associated with a specific URL that could be shared. Then I could send someone a URL and they could see all my contributions with links to the specific PRs.

This is already available: https://github.com/mne-tools/mne-python/commits?author=sappelhoff

> keep author names in tutorials and examples, AND write some loose guidelines

See above point; "loose guideline" does not, IMO, eliminate the bias toward advantaging some contribs and disadvantaging others, and it adds overhead for devs to adjudicate (or at least "keep an eye out" for whether names were (not) added (in)appropriately.

cbrnr, 2023-03-29:
Examples and tutorials are typically touched by far fewer individuals, so it should not be a problem to include all names. Even when someone "just" fixes typos (BTW I think that fixing typos is extremely valuable). So our guideline should be to include everyone who modified an example or tutorial.

cbrnr, 2023-03-29:
There's also https://github.com/all-contributors/all-contributors if that's something you would want to implement. I'm not sure I like it that much, but it's one way to increase our focus on contributors.

britta-wstnr, 2023-03-29:
To make my position clearer:

I do not think that fixing typos is "not valuable". But I think we need to find some way to give people credit who spend a lot of their time on open source work. And in a way that has little overhead for them and is easy to understand by assessment committees etc.

I fully agree that the way we do it now is not fair.

However, just erasing the names does not fix that. Adding all names does not fix that either. 
--Let me be a bit more specific: if we add all names, it will again be the ones that feel less confident who will suffer. They will not feel like they can put their work in a CV (or elsewhere), _because they have nothing to back it up with_. After all, the names now just mean you touched a file at some point. 

So in my opinion, we need to fix the fairness of the name lists (or come up with an entirely different system). Because dropping them/putting all names is not fair either.

hoechenberger, 2023-03-29:
I suppose the easiest objective measure is the number of lines a contributor touched...

jasmainak, 2023-03-29:
My understanding of the names on the top was that it is not just for visibility, but also responsibility. Those who put their names at the top of the file take some level of responsibility for the code having substantially contributed to it and *understood* how it works. For instance, I would be uncomfortable putting my name in `bem.py` even if I may have refactored some functions ... I may not be able to help someone figure out how the equations relate to Munck's paper. Presumably, by the time someone is making such substantial contributions, they would not be shy adding their name. For small and new contributions, the changelog is always there.

cbrnr, 2023-03-29:
@jasmainak that's what codeowners is for.

jasmainak, 2023-03-29:
Codeowners is for the entire repo or file-level? Would you know who to ask if something is not clear in some part of a particular file?

cbrnr, 2023-03-29:
It's file-based. You specify patterns to assign code to a person.

hoechenberger, 2023-03-29:
> Would you know who to ask if something is not clear in some part of a particular file?

Honestly in cases like this I'd just open an issue and maybe tag the dev(s) who touched that code last, as reported by `git blame`

hoechenberger, 2023-03-29:
I feel we're digressing. I believe we didn't really mean to discuss things like, which commits belong to whom and who is responsible for which subsystem / module – all stuff that's handled for us via git and GitHub; but we wanted to find ways to acknowledge contributors. Let's try to focus on this aspect! :)

jasmainak, 2023-03-29:
git and github provide powerful tools for tracking commits, modules etc. But at the end of the day, these tools cannot track everything. I have known of cases where someone contributed during a pair programming session and their contribution had to be acknowledged in other ways ... yes, you can use co-authored commits but in some cases, these people may not even have a github account since they are senior researchers. As Britta also pointed out above, committees may not be so git-savvy and may only look at the header of the files superficially. I think we need to strike a balance between being easy for developers vs easy for users/non-developers in understanding who is contributing and/or responsible for the code.

drammock, 2023-04-25:
There was discussion of this a couple weeks ago at a dev meeting. Unfortunately nobody took notes, so I'll do my best to reconstruct the conclusions reached:

1. the current way things are done is both inaccurate and unfair
2. simply removing the names (from either source files or examples/tutorials) risks doing some harm to folks whose names are there (i.e., if they rely on their name being there as public demonstration of their contributor efforts).
3. until we have a proposal that does not risk such harm, no action should be taken.

cc @britta-wstnr @agramfort @larsoner who I think were the ones present for that discussion; please correct me if I'm mistaken.

cbrnr, 2023-04-25:
I already showed a technical way to extract all authors for a file using git. So this information is there if anyone needs it. Can we make a list of persons who require this information? If there are not too many, maybe individual custom solutions are feasible (like generating a report of contributions). Are there other suggestions?

cbrnr, 2023-04-25:
We could create that solution (a report) for them.

drammock, 2023-04-25:
> maybe it could be an addition to the website with a search-like tool. It would make this summary more accessible and shareable.

hosting such an interface on our website would also allow us to curate / customize what metric(s) we choose to display. Some possibities (besides number of commits):

- number of PR reviews
- number of issues reported
- sprint or GSoC mentorship
- IDK if Discourse has an API but if they do we could potentially pull in forum stats too
- Getting funding to support the project

Separately, I wonder if it would help to try to separate the question of "what would be optimal going forward?" from the question of "what would preserve or improve on the imperfect situation we've inherited?" My hope in suggesting this is that we could reach some consensus about what is future-optimal, and then add to it an appropriate "patch" to try to minimize whatever harm it would do if we ignored the imperfect current/past state of affairs. Thoughts?

hoechenberger, 2023-04-25:
> IDK if Discourse has an API

They do, https://docs.discourse.org/

cbrnr, 2023-04-26:
I like the suggestion to separate the optimal solution from incremental improvements. Otherwise, we might not implement anything at all for a long time. Again, I would like to ask if anyone can give _concrete_ examples of why and which specific statistics are important to have. This would greatly help us reduce efforts in the spirit of YAGNI.

A simple and quick (but maybe still good) improvement could be to archive all author headers in a common file, so that this information is still visible for anyone who needs it. This could be a simple text/Markdown file demonstrating contributions from particular authors (those who happened to be in the file headers) to particular files. Then we could remove authors from sources (but keep and encourage them in examples/tutorials).

sappelhoff, 2023-04-26:
> A simple and quick (but maybe still good) improvement could be to archive all author headers in a common file, so that this information is still visible for anyone who needs it.

don't we have something like that here? https://github.com/mne-tools/mne-python/blob/909e45821bd004b6072a6426e4e8ea88e51b72ad/doc/changes/names.inc#L1

cbrnr, 2023-04-26:
I understand these problems, but what I don't understand is why this is tied to having a name in the header of the source file. It's what MNE happened to choose, but I don't think that this proves contributions/competence/worth etc. when we already have other ways to demonstrate this (GitHub, git, our changelog with contributors). I think having selected names in source headers does more harm than good, so we should try to improve on the current situation rather sooner than later.

cbrnr, 2023-04-26:
> I have personally heard from folks that they used the fact that their name was in the source file to prove the extent of their contributions in the context of a job hire or promotion interaction. This is not a hypothetical. Removing their name removes their power to do that.

OK. I think they could have proved it with some other measure as well. That's exactly my point,  just because names in the source seemed to have worked doesn't mean that there are no alternatives.

hoechenberger, 2023-04-26:
I wonder if specifically for the act of job applications we could create a script that uses git to generate some stats PLUS allows for manual insertion of additional accomplishments / comments that can be added by a core team member. This information is then used to create a digitally signed certificate whose authenticity can be verified online. Just like the certificates of completion you can get via LinkedIn Learning

drammock, 2023-04-26:
> OK. I think they could have proved it with some other measure as well.

Why do you think that?  Why not believe them when they say "this is what worked"?

hoechenberger, 2023-04-26:
> I wonder if specifically for the act of job applications we could create a script that uses git to generate some stats PLUS allows for manual insertion of additional accomplishments / comments that can be added by a core team member. This information is then used to create a digitally signed certificate whose authenticity can be verified online. Just like the certificates of completion you can get via LinkedIn Learning. We could vouch for the contributions and skills of a contributor. 
> 
> 

I'd volunteer to draft such a document template

hoechenberger, 2023-04-26:
> > OK. I think they could have proved it with some other measure as well.
> 
> 
> 
> Why do you think that?  Why not believe them when they say "this is what worked"?

He was just suggesting that because one approach worked doesn't mean that another would not have worked. However if we want to work like scientists (evidence-based), I'd also say: what ya worked in the past has been shown to work, so why take the risk of switching to something else that might fail.

cbrnr, 2023-04-26:
> Why do you think that? Why not believe them when they say "this is what worked"?

I never said I didn't believe them. I just think that very likely other things might have worked as well. Re why not sticking to what has worked in the past, I think we have now identified the points why the current situation is less than ideal.

hoechenberger, 2023-04-26:
> > > OK. I think they could have proved it with some other measure as well.
> > 
> > 
> > 
> > Why do you think that?  Why not believe them when they say "this is what worked"?
> 
> He was just suggesting that because one approach worked doesn't mean that another would not have worked. However if we want to work like scientists (evidence-based), I'd also say: what has worked in the past has been shown to work, so why take the risk of switching to something else that might fail. 

The more I think about it, the more I like this idea. Imagine you could produce a document that says you've worked on MNE **and** it bears the signature of a certain Alexandre Gramfort (if Alex willing to play along 😅) This would provide a **huge** credibility boost, which, as was pointed out, might be especially beneficial to contributors from marginalized groups

hoechenberger, 2023-04-26:
> I'd volunteer to draft such a document template

The more I think about it, the more I like this idea. Imagine you could produce a document that says you've worked on MNE and it bears the signature of a certain Alexandre Gramfort (if Alex is willing to play along 😅) This would provide a huge credibility boost, which, as was pointed out, might be especially beneficial to contributors from marginalized groups

jasmainak, 2023-04-27:
I think it's not easy to fix the problem of credit assignment at the level of MNE. The person evaluating a certain application will use their own metrics. If it's a person in industry, they will see github contributions favorably, but in academia, the majority still don't understand open source. If there are 20-30 authors/contributors, the evaluators have a hard time wrapping their head around your contribution percentage. I don't think it's possible to put a number on this and this is the core problem of open source cred assignment. 

Job applications and grant or faculty hiring committees are one example, green card applications are another (which in turn has exponential impact on other stuff). The evaluator could be a medical doctor or an army veteran who has no idea Alex is (famous as he is :)

cbrnr, 2023-04-27:
@jasmainak the same argument applies to having a name in a source file header. It's all pretty random/arbitrary, so I am sure that we do not have to keep names for the sake of some committee probably wanting to take a look at some source file. I completely agree with you that credit assignment for open source work is a huge problem in academia (normally this doesn't count at all).

hoechenberger, 2023-04-27:
@jasmainak 
> I think it's not easy to fix the problem of credit assignment at the level of MNE. The person evaluating a certain application will use their own metrics. If it's a person in industry, they will see github contributions favorably, but in academia, the majority still don't understand open source. If there are 20-30 authors/contributors, the evaluators have a hard time wrapping their head around your contribution percentage. I don't think it's possible to put a number on this and this is the core problem of open source credit assignment.

I think we're somewhat going in circles here. We have evidence that names in source files have worked for job applicants in the past. It was also argued that Git history may or may not work, depending on the job you're applying for. Lastly it was brought up that esp. for marginalized groups, **even providing proof** may not be enough (at least that's how I understood it) unless a respectable, trusted source vouches for these contributors. **This is the specific issue I'd wanted to address with my proposal.** I didn't mean to invalidate or suggest to get rid of any of the other approaches, but to introduce a new service (a certificate of contribution, if you will) for folks who believe it could help them land their next job / enter the next career stage. These certs would only be issued upon request, and the way to issue the request needs to be designed such that it is super low-threshold (filling in a simple form on our website or so).

> Job applications and grant or faculty hiring committees are one example, green card applications are another (which in turn has exponential impact on other stuff). The evaluator could be a medical doctor or an army veteran who has no idea Alex is (famous as he is :)

But at least there would be **some** reference, which can be important for job applications.

If I've worked at a company, I can put my old boss as a reference. If I've contributed to MNE-Python once, of course I **could** reach out to the core team and ask if somebody would be willing to act as my reference, but there's a huge barrier there. Issuing a certificate that credibly proves I've been involved with the project, **and** has the name and email address of a responsible person (core dev team member) on it, creates an aura of trustworthiness and credibility.

jasmainak, 2023-04-27:
What I'm trying to say is that it depends on the job application and the level of the candidate. In some situations, it may be enough to show "some" contributions. In other kind of job applications, you have to come across as a major contributor. If you are a Masters student applying for a PhD or to a company, I can see how such a certificate or a recommendation letter may help. But in other situations, it is pretty unhelpful if the person evaluating cannot see that you made important contributions. I'm suggesting mindfulness to the fact that there is a diversity of situations to consider. 

Being able to pinpoint yourself as a major contributor to specific submodules or sister packages of MNE will probably be more helpful in more situations than having your name buried in a list of 50 or even 10 names. 

@hoechenberger to avoid going in circles :)

> We have evidence that names in source files have worked for job applicants in the past. It was also argued that Git history may or may not work, depending on the job you're applying for. 

yes, I agree with you here

> Lastly it was brought up that esp. for marginalized groups, even providing proof may not be enough (at least that's how I understood it) 

I just don't know if a generic certificate would be helpful in addressing this issue. Maybe it would help in some limited circumstances ... I would ask a person who might use it.

cbrnr, 2023-04-27:
Well, then we can make sure to put something on our website that this does not happen. My point is that names in source files are at least as flaky given that there aren't even rules who gets to be in that list.

cbrnr, 2023-04-27:
And if this is really critical for someone, I think this person will find ways to show his/her contributions to the committee even when there are no names in source headers. It just involves a little work, but such things are very individual that I think this is best handled on a case by case basis instead of relying on our current source name bingo (sorry, but that's what it basically is right now).

hoechenberger, 2023-04-27:
> I can vouch that this actually does happen. I had a certain review committees dismiss some of my contributions to some other open source work based on my name missing from a website (which it wasn't but whatever) ...

But the website isn't the source code either ... so in this case a cert like I proposed would've helped, or a page where all contributors are listed in alphabetical order, potentially with a link to that GH page that would list all commits by that user, which was suggested somewhere earlier in this thread (I cannot seem to find the posting right now), no?

drammock, 2023-04-27:
> a page where all contributors are listed in alphabetical order, potentially with a link to that GH page that would list all commits by that user, which was suggested somewhere earlier in this thread (I cannot seem to find the posting right now)

Perhaps you mean [Mathieu's comment](https://github.com/mne-tools/mne-python/issues/11605#issuecomment-1522422187) and/or [my response to it](https://github.com/mne-tools/mne-python/issues/11605#issuecomment-1522436320)

hoechenberger, 2023-04-27:
> Perhaps you mean [Mathieu's comment](https://github.com/mne-tools/mne-python/issues/11605#issuecomment-1522422187) and/or [my response to [...]

Yea, that's it!

britta-wstnr, 2023-04-28:
> instead of relying on our current source name bingo (sorry, but that's what it basically is right now).

I think the discussion never was about _keeping the source file names forever_ but more about _keeping them until we have a better system in place_.

In principle, I like the idea of having this be a part of our homepage and especially extending it by things like the list @drammock created.

One of the things I am not so sure about yet is what @jasmainak mentioned as well: if your name is on a source file that you can easily link in an application, this (for better or for worse) shows you had _substantial_ contribution to a method. If we generate a report with all PRs, this is harder to see. Granted, there is PR names etc., but this would mean the search committee has to look through the PR list, know what a "substantial" PR name looks like or believe the candidate when the candidate says "PR x was substantial". Regarding this:

>  I think this person will find ways to show his/her contributions to the committee even when there are no names in source headers. It just involves a little work

This I would like to contest, as we cannot just assume that everybody has the privilege to be believed after putting in "a little work". The playing field unfortunately is not the same for everyone here.

Now, again, I don't thinkwe have to keep the header _as is_ forever, but I think substantial contributions is one of the major "features" of source file names (biased as it might be right now) we maybe do not translate well yet.

Another one is visibility - while users can stumble across names in source files and then connect a persons name to a certain method/MNE-Python, a report on a homepage needs to be produced to be visible.

And a last thought for now: if we report those data on our webpage, we probably also need a way to be contacted about it, whether that be for wanting to be erased from the build-report-button* or whether someone thinks the reported info is not correct (e.g. because they were part of one more sprint or something like this).

*while PR lists are public information, other things might not be, e.g. sprint mentoring/attendance and the like.

drammock, 2023-04-30:
Just came across this: https://github.com/mntnr/name-your-contributors

posting it as an addition to the possibilities already mentioned in [this](https://github.com/mne-tools/mne-python/issues/11605#issuecomment-1522436320) and [this](https://github.com/mne-tools/mne-python/issues/11605#issuecomment-1523611259) previous comment.

larsoner, 2023-10-03:
@agramfort, @drammock and I did a little bit of investigation. When you google authors (we tried @jasmainak and @britta-wstnr for example) you do not get links to source files, even if you go pretty deep in the results. You do (sometimes as the first result!) get links to examples/ and tutorials/. Based on this, the cumulative discussion and ideas above, and some additional discussion today with @britta-wstnr, @wmvanvliet, and @jasmainak (hooray sprint time making chatting much easier!) we came up with the following tentative plan.

1. Remove names from `mne/` source files , replaced with:
   ```
   # Authors: MNE-Python developers <https://mne.tools/stable/credit.html>
   # License: BSD-3-Clause
   ```
   Source files are not easily found by Google so don't give great credit anyway. Also sometimes people email the authors for help with code *which is a bug not a feature* -- people should use standard communication channels (e.g., discourse).

2. Create a GitHub action that:

   1. Populates a `doc/credit.rst` that is roughly organized with public submodules as headings. Under each heading is a list of all authors of the file (ordered by credit, probably +/- lines over the history). This should be at least as discoverable / reportable for people who need it as having the names in the source files since it will be on our webpage, and we can link to it directly from the front page somehow.
   2. Populates `examples/` and `tutorials/` Python files with all contributor names to a given example/tutorial. Since these files are already discoverable by search, we can easily keep it this way by preserving and extending/completing these automatically.

Hopefully this preserves credit "backward compatibility" sufficiently while improving things overall, especially in terms of needing to make decisions about what to include or exclude with each contribution/PR. Someone (me probably!) "just" needs to implement this stuff but really I don't think it will be too onerous having played around a bit in #11975 and with GitHub actions recently. And critically it's extensible and adjustable as needed, for example we can add sections on grant support and Discourse activity that we currently don't credit sufficiently (if at all).

[13 more replies]

PR #12781 (2024-08-14): DOC: Fix headers
1. Format top of lines everywhere in MNE except examples/tutorials as:
   ```
   # Authors: The MNE-Python contributors.
   # License: BSD-3-Clause
   # Copyright the MNE-Python contributors.
   ```
2. In tutorials/examples, existing `Authors:` lines will be preserved, otherwise lines will be as above.
3. Add `ensure_headers.py` to `autofix` so the next time we add a file it will get this info automatically. I expect it to almost immediately push a (large!) commit to this PR. So after we merge this, we can add the squashed commit to `.git-blame-ignore-revs`. This will lose the [... cut]
closed 2024-08-14 by larsoner