mne-tools/mne-python #11608: mne.channels.read_ch_adjacency 'picks' argument does not function as described

aaronjnewman, 2023-03-28:
### Proposed documentation enhancement

The explanation of the `picks` for mne.channels.read_ch_adjacency reads:

> Channels to include. Slices and lists of integers will be interpreted as channel indices. In lists, channel type strings (e.g., ['meg', 'eeg']) will pick channels of those types, channel name strings (e.g., ['MEG0111', 'MEG2623'] will pick the given channels. Can also be the string values “all” to pick all channels, or “data” to pick [data channels](https://mne.tools/stable/glossary.html#term-data-channels). None (default) will pick all channels. Note that channels in info['bads'] will be included if their names or indices are explicitly provided. Picks must match the template.

However, `mne.channels.read_ch_adjacency('elec1005', picks=['eeg'])` generates an error:

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
<ipython-input-81-6e28f3bd522a> in <module>
----> 1 mne.channels.read_ch_adjacency('elec1005', picks=['eeg'])

/opt/conda/lib/python3.9/site-packages/mne/channels/channels.py in read_ch_adjacency(fname, picks)
   1504     nb = loadmat(fname)['neighbours']
   1505     ch_names = _recursive_flatten(nb['label'], str)
-> 1506     picks = _picks_to_idx(len(ch_names), picks)
   1507     neighbors = [_recursive_flatten(c, str) for c in
   1508                  nb['neighblabel'].flatten()]

/opt/conda/lib/python3.9/site-packages/mne/io/pick.py in _picks_to_idx(info, picks, none, exclude, allow_empty, with_ref_meg, return_kind)
   1011         raise ValueError('picks must be 1D, got %sD' % (picks.ndim,))
   1012     if picks.dtype.char in ('S', 'U'):
-> 1013         picks = _picks_str_to_idx(info, picks, exclude, with_ref_meg,
   1014                                   return_kind, extra_repr, allow_empty,
   1015                                   orig_picks)

/opt/conda/lib/python3.9/site-packages/mne/io/pick.py in _picks_str_to_idx(info, picks, exclude, with_ref_meg, return_kind, extra_repr, allow_empty, orig_picks)
   1046     # special case for _picks_to_idx w/no info: shouldn't really happen
   1047     if isinstance(info, int):
-> 1048         raise ValueError('picks as str can only be used when measurement '
   1049                          'info is available')
   1050 

ValueError: picks as str can only be used when measurement info is available
```

The same error is generated if I provide a list of channel names, e.g.: 

```
chs = ['Fp1', 'Fp2', 'F7', 'F3', 'Fz', 'F4', 'F8', 'FC5', 'FC1', 'FC2', 'FC6', 'T7', 'C3', 'Cz', 'C4', 'T8', 'TP9', 'CP5', 'CP1', 'CP2', 'CP6', 'TP10', 'P7', 'P3', 'Pz', 'P4', 'P8', 'PO9', 'O1', 'Oz', 'O2', 'PO10']
mne.channels.read_ch_adjacency('elec1005', picks=chs)
```

Am I mis-reading the docs, or is this an error either in the code or the docs?

drammock, 2023-03-28:
I think the docstring is wrong. The built-in neighbor files don't keep track of channel type information, so picking by channel type can't work.  The second case (giving a list of channel names) could work though...  easiest might be to create a fake Info from the ch_names present in the neighbors file, and pass that to `_picks_to_idx` instead of passing `len(ch_names)`.  @aaronjnewman do you feel up to making a PR that does that?

sappelhoff, 2023-11-13:
Note: @dominikwelke is working on this

dominikwelke, 2023-11-13:
actually, I just see that this has already been adressed (also the docstring was changed, i just missed this as i first looked at the stable doc).

so this can be closed I guess

closed 2023-11-13 by sappelhoff