mne-tools/mne-python #11611: In EDF+, for backwards compatibilty with EDF, the date must be in the range 01-01-1985 to 31-12-2084 inclusive.

larsoner, 2023-03-29:
In EDF+, for backwards compatibilty with EDF, the date must be in the range 01-01-1985 to 31-12-2084 inclusive.  
 
              The string "Startdate X X X X " in the recording field is valid in EDF+ but the short date notation at offset  
0xA8 must always be present.  
  
              For anonymization purposes, the consensus is to use 01-01-1985 00:00:00.  

              By being tolerant/lenient you'll acomplish to stay friends with your userbase (at least in the short term),
              but in the long term you'll hurt EDF(+) compatibility and ultimately the people who invested in EDF recording data bases.  
              It's better to disappoint them now (the people having trouble reading non-compliant files),  
              than to end up with a big pile of non-compliant files.  
  
              All non-compliant files that can still be accessed by being tolerant, can also be fixed by EDFbrowser.  
              If a non-compliant file can not be fixed by EDFbrowser, it can not be read (and have useful data) by being tolerant.  
              If you don't like the dependency to EDFbrowser for fixing non-compliant files, feel free to have a look at  
              the code and write your own EDF-fixer in Python.  
              That's how you help people and preserve the future of the EDF format at the same time.  
  
              Rationale: By refusing to open non-compliant files but offering a tool that modifies these files and fix them,
              you make people aware of the problem but still give them a solution.  
              The advantage is that they will usually report back / complain to the person responsible for the non-compliant  
              files which is a good thing.

_Originally posted by @Teuniz in https://github.com/mne-tools/mne-python/issues/11603#issuecomment-1488676849_

cbrnr, 2023-03-29:
I'm OK with being more strict as long as we provide some way to read certain non-standard files. I do not want to end up being able to read fewer EDFs than before. One way to implement this could be a `mode` flag, which by default could be `"strict"` and only accept 100% valid EDFs. A `"loose"` mode (which would have to be manually set) could then try to read as much data as possible while ignoring any EDF standard violations. Actually, not any violations, but most things that we encounter here are invalid characters (not using ASCII but some other encoding) and invalid dates.

larsoner, 2023-03-29:
While it's a bit invasive for users, I agree supporting community compliance would be nice.

> One way to implement this could be a mode flag, which by default could be "strict" and only accept 100% valid EDFs. A "loose" mode (which would have to be manually set)

I was thinking something similar -- but a bit stricter. I was thinking a new `incompliant='error' | 'warn'` which raises an error (by default) when a non-compliant file is found saying "please fix using EDFBrowser". We could have this in our code for 2 releases (longer than usual deprecation cycles) saying `incompliant` will be removed in 1.6 or so. This gives people time to convert their files, and for users to complain if the time seems too short. Then after that window, non-compliant files need to be fixed using the tool designed to fix them (EDFBrowser).

This is also nicer in terms of maintainability for us because we don't have to keep deciding and implementing code to handling more and more noncompliant EDF files. We can leave that to EDFBrowser.

cbrnr, 2023-03-29:
If we really do this, I'd like to check some of the EDF files I've been using, especially those available in public databases. I have the impression that many of those files are non-compliant, and I would not want to be forced to manually convert them before I can use them.

larsoner, 2023-03-29:
> especially those available in public databases. I have the impression that many of those files are non-compliant, and I would not want to be forced to manually convert them before I can use them.

Following @Teuniz's argument ideally what would happen here is, if you find non-compliance, you (or someone else) would fix them and update the public database to everyone's benefit.

cbrnr, 2023-03-29:
That's not going to happen in practice.

DimitriPapadopoulos, 2023-04-04:
If we keep a “_loose_” mode, it could be based on the current code.

An “_error_” mode could be based on [EDFlib-Python](https://pypi.org/project/EDFlib-Python/). That's one additional dependency, but why reinvent the wheel?

cbrnr, 2023-04-04:
Or maybe `engine="mne"` and `engine="pyedflib"` to be even more flexible (and in line with e.g. what pandas does)?

DimitriPapadopoulos, 2023-04-04:
The drawback of both [EDFlib-Python](https://pypi.org/project/EDFlib-Python/) and [pyEDFlib](https://pypi.org/project/pyEDFlib/) is that they only emit the general error message "_File is not valid EDF(+) or BDF(+)._" I would prefer more precise error messages. Is that enough a drawback to re-invent the wheel? I think not.

cbrnr, 2023-04-04:
Maybe. We could do it piecemeal. If you have specific warnings you need to add, we could do that in a strict mode. I'd prefer that slightly over using the two external dependencies, because pyedflib is not pure Python and edflib-python is slow.