mne-tools/mne-python #11628: Plot raw data - visualization bug?

vpKumaravel, 2023-04-14:
### Description of the problem

Hello!

Today I noticed something strange with the plot UI. I describe it as follows:

1) I used Maxwell function to detect bad channels and I plot the raw data without "Butterfly" effect and the data looks as below.

[image]

2) Then, I clicked on the channel MEG2032 because my algorithm predicts that as a bad channel, too. However, looking at the plot above, MEG2032 seems clean. While trying to investigate, I clicked on that channel and the UI marks it in red color - which is fine, but the waveform changed abruptly to something below.

[image]

I wonder if the UI is designed this way or a bug. Please provide your comments.

### Steps to reproduce

```Python
1. Import raw data
2. Detect bad channels using Maxwell filter by executing the following snippet:

    raw_check.info['bads'] = []
    auto_noisy_chs, auto_flat_chs = find_bad_channels_maxwell(
        raw_check, return_scores=False, verbose=True)
    print(auto_noisy_chs)  # we should find them!
    print(auto_flat_chs)  # none for this dataset

    raw_check.info['bads'] = auto_noisy_chs + auto_flat_chs
    for title, data in zip(['Maxwell Detection'], [raw_check]):
        with mne.viz.use_browser_backend('matplotlib'):
            fig = data.plot(butterfly=False, color='#00000022', bad_color='r')
        fig.subplots_adjust(top=0.9)
        fig.suptitle(title, size='xx-large', weight='bold')

3. Click on a channel not detected as bad by Maxwell function (i.e., the ones in grey color).
4. You might observe the change in shape of the data as I experienced.
```

### Expected results

I'd expect the shape of the data to be the same whether I click on a particular sensor or not.

### Actual results

As described above, the shape changes with a mouse click over the sensor's data.

### Additional information

mne.sys_info()
Platform:         Windows-10-10.0.19044-SP0
Python:           3.8.16 (default, Mar  2 2023, 03:18:16) [MSC v.1916 64 bit (AMD64)]
Executable:       C:\ProgramData\Anaconda3\envs\NEAR\python.exe
CPU:              Intel64 Family 6 Model 158 Stepping 10, GenuineIntel: 8 cores
Memory:           Unavailable (requires "psutil" package)
mne:              1.3.1
numpy:            1.24.2 {OpenBLAS 0.3.21 with 8 threads}
scipy:            1.10.1
matplotlib:       3.7.1 {backend=module://backend_interagg}
sklearn:          1.2.2
numba:            Not found
nibabel:          Not found
nilearn:          Not found
dipy:             Not found
openmeeg:         Not found
cupy:             Not found
pandas:           1.5.3
pyvista:          Not found
pyvistaqt:        Not found
ipyvtklink:       Not found
vtk:              Not found
qtpy:             Not found
ipympl:           Not found
pyqtgraph:        Not found
pooch:            v1.7.0
mne_bids:         Not found
mne_nirs:         Not found
mne_features:     Not found
mne_qt_browser:   Not found
mne_connectivity: Not found
mne_icalabel:     Not found

larsoner, 2023-04-14:
Are projectors on? Like if you click the "Prj" button in the lower right are things checked? If so, turn them all off and you should see that the data does not change. If this is the case then everything is working correctly. A good channel will be included in projection, but once you mark it as bad it gets excluded from projection, which will cause its values to change (along with values of other channels still included in the projection, albeit less noticably in most cases!).

vpKumaravel, 2023-04-14:
Thanks, @larsoner. If I toggle them all projectors to OFF, there is no change. I didn't know about the projectors earlier. Is there a way to disable them programmatically?

closed 2023-04-14 by larsoner